arendst/Tasmota #3387: STATUS on mqtt

provenza, 2018-07-31:
hi,

is there a way to receive status of a sonoff via command line and mosquitto?

`$ mosquitto_pub -h localhost -p 1883 -t '/cmnd/sonoffbasic02/status' -m ''  `

provide status on mqt console, is there a way to receive a response in terminal?

Frogmore42, 2018-08-01:
Mosquitto_sub

provenza, 2018-08-01:
there's non way to await for a new update and receive immediatly using pub?

Jason2866, 2018-08-01:
I dont understand your question, you get every change immediatly via mqtt through the broker
Tasmota is a software to make out of sonoff devices a mqtt device.

Frogmore42, 2018-08-01:
Mosquitto_pub publishes messages
Mosquito_sub subscribes to topics to get messages that are published.
It sounds like you would benefit from reading a tutorial on MQTT.  Search the internet and I am sure you will find one in your native language.

closed 2018-08-01 by provenza