arendst/Tasmota #3388: Please provide a <noscript> element in the web interface

josch, 2018-08-01:
Without JavaScript activated, one sees this in the settings interface:

[image]

None of the drop-downs are clickable.

Because the interface is so basic and simple, I didn't expect that it would require JavaScript and assumed the problem to be elsewhere.

It would be great if a `<noscript>` entry could be put at the top of the page informing the user that JavaScript is required to use the web interface.

Thanks!

arendst, 2018-08-02:
It's heavily depending on javascvript as all select options are only send once to the client as javascript array to be used by the client page rendering. This way the page itself is small enough to fit in ESP8266 RAM before it is send to the webserver.

ascillato2, 2018-08-18:
Hi,

Closing this issue as your questions has been addressed. Ask to reopen this issue if you need it. Thanks

closed 2018-08-18 by ascillato2