arendst/Tasmota #3389: Sonoff Basic - WiFi unstable after upgrade from 5.13.1 to 6.1.1

tlc76, 2018-08-01:
**Describe the bug**

**Before upgrade:**
- ICMP response time was 1-2 ms (sometimes but very rarely 10-12 ms)
- zero packets lost
- compiled with Arduino IDE
```
Reply from 10.0.0.178: bytes=32 time=44ms TTL=128
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128
Reply from 10.0.0.178: bytes=32 time=10ms TTL=128
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=5ms TTL=128
Reply from 10.0.0.178: bytes=32 time=6ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128
Reply from 10.0.0.178: bytes=32 time=11ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=1ms TTL=128
Reply from 10.0.0.178: bytes=32 time=6ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128
```

After upgrade:
- average ICMP is ~50 ms (going up to 1000-2000 ms)
- 2% packet loss
- compiled with ESP library 2.3.0 under PlatformIO
```
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=60ms TTL=128
Reply from 10.0.0.178: bytes=32 time=30ms TTL=128
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128
Reply from 10.0.0.178: bytes=32 time=37ms TTL=128
Reply from 10.0.0.178: bytes=32 time=51ms TTL=128
Reply from 10.0.0.178: bytes=32 time=72ms TTL=128
Reply from 10.0.0.178: bytes=32 time=39ms TTL=128
Reply from 10.0.0.178: bytes=32 time=48ms TTL=128
Reply from 10.0.0.178: bytes=32 time=4ms TTL=128
Reply from 10.0.0.178: bytes=32 time=40ms TTL=128
Reply from 10.0.0.178: bytes=32 time=19ms TTL=128
Reply from 10.0.0.178: bytes=32 time=15ms TTL=128
Reply from 10.0.0.178: bytes=32 time=23ms TTL=128
Reply from 10.0.0.178: bytes=32 time=4ms TTL=128
Reply from 10.0.0.178: bytes=32 time=39ms TTL=128
Request timed out.
Request timed out.
Reply from 10.0.0.178: bytes=32 time=2043ms TTL=128
Reply from 10.0.0.178: bytes=32 time=7ms TTL=128
Reply from 10.0.0.178: bytes=32 time=1118ms TTL=128
Reply from 10.0.0.178: bytes=32 time=124ms TTL=128
Reply from 10.0.0.178: bytes=32 time=46ms TTL=128
Reply from 10.0.0.178: bytes=32 time=7ms TTL=128
Reply from 10.0.0.178: bytes=32 time=12ms TTL=128
Reply from 10.0.0.178: bytes=32 time=87ms TTL=128
Reply from 10.0.0.178: bytes=32 time=343ms TTL=128
```

- [x ] Searched the problem in issues and in the wiki
- [x] Hardware used : 
Sonoff Basic R2 v1.0
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE (command executed only under 6.1.1 firmware):
12:36:55 CMD: status 0
12:36:55 MQT: tasmota/sonoff1/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
12:36:55 MQT: tasmota/sonoff1/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:19:19","StartupUTC":"2018-08-01T09:17:36","Sleep":0,"BootCount":5,"SaveCount":12,"SaveAddress":"F8000"}}
12:36:55 MQT: tasmota/sonoff1/stat/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-08-01T12:10:37","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:36:55 MQT: tasmota/sonoff1/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":1,"LogHost":"10.0.0.3","LogPort":514,"SSId":["tlc3","tlc"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
12:36:55 MQT: tasmota/sonoff1/stat/STATUS4 = {"StatusMEM":{"ProgramSize":537,"Free":464,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","0C000000","23B6179E","00000000"]}}
12:36:55 MQT: tasmota/sonoff1/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff1","IPAddress":"10.0.0.178","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.1","Mac":"5C:CF:7F:7A:73:27","Webserver":2,"WifiConfig":3}}
12:36:55 MQT: tasmota/sonoff1/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.3","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7A7327","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:36:55 MQT: tasmota/sonoff1/stat/STATUS7 = {"StatusTIM":{"UTC":"Wed Aug 01 09:36:55 2018","Local":"Wed Aug 01 12:36:55 2018","StartDST":"Sun Mar 25 03:00:00 2018","EndDST":"Sun Oct 28 04:00:00 2018","Timezone":99,"Sunrise":"06:02","Sunset":"20:40"}}
12:36:55 MQT: tasmota/sonoff1/stat/STATUS10 = {"StatusSNS":{"Time":"2018-08-01T12:36:55"}}
12:36:55 MQT: tasmota/sonoff1/stat/STATUS11 = {"StatusSTS":{"Time":"2018-08-01T12:36:55","Uptime":"0T00:19:19","Vcc":3.140,"POWER":"OFF","Wifi":{"AP":1,"SSId":"tlc3","RSSI":100,"APMac":"F0:B4:29:4F:18:45"}}}
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
Sometimes this bug causes a delay in executing Toggle commands from the web interface or even the command is not executed at all.

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
- No changes in WiFi AP configuration
- The Sonoff Basic device is in the same physical location (signal strength 100%)
- Upgrade was done via OTA (two steps method via BE_MINIMAL, both steps with 6.1.1 binaries compiled under Visual Studio Code + PlatformIO)

**(Please, remember to close the issue when the problem has been addressed)**

tlc76, 2018-08-01:
Checked again 10 minutes later, ICMP packet loss is massive: 22%
```
Pinging sonoff1.lan [10.0.0.178] with 32 bytes of data:
Reply from 10.0.0.178: bytes=32 time=81ms TTL=128
Reply from 10.0.0.178: bytes=32 time=9ms TTL=128
Reply from 10.0.0.178: bytes=32 time=7ms TTL=128
Request timed out.
Reply from 10.0.0.178: bytes=32 time=54ms TTL=128
Reply from 10.0.0.178: bytes=32 time=15ms TTL=128
Reply from 10.0.0.178: bytes=32 time=18ms TTL=128
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128
Request timed out.
Reply from 10.0.0.178: bytes=32 time=4ms TTL=128
Request timed out.
Reply from 10.0.0.178: bytes=32 time=9ms TTL=128
..........
Reply from 10.0.0.178: bytes=32 time=8ms TTL=128
Reply from 10.0.0.178: bytes=32 time=67ms TTL=128
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128

Ping statistics for 10.0.0.178:
    Packets: Sent = 62, Received = 48, Lost = 14 (22% loss),
Approximate round trip times in milli-seconds:
    Minimum = 3ms, Maximum = 135ms, Average = 29ms
```

Jason2866, 2018-08-01:
No problem here with 6.1.1c with core 2.3.0

d:\Sonoff-Tasmota-development.6.1.1c.2907\sonoff>ping -t 192.168.2.126

```
Ping wird ausgeführt für 192.168.2.126 mit 32 Bytes Daten:
Antwort von 192.168.2.126: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=113ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=114ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=27ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=5ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.126: Bytes=32 Zeit=3ms TTL=128

Ping-Statistik für 192.168.2.126:
    Pakete: Gesendet = 38, Empfangen = 38, Verloren = 0
    (0% Verlust),
Ca. Zeitangaben in Millisek.:
    Minimum = 3ms, Maximum = 114ms, Mittelwert = 9ms
```

tlc76, 2018-08-01:
are you using the 6.1.1 stable version or the development one?
thanks

Jason2866, 2018-08-01:
Using 6.1.1c development. Tried on a S20:

```
d:\Sonoff-Tasmota-development.6.1.1c.2907\sonoff>ping -t 192.168.2.119

Ping wird ausgeführt für 192.168.2.119 mit 32 Bytes Daten:
Antwort von 192.168.2.119: Bytes=32 Zeit=42ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=50ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=1ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=1ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=1ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=26ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=1ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=5ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=1ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128
Antwort von 192.168.2.119: Bytes=32 Zeit=2ms TTL=128

Ping-Statistik für 192.168.2.119:
    Pakete: Gesendet = 64, Empfangen = 64, Verloren = 0
    (0% Verlust),
Ca. Zeitangaben in Millisek.:
    Minimum = 1ms, Maximum = 50ms, Mittelwert = 3ms
```

Jason2866, 2018-08-01:
I have completly disabled wifimanager. Maybe?
Here is the part of my user_config_override.h

```
#undef  WIFI_CONFIG_NO_SSID                                                 
#define WIFI_CONFIG_NO_SSID    WIFI_SERIAL       // Default tool if wifi fails to connect and no SSID is configured
                                                 //   (WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_SERIAL)
                                                 //   *** NOTE: When WPS is disabled by USE_WPS below, WIFI_WPSCONFIG will execute WIFI_MANAGER ***
                                                 //   *** NOTE: When WIFI_MANAGER is disabled by USE_WEBSERVER below, WIFI_MANAGER will execute WIFI_SMARTCONFIG ***
                                                 //   *** NOTE: When WIFI_SMARTCONFIG is disabled by USE_SMARTCONFIG below, WIFI_SMARTCONFIG will execute WIFI_SERIAL ***

#undef  WIFI_CONFIG_TOOL
#define WIFI_CONFIG_TOOL  WIFI_WAIT             // [WifiConfig] Default tool if wifi fails to connect
                                                //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT, WIFI_SERIAL)
// -- Wifi Config tools ---------------------------
#undef  USE_WPS                                  // Add support for WPS as initial wifi configuration tool (+33k code, 1k mem (5k mem with core v2.4.2+))
#undef  USE_SMARTCONFIG                          // Add support for Wifi SmartConfig as initial wifi configuration tool (+23k code, +0.6k mem)
//#define USE_WPS                                  // Add support for WPS as initial wifi configuration tool (+33k code, 1k mem (5k mem with core v2.4.2+))
//#define USE_SMARTCONFIG                          // Add support for Wifi SmartConfig as initial wifi configuration tool (+23k code, +0.6k mem)

 
```

tlc76, 2018-08-01:
I am preparing the 6.1.1c version then, if issue still persists, I'll disable wifimanager to check if any improvement.
Thanks for the info!

tlc76, 2018-08-01:
hmm, with 6.1.1c minimal issue is not present:

```
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128
Reply from 10.0.0.178: bytes=32 time=8ms TTL=128
Reply from 10.0.0.178: bytes=32 time=4ms TTL=128
Reply from 10.0.0.178: bytes=32 time=4ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=5ms TTL=128
Reply from 10.0.0.178: bytes=32 time=4ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=3ms TTL=128
Reply from 10.0.0.178: bytes=32 time=4ms TTL=128
Reply from 10.0.0.178: bytes=32 time=4ms TTL=128
Reply from 10.0.0.178: bytes=32 time=2ms TTL=128
Reply from 10.0.0.178: bytes=32 time=4ms TTL=128
Reply from 10.0.0.178: bytes=32 time=5ms TTL=128
```

tlc76, 2018-08-01:
Something is really strange here... there are packets with huge response time.
Any idea what might cause this behavior? It seems it is affecting (so far) only my devices...

Thanks

```
Reply from 10.0.0.178: bytes=32 time=8ms TTL=128
Reply from 10.0.0.178: bytes=32 time=9ms TTL=128
Reply from 10.0.0.178: bytes=32 time=12ms TTL=128
Reply from 10.0.0.178: bytes=32 time=8ms TTL=128
Reply from 10.0.0.178: bytes=32 time=1666ms TTL=128
Reply from 10.0.0.178: bytes=32 time=8ms TTL=128
Reply from 10.0.0.178: bytes=32 time=298ms TTL=128
Reply from 10.0.0.178: bytes=32 time=15ms TTL=128
Reply from 10.0.0.178: bytes=32 time=13ms TTL=128
Reply from 10.0.0.178: bytes=32 time=11ms TTL=128
Reply from 10.0.0.178: bytes=32 time=9ms TTL=128
Reply from 10.0.0.178: bytes=32 time=8ms TTL=128
```

wasi54, 2018-08-01:
Did you also tried bold settings on platformio ? 
(Debian 4.9.88-1+deb9u1 (2018-05-07) x86_64 GNU/Linux with python 2.7.13)

; *** Esp8266 core for Arduino version 2.4.1
**platform = espressif8266@1.7.3**
framework = arduino
board = esp01_1m
board_build.flash_mode = dout

build_flags =
  -Wl,-Tesp8266.flash.1m0.ld
  -DVTABLES_IN_FLASH
  **-DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH**

pio update
pio upgrade
pio run
------------------------
PlatformIO 3.5.4 is currently the newest version available.
Checking tool-scons                      @ 2.20501.4      [Up-to-date]

Platform Manager
================
Platform Espressif 8266
--------
Checking espressif8266                   @ 1.7.3          [Up-to-date]
Checking tool-mkspiffs                   @ 1.200.0        [Up-to-date]
Checking tool-esptool                    @ 1.413.0        [Up-to-date]
Checking tool-espotapy                   @ 1.1.2          [Up-to-date]
Checking framework-arduinoespressif8266  @ 1.20401.2      [Up-to-date]
Checking toolchain-xtensa                @ 1.40802.0      [Up-to-date]

---------------------------------

64 bytes from 172.16.0.130: icmp_seq=4 ttl=128 time=1.39 ms
64 bytes from 172.16.0.130: icmp_seq=5 ttl=128 time=1.29 ms
64 bytes from 172.16.0.130: icmp_seq=6 ttl=128 time=1.49 ms
64 bytes from 172.16.0.130: icmp_seq=7 ttl=128 time=1.22 ms
64 bytes from 172.16.0.130: icmp_seq=8 ttl=128 time=1.44 ms
64 bytes from 172.16.0.130: icmp_seq=9 ttl=128 time=1.22 ms
64 bytes from 172.16.0.130: icmp_seq=10 ttl=128 time=1.21 ms
64 bytes from 172.16.0.130: icmp_seq=11 ttl=128 time=1.04 ms
64 bytes from 172.16.0.130: icmp_seq=12 ttl=128 time=1.27 ms
64 bytes from 172.16.0.130: icmp_seq=13 ttl=128 time=1.34 ms
64 bytes from 172.16.0.130: icmp_seq=14 ttl=128 time=1.19 ms
64 bytes from 172.16.0.130: icmp_seq=15 ttl=128 time=1.23 ms
64 bytes from 172.16.0.130: icmp_seq=16 ttl=128 time=1.25 ms
64 bytes from 172.16.0.130: icmp_seq=17 ttl=128 time=1.80 ms
64 bytes from 172.16.0.130: icmp_seq=18 ttl=128 time=1.21 ms
64 bytes from 172.16.0.130: icmp_seq=19 ttl=128 time=1.15 ms
64 bytes from 172.16.0.130: icmp_seq=20 ttl=128 time=1.13 ms
64 bytes from 172.16.0.130: icmp_seq=21 ttl=128 time=1.34 ms
64 bytes from 172.16.0.130: icmp_seq=22 ttl=128 time=1.28 ms
64 bytes from 172.16.0.130: icmp_seq=23 ttl=128 time=1.21 ms
64 bytes from 172.16.0.130: icmp_seq=24 ttl=128 time=1.51 ms
64 bytes from 172.16.0.130: icmp_seq=25 ttl=128 time=1.27 ms
64 bytes from 172.16.0.130: icmp_seq=26 ttl=128 time=3.09 ms
64 bytes from 172.16.0.130: icmp_seq=27 ttl=128 time=1.36 ms
64 bytes from 172.16.0.130: icmp_seq=28 ttl=128 time=1.32 ms
64 bytes from 172.16.0.130: icmp_seq=29 ttl=128 time=1.39 ms
64 bytes from 172.16.0.130: icmp_seq=30 ttl=128 time=1.35 ms
64 bytes from 172.16.0.130: icmp_seq=31 ttl=128 time=1.38 ms

Jason2866, 2018-08-01:
@wasi54 
2.4.1 generates often problems ....
2.3.0 is (until now) best choice for Tasmota

The strange behaviour from @tlc76 is not from Arduino core. Two reasons:
- it is working on all my devices (i have many ;-))
- his minimal version 6.1.1c has perfect ping responses

so my conclusion a combination of different settings causeses the issue.
Since 5.13x to 6.1.x there are many changes and new options!

Jason2866, 2018-08-01:
The platform where compiled could be a issue. Theo had problems with wrong generated code under Linux.
He is compiling now under Windows. See:
https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips

wasi54, 2018-08-01:
@Jason2866 OK, but i don't have Windows, sorry!
[image]

tlc76, 2018-08-01:
@wasi54 : I am using the default settings from the platformio.ini config file, which is delivered within the Tasmota source code (except esp core which is 2.3.0):

```
[common] ; ************************************************************
; *** Esp8266 core for Arduino version 2.3.0
**platform = espressif8266@1.5.0**
; *** Esp8266 core for Arduino version 2.4.0
;platform = espressif8266@1.6.0
; *** Esp8266 core for Arduino version 2.4.1
;platform = espressif8266@1.7.3
; *** Esp8266 core for Arduino version latest beta
;platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
; *** Esp8266 core for Arduino current version
;platform = espressif8266

framework = arduino
board = esp01_1m
board_build.flash_mode = dout

; *** Fix espressif8266@1.7.0 induced undesired all warnings
build_unflags = -Wall

build_flags =
  -Wl,-Tesp8266.flash.1m0.ld
  -DVTABLES_IN_FLASH
  -DPIO_FRAMEWORK_ARDUINO_LWIP_HIGHER_BANDWIDTH 
```

Could this behavior appear due to this setting?
```
build_unflags = -Wall
```

@Jason2866 : both 5.13.1 and 6.1.1 versions were compiled under windows 10:
- v5.13.1 under Arduino IDE (very difficult to manage)
- v6.1.1 (and 6.1.1c) under Visual Studio Code w/ PlatformIO  extension.

The issue DOES NOT APPEAR when using the minimal version (tested both 6.1.1 and 6.1.1c) and core 2.3.0!

tlc76, 2018-08-01:
PROBLEM SOLVED!

Followed @Jason2866 's advice and disabled wifimanager by uncommenting these lines:
```
// -- Wifi Config tools ---------------------------
#define USE_WPS                                  // Add support for WPS as initial wifi configuration tool (+33k code, 1k mem (5k mem with core v2.4.2+))
#define USE_SMARTCONFIG                          // Add support for Wifi SmartConfig as initial wifi configuration tool (+23k code, +0.6k mem)
```

Ping is now around 4ms average:
```
64 bytes from 10.0.0.178: seq=63 ttl=128 time=2.799 ms
64 bytes from 10.0.0.178: seq=64 ttl=128 time=3.403 ms
64 bytes from 10.0.0.178: seq=65 ttl=128 time=1.760 ms
64 bytes from 10.0.0.178: seq=66 ttl=128 time=5.647 ms
64 bytes from 10.0.0.178: seq=67 ttl=128 time=6.625 ms
64 bytes from 10.0.0.178: seq=68 ttl=128 time=3.922 ms
64 bytes from 10.0.0.178: seq=69 ttl=128 time=1.680 ms
64 bytes from 10.0.0.178: seq=70 ttl=128 time=2.235 ms
64 bytes from 10.0.0.178: seq=71 ttl=128 time=3.321 ms
64 bytes from 10.0.0.178: seq=72 ttl=128 time=4.712 ms
64 bytes from 10.0.0.178: seq=73 ttl=128 time=5.590 ms
64 bytes from 10.0.0.178: seq=74 ttl=128 time=5.274 ms
64 bytes from 10.0.0.178: seq=75 ttl=128 time=9.628 ms
64 bytes from 10.0.0.178: seq=76 ttl=128 time=5.709 ms
64 bytes from 10.0.0.178: seq=77 ttl=128 time=2.488 ms
64 bytes from 10.0.0.178: seq=78 ttl=128 time=2.158 ms
64 bytes from 10.0.0.178: seq=79 ttl=128 time=2.712 ms
64 bytes from 10.0.0.178: seq=80 ttl=128 time=3.071 ms
64 bytes fr om 10.0.0.178: seq=81 ttl=128 time=4.233 ms
64 bytes from 10.0.0.178: seq=82 ttl=128 time=3.695 ms
64 bytes from 10.0.0.178: seq=83 ttl=128 time=6.169 ms
64 bytes from 10.0.0.178: seq=84 ttl=128 time=2.458 ms
^C
--- sonoff1 ping statistics ---
85 packets transmitted, 85 packets received, 0% packet loss
round-trip min/avg/max = 1.680/4.296/20.103 ms
```

Thank you for your support!
Cristian

ascillato, 2018-08-01:
Hi,

If your problem is solved, please close your issue. Thanks

tlc76, 2018-08-01:
Gladly.
FYI I was still monitoring the device for odd behavior - none so far

closed 2018-08-01 by tlc76

brainnex, 2018-08-01:
hello, i don't think this issue is really solved. i have 2 sonoffs in same room. one with 5.10.0 and one with 6.1.1 precompiled bin  from Update 20180731 and 2 I2C sensors. The pings to the 6.1.1 are ways higher with lost messages and the web interface will even not load. The one with 5.10 is working fine and very responsive. I thing here is a bug more to find and many users have problems here.