arendst/Tasmota #3390: Working watermeter based on Tasmota

kajmaj, 2018-08-01:
If anybody is interested.

I use water meter on Wemos. It is based on following 2 rules:
`Rule1 on COUNTER#C1>4500 do backlog Add1 10; publish stat/wemos_1/counter1 %var1%; mem1 %var1%; counter1 0 endon`  
`Rule2 on system#boot do backlog var1 %mem1%; add1 10 endon`

Because value in `var1` does not survive reboot, powercut etc., `mem` is used for persistence.
To continue after reboot it requires var1 "refresh" - rule2

My sensor gives 4500 pulses per 10 liters, so it is possible to configure anyhow :)
It might be not the best, but it works well for me

Frogmore42, 2018-08-01:
Hmm, I have a tipping bucket rain gauge that I have hooked up to a gpio and just use a standard counter.  I do the math at a higher level, using NodeRED.

Not sure how mem works, but if it actually writes to flash every time, you will wear out your flash pretty quickly.

kajmaj, 2018-08-01:
Hmm, good remark. I was not thinking about it in this way. I might to decrease number of writing by volume (like 100 liters, or cubic meter)
Just quick look to wiki:
...flash drives allow reading, writing, and erasing of data, with some allowing 1 million write/erase cycles in each cell of memory: if 100 uses per day, 1 million cycles could span 10,000 days or over 27 years. Some devices level the usage by auto-shifting activity to underused sections of memory."

Nonetheless you are right, will work on "off sonoff" math in the future.

roblad, 2018-08-03:
Hi,

See that COUNTERDEVIDER 

https://github.com/stefanbode/Sonoff-Tasmota

I am using it 

https://github.com/roblad/Sonoff-Tasmota

Svellem, 2018-08-17:
@kajmaj 
What type of sensor do you use as water sensor for the Wemos?

I was hoping to use photoresistive and line sensor with two Wemos D1 minis. The first to read elecrical usage off a blinking electical meter and the latter to track water usage from a "spinning disk" water meter. I can not get either to report any values as "counter" sensors on pin D0, therefore I am curious as to how others have done watermeter tracking with Tasmota and Wemos.

kajmaj, 2018-08-18:
@Svellem 
I use magnetic hall flow sensor. To be more specific :[https://www.aliexpress.com/item/B10-DN25-G1-1inch-water-diesel-Brass-Hall-effect-Turbine-flow-sensor-meter-Turbine-flowmeter-2/32869938988.html?spm=a2g0s.9042311.0.0.27424c4d9xJMA9](url)
So far it works quite well.

ascillato2, 2018-08-22:
Hi,

Thanks a lot for share.

closed 2018-08-22 by ascillato2