arendst/Tasmota #3392: ntpserver2 is locked to nl.pool.ntp.org

jcheger, 2018-08-02:
**ntpserver2 cannot be changed**, it is locked to nl.pool.ntp.org

NtpServer(1) and NtpServer3 let change the settings, but not NtpServer2 

- [X] Searched the problem in issues and in the wiki
- [X] Hardware used : Sonoff POW
- FW versions: 6.1.0, 6.1.1
```
ntpserver1 0.ch.pool.ntp.org
.../stat/RESULT = {"NtpServer1":"0.ch.pool.ntp.org"}
ntpserver2 1.ch.pool.ntp.org
.../stat/RESULT = {"NtpServer2":"nl.pool.ntp.org"}
ntpserver3 2.ch.pool.ntp.org
.../stat/RESULT = {"NtpServer3":"2.ch.pool.ntp.org"}```
```

**To Reproduce**
```
 ntpserver2 1.ch.pool.ntp.org
```

arendst, 2018-08-02:
Ah, that's not a bug but a feature....

If the ntp server starts with a "1" the software detects it as using the default ntp server from user_config.h

I guess a valid ntp server in Switzerland is also "ch.pool.ntp.org" so without the "1".

jcheger, 2018-08-02:
Ok, you are right. Without starting with « 1 », the server is set.
```
npserver2 ch.pool.ntp.org
.../stat/RESULT = {"NtpServer2":"ch.pool.ntp.org"}
```
Thanks for your answer

closed 2018-08-02 by jcheger

arendst, 2018-08-27:
v6.1.1.12 allows the use of a 1 as start of server name.

jcheger, 2018-08-27:
oh, nice ! Thank you.