arendst/Tasmota #3395: iFan02 RF remote stopped working

ExplodingLemur, 2018-08-02:
My iFan02 stopped responding to its RF remote.  I've tried power-cycling the fan controller, I've tried holding the app bind button on the remote to get it to re-bind, but it's just not responding to it.  Has anyone else experienced this?  Is there a fix or did I just get a bad unit?  (It did work fine for about a week...)

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [X] Hardware used : iFan02
- [ ] Provide the output of command ``status 0`` :
```
[will provide this later today]
```

**(Please, remember to close the issue when the problem has been addressed)**

firzanarmani, 2018-08-05:
it's happened to me too. it only lasted less than a day. i didnt flash tasmota on it at first, was using ewelink then it just stopped responding to the remote. i suspect that it's a hardware problem but i don't really know how to go about it

ExplodingLemur, 2018-08-05:
I've opened a ticket with iTead as well but they don't seem to answer any of their support tickets :/

ExplodingLemur, 2018-08-14:
iTead's suggestion is to pop open the case of the RF remote, hold the battery in place and press and hold the button on the back of the remote PCB while power-cycling the fan controller.  I've not gotten that to work yet though.  I plan on replacing the capacitors in my unit to match the values from my old speed control so I'll touch up the RF RX module's solder joints while I'm in there.

ExplodingLemur, 2018-08-20:
I reflowed all the pins for the little RF daughterboard and now it's working again.  I guess this part had poor solder joints.

closed 2018-08-20 by ExplodingLemur

firzanarmani, 2018-08-31:
@ExplodingLemur how did you reflow the pins of the RF daugtherboard?