arendst/Tasmota #3396: Using an ESP-01 for multiple reed-switches, but no relay

mattpurnell, 2018-08-03:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used :  ESP-01 (black)
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT - Note that this is a little different to the a scenario descibed - I want to get this one right before I modify the main unit I rely on:

18:39:21 MQT: stat/garagedoor/STATUS = {"Status":{"Module":18,"FriendlyName":["garagedoor"],"Topic":"garagedoor","ButtonTopic":"doorbell","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
18:39:21 MQT: stat/garagedoor/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://dsm.lcl:8008/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:16:36","StartupUTC":"2018-08-03T06:22:45","Sleep":0,"BootCount":29,"SaveCount":190,"SaveAddress":"F6000"}}
18:39:21 MQT: stat/garagedoor/STATUS2 = {"StatusFWR":{"Version":"6.1.1c","BuildDateTime":"2018-08-02T23:35:07","Boot":5,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
18:39:21 MQT: stat/garagedoor/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":3,"LogHost":"dsm.lcl","LogPort":514,"SSId":["Geogaddi_5Ghz","Geogaddi"],"TelePeriod":300,"SetOption":["040A8009","55A18000","00000000"]}}
18:39:21 MQT: stat/garagedoor/STATUS4 = {"StatusMEM":{"ProgramSize":447,"Free":556,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","08C02794","00000000","35B6179E","00000000"]}}
18:39:21 MQT: stat/garagedoor/STATUS5 = {"StatusNET":{"Hostname":"garagedoor-1868","IPAddress":"192.168.2.211","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"DC:4F:22:0A:07:4C","Webserver":2,"WifiConfig":1}}
18:39:21 MQT: stat/garagedoor/STATUS6 = {"StatusMQT":{"MqttHost":"dsm.lcl","MqttPort":1883,"MqttClientMask":"garagedoor","MqttClient":"garagedoor","MqttUser":"garagedoor","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:39:21 MQT: stat/garagedoor/STATUS7 = {"StatusTIM":{"UTC":"Fri Aug 03 06:39:21 2018","Local":"Fri Aug 03 18:39:21 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":12}}
18:39:21 MQT: stat/garagedoor/STATUS10 = {"StatusSNS":{"Time":"2018-08-03T18:39:21","Switch1":"ON","Switch2":"ON"}}
18:39:21 MQT: stat/garagedoor/STATUS11 = {"StatusSTS":{"Time":"2018-08-03T18:39:21","Uptime":"0T00:16:36","Vcc":3.410,"Wifi":{"AP":2,"SSId":"Geogaddi","RSSI":100,"APMac":"00:11:32:45:90:1F"}}}
```
Hi,

I think my question is not a totally uncommon one; but I cannot get my head around the other threads I find that touch on the subject. Either that, or something isn't working 🤔 

--- Basically, I want two switches (with no relays) to send non-identical MQTT messages.. e.g. with identifiers like "POWER1" and POWER2", which is apparently complicated when there are no Relays present?

I have an ESP-01 flashed with Tasmota and set to emulate the Wemo (Generic). I already successfully have GPIO1 (TX) used for a temperature sensor, and GPIO3 (RX) is used for a reed switch already. So that leaves GPIO0 and GPIO2 to add an extra Doorbell switch. 

The Doorbell will be exactly the same as the reed switch, but a push-button rather than magnetic. The only problem I have is that any switches give the same "POWER" identifier.

What command is required, to get each switch (Reed on GPIO3, Doorbell on GPIO0/2) to send non-identical MQTT messages? e.g. POWER1 and POWER2?

-----
PT 2 Bonus question 😄 
This page confuses me: https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic

In my understanding, at least the below doesn't work - If I set cmnd/doorbell/SwitchTopic  with a payload of 2, upon triggering the switches I get MQTT messages like cmnd/2/POWER. So the 'SwitchTopic' command should be described as naming the custom topic?

> SwitchTopic=0 controls the relay directly.
> SwitchTopic=1 sends a message with the default topic to MQTT. This message will get picked up by the device itself and sets the state of the relay accordingly.
> While SwitchTopic=2 sends a command with a custom topic via MQTT.

Jason2866, 2018-08-03:
Read section commands in wiki
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#mqtt-specific

It works as described.

Jason2866, 2018-08-03:
You can get POWER1 and POWER2 on off messages without using custom topic.
Just define virtual (non existing!) relais. Do this on GPIO pins which are not routed outside.
For example GPIO 12, 13, 14, 4, 5
Now you get for every switch with a virtual connected relais an On Off message.

mattpurnell, 2018-08-04:
@Jason2866  Thanks mate that was the bit I had not absorbed properly..! Configure several dummy relays - done 😄

closed 2018-08-04 by mattpurnell