arendst/Tasmota #3397: Is there a way to download test build?

witek1308, 2018-08-03:
Hi
I am interested in features of 6.1.1c which is not released yet officialy.
Is there a repository to download test builds to avoid building myself?

andrethomas, 2018-08-03:
Currently not to my knowledge although we will soon have a web-based builder that may be used to test code which is in the current development branch.

andrethomas, 2018-08-03:
If you're willing to take a step on the wild side and test the current build outputs you may check them from http://thehackbox.org/Sonoff-Tasmota/build/

Currently only builds:
sonoff
sonoff-minimal
sonoff-classic
sonoff-knx
sonoff-allsensors

But a web interface is in the works to enable/disable sensors etc for custom builds.

witek1308, 2018-08-04:
Thanks. Definitely will give it a try.

witek1308, 2018-08-07:
As I don't understand deep dive OTA mechanism/eligibility, is this a known thing the test build can't go OTA?
I'm getting buffer miscompare for the existing 6.1.1. I can try to flash it via cable but where I want to test this device it is for already mounted scenario in a very incovenient place and unmounting will be a hassle. OTA would be perfect thing.

andrethomas, 2018-08-07:
Flashing a test binary to a unit that is in an inconvenient location is not advised... the current codebase should be released by arendst in a week or so and be available from the usual locations... I would wait for that on if testing the current codebase on a device that is in production/running something important.

andrethomas, 2018-08-07:
Short answer - flash only to a test device using a cable :)

ascillato, 2018-08-29:
Hi,

This information has been added to the readme file. 

Thanks

closed 2018-08-29 by ascillato2