arendst/Tasmota #3400: Arduino Core 2.4.2 Released

Courty40, 2018-08-03:
I saw Core 2.4.2 Released today on Arduino, any thoughts on if it will fix some of the issues we've been seeing ?

Courty

Jason2866, 2018-08-03:
I tried it (my first build from Tasmota on Arduino IDE, i know why using VSC!!!)
Seems good at first look, boots up ;-)

[image]

akpuggy, 2018-08-03:
Can you please share how you were able to test since latest espressif8266 package is still 1.7.3? I only have 1.73 and 1.50 platforms installed in Atom, so not sure what to use in the platform.io file (platform = espressif8266@?).

Jason2866, 2018-08-03:
On Platformio still not available... 
That was the reason i setup Arduino for Tasmota. And i know why i hate it Arduino and changed!

andrethomas, 2018-08-03:
Works for me too - Can't help but wonder what the extra ~18k of flash will give us!

andrethomas, 2018-08-03:
@Jason2866 
[image]

You do know you can use a 4M linker script, right?

[image]

Jason2866, 2018-08-03:
@andrethomas 
Yes i know, was build quick and dirty on Arduino (did i say i hate it?).
I have the 4M definition in platformio.ini....

andrethomas, 2018-08-03:
@Jason2866 Gotcha... was just checking :)

Jason2866, 2018-08-03:
HueEmulation doesnt work :-(

andrethomas, 2018-08-06:
I also find build time to be excessive compared to 2.4.1

Frogmore42, 2018-08-13:
Looks like PlatformIO just released an update that includes this core:
https://github.com/platformio/platform-espressif8266/releases/tag/v1.8.0

akpuggy, 2018-08-14:
Sweet! Can now try it out next week... haulted all sonoff projects waiting for this.

Frogmore42, 2018-08-14:
Yeah, me too.

Jason2866, 2018-08-14:
Compiles correct under Atom / Platformio and VSC / Platformio.
WemosMini boots up and behaviour seems normal.

[image]

Jason2866, 2018-08-14:
Works on Sonoff Basic

[image]

arendst, 2018-08-14:
Stage is not 2.4.2 but the next beta version. I cannot get platformio to compile 1.8.0 as it complains about SCON python problems. Probably platformio problem but I won't reinstall platformio just now. I'll wait for someone to succeed ;-)

andrethomas, 2018-08-14:
For me it compiles under Linux CLI but I just flashed it to a device and it reboots turning the relays on momentarily and then reboots again... damn... no i have to open something LOL

Jason2866, 2018-08-14:
@arendst 
I have in platformio.ini
platform = espressif8266@1.8.0
I think the havent changend the name in release 1.8.0 (cfd48f3 is the same as in arduino release 2.4.2!)

andrethomas, 2018-08-14:
My linux cli version still has this

```
;platform = espressif8266@1.7.3
; *** Esp8266 core for Arduino version latest beta
;platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
; *** Esp8266 core for Arduino current version
platform = espressif8266

```

andrethomas, 2018-08-14:
Hmm

```
tasmota@tasmota:~/tasmota/development$ platformio platform update
Platform Espressif 8266
--------
Updating espressif8266                   @ 1.8.0          [Up-to-date]
Updating framework-arduinoespressif8266  @ 2.20402.0      [2.20402.1]
Uninstalling framework-arduinoespressif8266 @ 2.20402.0:        [OK]
PackageManager: Installing framework-arduinoespressif8266 @ 2.20402.1
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
framework-arduinoespressif8266 @ 2.20402.1 has been successfully installed!
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]

Platform Espressif 8266
--------
Updating espressif8266                   @ 1.5.0          [Fixed]
Updating framework-arduinoespressif8266  @ 1.20300.1      [Up-to-date]
Updating tool-esptool                    @ 1.409.0        [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]

Platform Espressif 8266
--------
Updating espressif8266                   @ 1.6.0          [Fixed]
Updating framework-arduinoespressif8266  @ 1.20400.3      [Up-to-date]
Updating tool-esptool                    @ 1.409.0        [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]

```

andrethomas, 2018-08-14:
[image]

Jason2866, 2018-08-14:
@andrethomas 
Looks good 👍

Courty40, 2018-08-14:
I’ve had a couple of devices (the older ones) that have needed the flash/eeprom cleared and to start again before the are stable. May be related to some of the issues here ?

Jason2866, 2018-08-14:
@Courty40 
Everyone (here in this thread) should the devices now get running with 1.8.0 aka 2.4.2 :-)
Using Platformio (Atom) with entry in platformio.ini
platform = espressif8266@1.8.0
and a `pio update`

andrethomas, 2018-08-14:
@Jason2866 I'm tentative on jumping to conclusions on devices that are hard to get to :)

Jason2866, 2018-08-14:
@andrethomas 
I did not say always the easy way :-) -> Dont fix aint broken

andrethomas, 2018-08-14:
@Jason2866 Now you're just reminding me of a new way of comparative statement methods I learnt here which never occurred to me before

I always did
`if (variable == known)`
Here I learn
`if (known == variable)`

LOL

arendst, 2018-08-15:
Well at least platformio now has the correct core 2.4.2 version when 1.8.0 is selected. See https://community.platformio.org/t/espressif-32-8266-new-boards-esp-prog-bug-fixes-for-esp-idf-and-arduino-frameworks/4766

Now someone has to find out if all problems with previous versions have been solved...

I know the Hue fix is not in the 2.4.2 webserver code so that still needs to be patched locally.

arendst, 2018-08-15:
Do you experience quick webpage updates like with previous versions? I tried two wemos with 2.4.2 and they respond very slow; the answer is shown on serial interface at least three seconds before the webpage console is updated...

wasi54, 2018-08-15:
Webpage update takes less than 0.1 sec. for Power on/off via mqtt and Wifi is stable and fast since 6.1.1c with STAGE!
I am using on Basic,Dual V2 and TH:
-DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH
 
[image]

[image]

Jason2866, 2018-08-15:
@arendst 
I have the feeling that sometimes with 2.4.2 the webconsole reacts some milliseconds later than with 2.3.0
But it isnt laggy in any way! If i do the same thing again the response is faster as it is with 2.3.0.
For me is 2.4.2 is working fine so far. I will measure if sleep is working. If yes i will update some more devices (which are easily reachable)...

arendst, 2018-08-15:
Just did a recompile and now it's working fine....

andrethomas, 2018-08-15:
Wish I could recompile myself... perhaps I will also work fine :)

Jason2866, 2018-08-15:
@andrethomas 
human code has self correction build in. No recompile neccersary;-)

andrethomas, 2018-08-15:
@Jason2866 Not having to reboot for a few days and still work properly would be nice though :-)

Jason2866, 2018-08-15:
@andrethomas Total off topic anyway. At my last reboot i lost some memory😎

andrethomas, 2018-08-15:
@Jason2866 Yeah, make us a channel on Discord so we can goof off there.

I've uploaded the new code to the sonoff dual that controls my overhead lights in my lab... easy to get to in case it fails.

Jason2866, 2018-08-15:
@andrethomas 
Just fired up Discord (first time): https://discord.gg/Ks2Kzd4

andrethomas, 2018-08-16:
No memory leak so far...

ascillato, 2018-08-25:
Hi,

Up to now, in my devices there was no freeze nor exception with v2.4.2 with lwIP 1.4 (at 80Mhz) since I flash them. It also seems that the webserver works faster. :smile: 

The only downside of v2.4.2 is that `sleep` will freeze the device. :disappointed: 

So, if you are going to use `SLEEP`, you need to use v2.3.0. If not, use v2.4.2 lwIP 1.4

In my tests, v2.4.2 with lwIP 2 led to reboots with exceptions.

ascillato, 2018-08-25:
I like the Discord chat you have opened. What about making that the official chat for Tasmota? We can add that to the readme / wiki.

Jason2866, 2018-08-25:
Hi @ascillato 
why not?! Let Us try. Maybe we can move  the non issue stuff out of here. 
A another benefit users can get a direct response there.

ascillato2, 2018-08-25:
Hi,

Closing this issue as esplib v2.4.2 is working fine except from sleep feature.

Please, ask to reopen if needed. Thanks all for the tests and comments. :+1:

closed 2018-08-25 by ascillato2