arendst/Tasmota #3401: PRODINo WiFi-ESP support

k4roshi, 2018-08-03:
Hi everyone, i'm in the process of making tasmota run on this hardware:
https://www.kmpelectronics.eu/en-us/products/prodinowifi-esp.aspx
I think it's an interesting platform, with 4 optoisolated inputs and 4 relay outputs on a wroom-2 based board with 2mb of flash. That makes it quite easy to use with existing light switches without having to solder anything.
The only weird thing on this board is that it uses an MCP23S08 through SPI for controlling both the 4 inputs and the 4 outputs. The good thing is that that leaves many of the gpio pins free to add other sensors. 
I'd like to do a clean port, create a driver and create a pull request if that's ok. I see that the MCP23008 is supported, but only as an input and through I2C, so using that same driver is not feasible.
Would this platform be interesting for you? Is it a problem that it uses an external chip to provide its outputs? Any suggestions on how to proceed?

andrethomas, 2018-08-03:
I see that the MCP23008 is supported, but only as an input and through I2C

The MCP23008/MCP23017 is supported over I2C in output mode also, but output mode is not committed to development.

The test code for using MCP230xx in output mode can be found here: https://github.com/andrethomas/Sonoff-Tasmota/tree/mcp230xx_output

You can follow the trailing end of issue https://github.com/arendst/Sonoff-Tasmota/issues/3350 of the original input code here for updates on the OUTPUT support but I am not sure when or if it will be implemented in development branch - perhaps if there's sufficient call for it - currently it requires an additional define USE_MCP230xx_OUTPUT in addition to the xsns_29_mcp230xx.ino file from the above repo to make outputs work.

andrethomas, 2018-08-03:
Perhaps to note, I think the major challenge in adding it as a newly supported device would be that the current code model does not make provision for the inputs/outputs of the board you mention to be listed in the conventional configuration web page such as these

[image]

The MCP230xx driver does not have a web configuration and is not available for configuration in the above fashion as normal gpio pins on the ESP8266. It does not have a web config page of its own due to RAM constraints of the ESP8266.

It uses a different method of interaction through a sensor29 command as documented in its Wiki here https://github.com/arendst/Sonoff-Tasmota/wiki/MCP23008-MCP23017

The only addition the OUTPUT code provides is
```
sensor29 pin,5,0  // Set pin as output and default to LOW on power-on
sensor29 pin,5,1  // Set pin as output and default to HIGH on power-on
```

If you have setoption0 1 (#define SAVE_STATE in user_config.h) then the above is ignored and the last known state is assumed during reset/power-up

Then to turn on and off
```
sensor29 pin,ON   // Turn pin ON (HIGH)
sensor29 pin,OFF  // Turn pin OFF (LOW)
sensor29 pin,T    // Toggle current state of pin
```

jsponz, 2018-08-05:
@k4roshi I am also interested on this device. How is your status?

ascillato2, 2018-09-03:
Hi,

As there is no feedback, I'm closing this issue. Please, ask to reopen if you need more help. Thanks

closed 2018-09-03 by ascillato2

k4roshi, 2018-10-17:
Hi everyone, i had to move house, thus had no access to my tools or time to tackle this. 
Now i'm back and would really like to use this board in my new smart home system.
What i'd like to do, if feasible and desirable, is add the MCP config to the main module parameters configuration page (the one attached earlier on). The sensor29 console interface is definitely usable (especially through MQTT) but to fully exploit the potential of the the added inputs and outputs i think they should be accessible through the web interface. The added inputs and outputs would then be treated as first class citizens, just as the internal ones. For instance relays attached to the MCP would show up and be triggered even in the home page.
I don't expect it to be particularly easy but, before starting this endeavour, i'd like to know whether there are unsurmountable limits of the platform that i should know about or if this kind of pull request would not be accepted because of the changes it would involve. For instance, would RAM still be a problem if we reuse the main board configuration interface?

andrethomas, 2018-10-18:
The MCP230xx expansion was initially for inputs only and output functionality was added later.

We did initially have a web ui config page for the MCP230xx but it just took too much ram and additional flash space to maintain this - in some cases users reported resets due to the fact that there was not enough ram.

For this reason web ui configuration of the MCP230xx was removed in favor of the sensor29 command as it has a lower demand for flash space and ram usage during running time.