arendst/Tasmota #3402: PowerOnstate ignores LedPower and LedState

pascalsaul, 2018-08-03:
**Describe the bug**
When the relay stays always it is ignoring LedState and PowerState 0.

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!
- [ ] Searched the problem in issues and in the wiki
https://github.com/arendst/Sonoff-Tasmota/issues/699
- [ ] Hardware used : 
Sonoff Touch as ButtonTopic
- [ ] Provide the output of command ``status 0`` :
```
tele/touch_zolder_onder/LWT Online
ha/switch/touch_zolder_onder_1/config {"name":"touch_zolder_onder","command_topic":"cmnd/touch_zolder_onder/POWER","state_topic":"stat/touch_zolder_onder/RESULT","value_template":"{{value_json.POWER}}","payload_off":"0","payload_on":"1","availability_topic":"tele/touch_zolder_onder/LWT","payload_available":"Online","payload_not_available":"Offline"}
ha/binary_sensor/touch_zolder_boven_1/config {"name":"touch_zolder_onder BTN","state_topic":"cmnd/touch_zolder_boven/POWER","payload_on":"TOGGLE","availability_topic":"tele/touch_zolder_onder/LWT","payload_available":"Online","payload_not_available":"Offline"}
cmnd/touch_zolder_onder/status 0
stat/touch_zolder_onder/STATUS {"Status":{"Module":10,"FriendlyName":["touch_zolder_onder"],"Topic":"touch_zolder_onder","ButtonTopic":"touch_zolder_boven","Power":1,"PowerOnState":4,"LedState":0,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
stat/touch_zolder_onder/STATUS1 {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.12/sonoff.bin","RestartReason":"Software/System restart","Uptime":"10T22:13:21","StartupUTC":"2018-07-23T21:36:45","Sleep":100,"BootCount":46,"SaveCount":182,"SaveAddress":"F9000"}}
stat/touch_zolder_onder/STATUS2 {"StatusFWR":{"Version":"6.1.1c","BuildDateTime":"2018-07-22T14:44:40","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
stat/touch_zolder_onder/STATUS3 {"StatusLOG":{"SerialLog":0,"WebLog":0,"SysLog":0,"LogHost":"192.168.1.17","LogPort":514,"SSId":["XXX","XXX"],"TelePeriod":30,"SetOption":["000A0009","55818000"]}}
stat/touch_zolder_onder/STATUS4 {"StatusMEM":{"ProgramSize":394,"Free":608,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","00400684","0C000000","00003CC4","00000000"]}}
stat/touch_zolder_onder/STATUS5 {"StatusNET":{"Hostname":"touch_zolder_onder","IPAddress":"192.168.1.60","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"60:01:94:84:4C:3C","Webserver":2,"WifiConfig":4}}
stat/touch_zolder_onder/STATUS6 {"StatusMQT":{"MqttHost":"192.168.1.18","MqttPort":1883,"MqttClientMask":"touch_zolder_onder","MqttClient":"touch_zolder_onder","MqttUser":"mqtt","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
stat/touch_zolder_onder/STATUS7 {"StatusTIM":{"UTC":"Fri Aug 03 19:50:07 2018","Local":"Fri Aug 03 21:50:07 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99}}
stat/touch_zolder_onder/STATUS10 {"StatusSNS":{"Time":"2018-08-03T21:50:07"}}
stat/touch_zolder_onder/STATUS11 {"StatusSTS":{"Time":"2018-08-03T21:50:07","Uptime":"10T22:13:22","POWER":"1","Wifi":{"AP":1,"SSId":"Peppelenwei 65","RSSI":98,"APMac":"F4:EA:67:0F:8A:21"}}}
```

**To Reproduce**
```'cmnd/touch_zolder_onder/backlog' -m 'LedPower 0;LedState 0;'```

**Expected behavior**
The LED stays OFF and the relay is disabled in OFF modus

**Additional context**
I would like to request a PowerOnState | 6 | Turn relay(s) off and disable further relay control to address these issues

ascillato, 2018-08-26:
Hi,

So, for your sonoff touch, you want :

- the relay to be off all the time
- the blue and the red Leds to be off all the time
- And send MQTT messages when touching the sonoff

right?

Yes, you are right because there is no `buttonmode` command as `switchmode` does so as to override relay control and make the red led to be off all the time (the red led is hardwired to the relay, no control over that). Neither a `poweronstate 6`.

I'm adding the label enhancement to address this.

But by now, you can use the following workaround:

Go to Module Configuration in the Tasmota Webmenu and choose **GENERIC**
After reboot, only set **GPIO00** as **switch1**
Go to the console and type:

```
switchtopic 0
switchmode1 1
rule 1
rule on switch1#state=1 do publish cmnd/topic/power 1 endon
```

This should make your sonoff touch behave as you expected.

ascillato, 2018-08-27:
Hi,

So, for your sonoff touch, you want :

* the relay to be off all the time
* the blue and the red Leds to be off all the time
* And send MQTT messages when touching the sonoff

right?

So, adapting [this](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#13-toggle-a-relay-only-when-holding-the-button-for-2-seconds) example of the wiki to your requirements, there is no need for new `poweronstate 6` command.

Please, using sonoff touch as module type, write in the console the following:

```
poweronstate 0
ledpower 0
ledstate 0
buttontopic 0
setoption1 1
rule on button1#state=2 do publish cmnd/topic/power 2 endon
rule 1
```

Hope this is helps.

ascillato2, 2018-08-28:
Hi,

Have you managed to solve your issue?

pascalsaul, 2018-08-29:
Nope, but fix doesn't seem to work. Need to debug it in more details when I'm back from a break.

ascillato2, 2018-09-03:
Hi,

Which part is not working for you? I have tested and works.

ascillato2, 2018-09-04:
Please, share your results when you have time.

Thanks.

closed 2018-09-04 by ascillato2

pascalsaul, 2018-09-17:
The relay is toggling and the LED also. 

The buttontopic isn't working anymore so the other switch is not being toggled.