arendst/Tasmota #3403: How to publish "Toggle" to topic

RaymondMouthaan, 2018-08-03:
Guess,

I've been reading [Understanding-SwitchMode-and-SwitchTopic](https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic), but it's not clear to me.

What i like to do, is to read "high" or "low" from a Wemos digital input and publish "toggle" to a topic, so basically when a button is pressed Tasmota publishes "toggle" to the topic and the same when the button is released.

Hope someone can point me in the right direction ...

Greetz,
Ray

andrethomas, 2018-08-03:
What MQT output is generated when you change the state of your input pin? How did you configure the pin?

andrethomas, 2018-08-04:
There is probably an easier way but I would just create a rule that responds to the Switch1=ON and Switch1=OFF and do publish xxx - Check the rules wiki for examples https://github.com/arendst/Sonoff-Tasmota/wiki/Rules

RaymondMouthaan, 2018-08-04:
Maybe there is an easier way, but for now I am trying rules, but so far no success, because i am facing rules for the first time. Not sure yet how to set-up the correct rule for this.

andrethomas, 2018-08-04:
`cmnd/wemos-1/POWER1`
Actually, the "wemos-1" is configured for a specific device in the MQTT configuration - in my test board below its still in the default of sonoff

[image]

RaymondMouthaan, 2018-08-04:
yeah thats what i know, but why is the toggle send to cmnd/2/POWER1 instead of cmnd/wemos-1/POWER1 ?

andrethomas, 2018-08-04:
Does your full-topic still look like mine above?

andrethomas, 2018-08-04:
You must use SwitchTopic=0, I guess

RaymondMouthaan, 2018-08-04:
Yes it is
[image]

andrethomas, 2018-08-04:
Issue command
switchtopic "wemos-1"

RaymondMouthaan, 2018-08-04:
I've set back SwitchTopic to 0 (which is the default)
[image]

Toggle is not published anymore

andrethomas, 2018-08-04:
Try
switchtopic "wemos-1"
?

andrethomas, 2018-08-04:
Actually without the "

switchtopic wemos-1

Then I get

15:08:55 MQT: cmnd/wemos-1/POWER = TOGGLE

RaymondMouthaan, 2018-08-04:
I just did, and your are the man!!!

That just did it, yeeaaaayyy 

However .. I think there is a bug ...

I tried the command like you said:
`switchtopic "wemos-1"`

which results in:
[image]

But those quotes should be there in the topic so I tried:
`switchtopic wemos-1` (without those quotes)

and its result is:
[image]

andrethomas, 2018-08-04:
You just forced me to learn about switchtopic :)

Well done!

RaymondMouthaan, 2018-08-04:
Hahaah great, but its not working for me without quotes :(

RaymondMouthaan, 2018-08-04:
alright this becomes more strange ... 

`switchtopic wemos-1` fails as you could see in my previous screenshot
`switchtopic wemos-2` succeeded with the toggle as expected

andrethomas, 2018-08-04:
Makes sense to use something other than current default topic

andrethomas, 2018-08-04:
I get the same result if I use the same topic as my board uses... so it has to be different from the default

RaymondMouthaan, 2018-08-04:
Yes, for testing I just changed the default topic wemos-1 from the mqtt config page to wemos-bla and set switchtopic wemos-1 on the command line, and now it works

RaymondMouthaan, 2018-08-04:
I just wonder what if you have 2 or more switches, seems to me all switches share the same topic defined by switchtopic <topic> .. this makes it not possible to distinguish the switch that was pressed ..

but for now I am helped, since I am planning to only attach 1 switch to one Wemos.

@andrethomas thanks for your help!!! 👍

closed 2018-08-04 by RaymondMouthaan

andrethomas, 2018-08-04:
You're welcome.

Just tested multiple switches and it changes the POWERx (x)

```
15:28:17 MQT: cmnd/wemos/POWER1 = TOGGLE
15:28:18 MQT: cmnd/wemos/POWER2 = TOGGLE
```

POWER1 = Switch1
POWER2 = Switch2

RaymondMouthaan, 2018-08-04:
Oooww thats really nice! .. could you post screenshots of your config and the exact commands ?

I didn't manage to get it work like you

andrethomas, 2018-08-04:
Nothing different just configured Switch2 on another pin

[image]

RaymondMouthaan, 2018-08-04:
Aha while you posted your config .. I found out to configure RELAY 1 and 2 as well, this shows up as POWER1 and POWER2 👍

andrethomas, 2018-08-04:
Good stuff, enjoy!

RaymondMouthaan, 2018-08-04:
I will! .. Thanks again and enjoy your day 👍