arendst/Tasmota #3406: How to configure common relay module (non-sonoff)

mattpurnell, 2018-08-04:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : ESP01 with a relay module - with builtin voltage drop
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT (please ignore the failure to connect to MQTT server)

17:57:42 RSL: stat/SPARE/STATUS = {"Status":{"Module":18,"FriendlyName":["SPARE","SPARE2"],"Topic":"SPARE","ButtonTopic":"0","Power":3,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
17:57:42 RSL: stat/SPARE/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://dsm.lcl:8008/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:58","StartupUTC":"2018-08-04T05:56:44","Sleep":0,"BootCount":4,"SaveCount":17,"SaveAddress":"FB000"}}
17:57:42 RSL: stat/SPARE/STATUS2 = {"StatusFWR":{"Version":"6.1.1c","BuildDateTime":"2018-08-04T17:37:08","Boot":5,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
17:57:43 RSL: stat/SPARE/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":3,"LogHost":"dsm.lcl","LogPort":514,"SSId":["Geogaddi_5Ghz","Geogaddi"],"TelePeriod":300,"SetOption":["00088009","55818000","00000000"]}}
17:57:43 RSL: stat/SPARE/STATUS4 = {"StatusMEM":{"ProgramSize":447,"Free":556,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","08C02794","00000000","35B6179E","00000000"]}}
17:57:43 RSL: stat/SPARE/STATUS5 = {"StatusNET":{"Hostname":"SPARE-4496","IPAddress":"192.168.2.212","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"5C:CF:7F:69:11:90","Webserver":2,"WifiConfig":3}}
17:57:43 RSL: stat/SPARE/STATUS6 = {"StatusMQT":{"MqttHost":"dsm.lcl","MqttPort":1883,"MqttClientMask":"SPARE","MqttClient":"SPARE","MqttUser":"SPARE","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:57:43 RSL: stat/SPARE/STATUS7 = {"StatusTIM":{"UTC":"Sat Aug 04 05:57:43 2018","Local":"Sat Aug 04 17:57:43 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":12}}
17:57:43 RSL: stat/SPARE/STATUS10 = {"StatusSNS":{"Time":"2018-08-04T17:57:43"}}
17:57:43 RSL: stat/SPARE/STATUS11 = {"StatusSTS":{"Time":"2018-08-04T17:57:43","Uptime":"0T00:00:59","Vcc":3.510,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":2,"SSId":"Geogaddi","RSSI":90,"APMac":"00:11:32:45:90:1F"}}}

```
Hello team

Firstly - I am a great fan of Tasmota and all you do 👍 - so I feel a bit bad asking for your help here  given that the module is non-Sonoff - 😟 - but I would really like to keep the Tasmota theme for all the ESPs I have.

Now recently I bought a few 5v relays that also handle the voltage drop (to 3.3v). I found using these we are somewhat abstracted from connecting to the relay - so I am only *mostly* sure that the relay is on GPIO2.

Anyway.. I have been totally unable to get the relay to actually switch over at all; I have tried two separate ESP units and 2 different relay modules, leading me to believe there is no hardware fault. I have tried numerous combinations of type & emulation, configured the module to have relays on all pins available on the 'Generic' setting, but have gotten no relay activity at all.

I have been testing for continuity with a multitester, e.g. hooking it up to "Normally Open" and trying to close the relay, and vice-versa. 

I know there will be something missing from my knowledge on this - can anyone please point me in the right direction?

Apologies for the lack of info on this unit:

[image]

mattpurnell, 2018-08-04:
I think this thread is relevant; https://github.com/arendst/Sonoff-Tasmota/issues/1714

I might try to document this requirement thoroughly to help others

mattpurnell, 2018-08-04:
Unfortunately - I think the code has diverged too much since this pull request: https://github.com/arendst/Sonoff-Tasmota/pull/1282/files

I was able to select the module inside the web console, but the physical relay is not triggered.

[image]

andrethomas, 2018-08-04:
That ESP module looks like a ESP-01 so it only breaks out GPIO0 and GPIO2. If you configure it as a GENERIC and put the relay on either one of those (most likely GPIO2) it should work.

andrethomas, 2018-08-04:
Most likely 
[image]

There is also Relay1i if the relay is wired inverted which is quite likely.

KPK75, 2018-08-04:
The relay is controlled via STM chip using mentioned above serial commands.

Jason2866, 2018-08-04:
Maybe the ESP-01 Relais Module (left on picture) does not work without hardware modification.
I bought one that has got 3! design errors (GPIO0 pulldown resistor, CHPD not pullup!, GPIO2 floated).

mattpurnell, 2018-08-05:
The ESP-01 is definitely fine, I have used it on a standard relay just fine - as @KPK75 points out, these things need a serial signal, not by GPIO pin going high/low.

I have tried using the following serial commands:
` echo -ne "\xA0\x01\x01\xA2" | mosquitto_pub -t "cmnd/GARAGE/serialsend4" -s -u [username] -P [password]`
...but after sending that - the unit just spits out the same MQTT message endlessly, even after a power bounce - so at the moment it looks like I will need a new firmware for the ESP-01:
[image]

Jason2866, 2018-08-05:
@mattpurnell 
Why do you think the module isnt faulty? Did you manage to switch the relais ever (with orig. Software)?
I have a exact looking module that never worked in its original version (via described serial commands from LC Tech Site).  That the ESP-01 Module itself is working i believe (mine was ok too).

mattpurnell, 2018-08-05:
@Jason2866  Ah I thought you meant the ESP might be faulty.

No you are right it's a possibility; while I have two of them, and the seller had no negative feedback for that unit.. but I definitely can't be certain. 

The AP that the ESP put out had an SSID of "AI-THINKER' so I could track down that firmware and see.

pmknowles, 2018-08-19:
I can confirm that https://github.com/arendst/Sonoff-Tasmota/pull/1282/files
works with these but you I think you can only use up to 5.12
[image]
One of the problems with this device is that one of the commands use a null character which is seen as a terminator by a lot of terminal software.
Theo has blocked transmitting the strings via the serial interface so KPK75's version is the only one I know that works.

pmknowles, 2018-08-19:
This is my openHAB item for the LCTech
Switch LRLamp1 "Living Room Lamp 1" <switch> (GroundFloor) {mqtt=">[mosquitto:cmnd/sonoff-LRLamp1/POWER:command:*:default],<[mosquitto:stat/sonoff-LRLamp1/POWER:state:default]"}
The modified bin does the serial send
However https://github.com/arendst/Sonoff-Tasmota/issues/3168 looks like SerialSend4 should work My version 6.11 (31072018) doesn't have SerialSend4 in it. I'll investigate tomorrow

mattpurnell, 2018-08-20:
Thanks! Looks like the breakthrough I needed to use these devices

pmknowles, 2018-08-20:
Compiled the latest release and it works (sort of)
in openHAB the ON command is
publish("mosquitto", "cmnd/sonoff-hob/SerialSend4", '\u00a0\u0001\u0001\u00a2')
and the OFF command is
publish("mosquitto", "cmnd/sonoff-hob/SerialSend4", '\u00a0\u0001\u0000\u00a1')

That works but I can't get it to function by typing SerialSend4 in console. as it sends everything you type as the exact character so 0x40 appears as 0x40 and binary strings appear as 0s and 1s

mattpurnell, 2018-08-20:
That's awesome thanks, I'll try it out tonight (NZL time)

mattpurnell, 2018-08-21:
Hi @pmknowles  was there any other configuration, beyond simply compiling the latest release?

Did you modify the code as per https://github.com/arendst/Sonoff-Tasmota/issues/1714 ?

alejlle, 2018-08-23:
I have one gagdet like this and sucessfully upload tasmota and under config set as generic and GPIO 0 to command relay, All working fine.

ascillato2, 2018-08-26:
Hi,

Let's gather all LC Wifi Relay config issues in #1282

closed 2018-08-26 by ascillato2

killerbees, 2019-04-13:
> I can confirm that https://github.com/arendst/Sonoff-Tasmota/pull/1282/files
> works with these but you I think you can only use up to 5.12
> [image]
> One of the problems with this device is that one of [...]

SendSerial5 ftw

killerbees, 2019-04-13:
SendSerial5

Senbei123, 2020-12-02:
Short hint also here:
use standard tasmota + configure it like it is described here:
https://templates.blakadder.com/LC-ESP8266.html
Works perfect.