arendst/Tasmota #3407: Two ultrasonic sensors on nodemcu (US-100 and JSN-SR04Tv2)

rajil, 2018-08-04:
Hello,
I have two ultrasonic sensors US-100 and JSN-SR04T which i would like to use with nodemcu. Both these sensors work if they are connected alone to nodemcu. However, when both the sensors are connected the webui only shows the output from a single sensor.

My module setup is as follows:
[image]

The devices connected are:
ADS1115 on D2 and D1
US-100 on D5 (trigger) and D6 (echo)
JSN-SR04T on D7 (trigger) and D0 (ech0)

Only US-100 output is being shown on webui,

[image]

How can i get both the ultrasonic sensors to show up?

andrethomas, 2018-08-04:
`Bonus question: Where is the LM75AD sensor reading coming from?`

You either have a LM75AD on the same I2C bus or another device which uses the same I2C address range... 0x48 - 0x4F and its a false detect

andrethomas, 2018-08-04:
Hmm, the ADS is on the same range so its a false detect for sure... will have to look at the LM75AD driver to see if I can avoid false detect but it pretty much gives valid responses on anything from what I remember... for now disable LM75AD in the config file.

andrethomas, 2018-08-04:
False detection should be solved by proposed PR https://github.com/arendst/Sonoff-Tasmota/pull/3408

andrethomas, 2018-08-04:
@rajil 
To get back to you on the original problem - from what I can see in the driver code only a single SR04 device is supported - I do not think the code knows how to differentiate between the two modules you have connected at this time.

ascillato2, 2018-08-25:
Hi,

> How can i get both the ultrasonic sensors to show up?

That feature is not supported by Tasmota.

If you feel your questions has been addressed, please close this issue. Thanks

ascillato2, 2018-08-27:
Hi,

Closing this issue as there is no feedback. Please, ask to reopen if needed. Thanks.

closed 2018-08-27 by ascillato2