arendst/Tasmota #3409: v6.1.1 from sources not working (with Sonoff S20)

dknubben, 2018-08-04:
I have compiled v6.1.1 and also latest github sources v6.1.1c with platformio and it is uploading sucessful to the device. But after re-plug in, the device is not working. With 5.14.0 no problem.
The binary release  v6.1.1 allsensors is starting (flashed with esptool), button works, but i don't have WPS.

[x ] Searched the problem in issues and in the wiki
[x ] Hardware used : Sonoff S20
[x ] Provide the output of command ``status 0`` : device not starting

Jason2866, 2018-08-04:
Which arduino core version do you use? At the moment still only 2.3.0 works reliable.
If you use 2.4.0 or 2.4.1 this is probably where your problem comes from

dknubben, 2018-08-17:
platformio (CLI) fails to flash the device with 6.1.x (from 5.14)
I solved the problem with this steps:

1. compiling the firmware with platformio

2. full erase the flash
esptool.py --port /dev/ttyUSB0 erase_region 0x0F4000 0x008000
esptool.py --port /dev/ttyUSB0 erase_flash

3. flash the new firmware:
esptool.py --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 .pioenvs/sonoff-allsensors/firmware.bin

ascillato2, 2018-08-22:
Closing as the issue is solved

closed 2018-08-22 by ascillato2