arendst/Tasmota #3412: Sonoff iFan02 fan light turns on when AC turned ON

donoo, 2018-08-05:
I have configured Sonoff ifan02 using 6.1.1 version. Everything works fine but I have one issue.

When AC power is cut and applied again fan light turns on. I m not able to change this using poweronstate setting. In my place power cuts are frequent. 

Anyone has any solution for this problem? I can use openhab rule to control this but want this to work as other sonoff devices with poweronstate  = 3

thanks in advance!

Frogmore42, 2018-08-05:
Provide the requested information from the template and it would be easier to help you.  Without it, we have to guess.

My guess is you have a retained message that is causing the behavior.  You have to reconfigure whatever is configured to use retained messages to not do that AND you HAVE to delete all retained messages.

donoo, 2018-08-12:
@arendst 
Thanks for the reply but in my case light turns on even if it was in off state. Fan behaviour is same as yours. 
Let me recheck everything once again. Will report back if it works fine.

Frogmore42, 2018-08-12:
You probably have a retained message.  Look at the log on power up and it should give a clue as to the problem.

jumblies, 2018-08-16:
I have this problem also and solved it with the following rule.  
```
rule1 on System#Boot do power1 off endon```

I think this is another issue with precompiled binaries.  I have not compiled my own yet to test whether this occurs  but I know there is a variable that can be set to set the poweron state to off.  In my case, this rule suffices.

jumblies, 2018-08-16:
Just tested this and it doesn't work.  The console log SAYS it's off, but it's on.  
```00:00:00 Project sonoff iFan1 (Topic iFan1, Fallback DVES_AB0E72, GroupTopic sonoffs) Version 6.1.1-2_3_0
00:00:00 WIF: Connecting to AP2 shmoo5 in mode 11N as iFan1...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on iFan1.local with IP address 10.10.10.160
00:00:04 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/iFan1/LWT = Online (retained)
00:00:05 MQT: cmnd/iFan1/POWER = 
00:00:05 MQT: tele/iFan1/INFO1 = {"Module":"Sonoff iFan02","Version":"6.1.1","FallbackTopic":"DVES_AB0E72","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/iFan1/INFO2 = {"WebServerMode":"Admin","Hostname":"iFan1","IPAddress":""}
00:00:05 MQT: tele/iFan1/INFO3 = {"RestartReason":"External System"}
00:00:06 MQT: stat/iFan1/RESULT = {"POWER1":"OFF"}
00:00:06 MQT: stat/iFan1/POWER1 = OFF
```

arendst, 2018-08-16:
Yes. I guess the rules option is the only viable way in this case as the microcontroller turns the relay on after power restore.

jumblies, 2018-08-17:
Thanks for comment @arendst .  I'm continually mystified that you keep such a close eye on the issues.  

You're correct that the MCU turns the light <POWER1 > on  at power up.  Even more bizarre is that the module has no awareness of the state of the relay and until the relay becomes actuated by a manual control.  I'll test a rule today to see if I can turn it on and then off to remedy this situation.  

```
00:00:00 Project sonoff iFan1 (Topic iFan1, Fallback DVES_AB0E72, GroupTopic sonoffs) Version 6.1.1-2_3_0
00:00:00 WIF: Connecting to AP1 shmoo3 in mode 11N as iFan1...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on iFan1.local with IP address 10.10.10.160
00:00:04 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/iFan1/LWT = Online (retained)
00:00:05 MQT: cmnd/iFan1/POWER = 
00:00:05 MQT: tele/iFan1/INFO1 = {"Module":"Sonoff iFan02","Version":"6.1.1","FallbackTopic":"DVES_AB0E72","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/iFan1/INFO2 = {"WebServerMode":"Admin","Hostname":"iFan1","IPAddress":"10.10.10.160"}
00:00:05 MQT: tele/iFan1/INFO3 = {"RestartReason":"External System"}
00:00:05 RUL: SYSTEM#BOOT performs "power1 off" **<----DOES NOTHING**
00:00:05 MQT: stat/iFan1/RESULT = {"POWER1":"OFF"}
00:00:05 MQT: stat/iFan1/POWER1 = OFF **<----DOES NOTHING**
00:00:06 MQT: stat/iFan1/RESULT = {"POWER1":"OFF"}
00:00:06 MQT: stat/iFan1/POWER1 = OFF **<----DOES NOTHING**
16:44:15 MQT: tele/iFan1/STATE = {"Time":"2018-08-17T16:44:15","Uptime":"0T00:00:13","Vcc":3.156,"POWER1":"OFF","FanSpeed":0,"Wifi":{"AP":1,"SSId":"shmoo3","RSSI":90,"APMac":"AC:9E:17:A1:21:B8"}}
16:45:38 CMD: rule1
16:45:38 MQT: stat/iFan1/RESULT = {"Rule1":"ON","Once":"OFF","Free":477,"Rules":"on System#Boot do power1 off endon"}
16:45:52 CMD: power1 off
16:45:52 MQT: stat/iFan1/RESULT = {"POWER1":"OFF"} ** <----DOES NOTHING **
16:45:52 MQT: stat/iFan1/POWER1 = OFF
16:45:57 CMD: power off
16:45:57 MQT: stat/iFan1/RESULT = {"POWER1":"OFF"}
16:45:57 MQT: stat/iFan1/POWER1 = OFF
16:46:09 CMD: power1 on
16:46:09 MQT: stat/iFan1/RESULT = {"POWER1":"ON"}
16:46:09 MQT: stat/iFan1/POWER1 = ON
16:46:13 CMD: power1 off
16:46:13 MQT: stat/iFan1/RESULT = {"POWER1":"OFF"} <----**DOES WHAT ITS SUPPOSED TO**
16:46:13 MQT: stat/iFan1/POWER1 = OFF
```

After using the ifan for about a month, I think it's a great product for the price, but not what I would consider "ready for prime time".  As usual, not a problem with tasmota, just the hardware.

arendst, 2018-08-20:
As the microcontroller controlling the relays only accepts commands after 4 seconds from power on you'll notice a delay in restoring the fan speed (and light).

Can't find a better solution.

closed 2018-08-20 by arendst

mikeyroth, 2018-11-19:
I am still having the issue with light turning on after power loss.  It never turns off:

Status 0
'22:48:15 SRC: WebConsole from 192.168.1.2
22:48:15 RSL: Received Topic /status, Data Size 1, Data 0
22:48:15 RSL: Group 0, Index 1, Command STATUS, Data 0
22:48:15 MQT: stat/sonoff_fan1/STATUS = {"Status":{"Module":44,"FriendlyName":["Sonoff Fan 1"],"Topic":"sonoff_fan1","ButtonTopic":"sonoff_fan1","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
22:48:15 MQT: stat/sonoff_fan1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:08:33","StartupUTC":"2018-11-19T03:39:42","Sleep":0,"BootCount":58,"SaveCount":219,"SaveAddress":"F8000"}}
22:48:15 MQT: stat/sonoff_fan1/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-10-30T17:33:53","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:48:15 MQT: stat/sonoff_fan1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Rothgeb",""],"TelePeriod":300,"SetOption":["00008029","55818000","00000000"]}}
22:48:15 MQT: stat/sonoff_fan1/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000183A0","23B617CE","00003BC0"]}}
22:48:16 MQT: stat/sonoff_fan1/STATUS5 = {"StatusNET":{"Hostname":"sonoff_fan1-7127","IPAddress":"192.168.1.40","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:AC:DB:D7","Webserver":2,"WifiConfig":1}}
22:48:16 MQT: stat/sonoff_fan1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.20","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_ACDBD7","MqttUser":"mike","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:48:16 MQT: stat/sonoff_fan1/STATUS7 = {"StatusTIM":{"UTC":"Mon Nov 19 03:48:16 2018","Local":"Sun Nov 18 22:48:16 2018","StartDST":"Sun Mar 11 02:00:00 2018","EndDST":"Sun Nov 04 02:00:00 2018","Timezone":99,"Sunrise":"02:03","Sunset":"11:06"}}
22:48:16 MQT: stat/sonoff_fan1/STATUS10 = {"StatusSNS":{"Time":"2018-11-18T22:48:16"}}
22:48:16 MQT: stat/sonoff_fan1/STATUS11 = {"StatusSTS":{"Time":"2018-11-18T22:48:16","Uptime":"0T00:08:34","Vcc":3.148,"POWER1":"OFF","FanSpeed":0,"Wifi":{"AP":1,"SSId":"Rothgeb","BSSId":"CC:40:D0:EF:82:F8","Channel":3,"RSSI":50}}}'

Console when powered on

'00:00:00 CFG: Loaded from flash at F9, Count 218
00:00:00 APP: Boot Count 58
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff Fan 1 (Topic sonoff_fan1, Fallback DVES_ACDBD7, GroupTopic sonoffs) Version 6.3.0-2_3_0
00:00:00 CFG: Saved to flash at F8, Count 219, Bytes 3584
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 Rothgeb in mode 11N as sonoff_fan1-7127...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 SRC: Retry
00:00:04 SRC: Retry
00:00:04 WIF: Checking connection...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff_fan1-7127.local with IP address 192.168.1.40
00:00:05 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/sonoff_fan1/LWT = Online (retained)
00:00:06 MQT: cmnd/sonoff_fan1/POWER = 
00:00:06 MQT: Subscribe to cmnd/sonoff_fan1/#
00:00:06 MQT: Subscribe to cmnd/sonoffs/#
00:00:06 MQT: Subscribe to cmnd/DVES_ACDBD7/#
00:00:06 MQT: tele/sonoff_fan1/INFO1 = {"Module":"Sonoff iFan02","Version":"6.3.0","FallbackTopic":"DVES_ACDBD7","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/sonoff_fan1/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff_fan1-7127","IPAddress":"192.168.1.40"}
00:00:06 MQT: tele/sonoff_fan1/INFO3 = {"RestartReason":"External System"}
00:00:06 MQT: stat/sonoff_fan1/RESULT = {"POWER1":"OFF"}
00:00:06 MQT: stat/sonoff_fan1/POWER1 = OFF (retained)
00:00:09 APP: (UTC) Mon Nov 19 03:39:51 2018, (DST) Sun Mar 11 02:00:00 2018, (STD) Sun Nov 04 02:00:00 2018
22:39:56 MQT: tele/sonoff_fan1/STATE = {"Time":"2018-11-18T22:39:56","Uptime":"0T00:00:14","Vcc":3.148,"POWER1":"OFF","FanSpeed":0,"Wifi":{"AP":1,"SSId":"Rothgeb","BSSId":"CC:40:D0:EF:82:F8","Channel":3,"RSSI":56}}
22:40:06 WIF: Checking connection...'

NicLogix, 2020-03-04:
I also have the Ifan02, and the same issue.
It seems not the ESP chip, but the Hunter microcontroller which switches the light on before handing over to the ESP.

My workaround was:
ON System#boot DO Backlog POWER1 ON; POWER1 OFF ENDON

Because it is the main bedroom where I sleep, I prefer the light off.  Now on power restore the light comes on and goes off within half a second.

Ifan02 not a fantastic product, a Hunter controller with ESP slapped on and rushed through production.

jumblies, 2020-03-04:
I agree with it not being a great product.  Since I haven't had any issues since the workaround I haven't followed closely.  How was it derived that it was a Hunter MC?

TMaYaD, 2020-06-07:
I'm seeing approximately 3 mins delay before tasmota kicks in. I'll settle for 5 seconds anyday; how do I get that?

rishabmehta7, 2020-07-29:
@TMaYaD See if the Fans SSID becomes visible on reboot, I see that every time and on login i see a reboot in 180 sec, forced reboot turns off the light and turn on the fan in previous state.
easy way to test is Set WiFiConfig to 4 or 5
and see if after a reboot it becomes 2