arendst/Tasmota #3417: CAUTION: Bricked Multiple TH16s when trying to upgrade to 5.14 OTA

virtzilla, 2018-08-06:
**Describe the bug**

I bricked a second (replacement for the 1st one that I thought was an an upgrade anomaly) TH16 this evening when trying to upgrade to 5.14 as a step on. the migration path to 6.x. This one was at 5.9 to start. I. tried to upgrade using over the network using a URL directly to the sonoff.bin binary under 5.14 under the releases. 

Seemed to proceed then the device went completely dead. No lights or sounds when plugged in. Have tried holding down the button to reset it countless times (for minutes at a time), nothing. Tried to connect to it via serial (to a pi) and it's lifeless (spent hours trying to resurrect the first unit). Nothing from esptool or on the console after countless attempts. Argh.

**To Reproduce**
Upgrade a unit from an older release (5.9 in this case where I was paying attention, not sure of the first one but it was 5.x) to 5.14 using the network mechanism and pointing it to 
https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.14.0/sonoff-minimal.bin

If you roll the dice, I hope your luck is better than mine.

**Expected behavior**
A successful or failed upgrade and. not a bricked unit. I know in the past, when uploading a file via the file method, I've gotten mismatch warnings and the like. Would expect those same safeties via the network upgrade mechanism.

Frogmore42, 2018-08-06:
You can't actually brick an esp8266.  It has a built-in bootloader.  Since it is not working with the firmware that is there now, you will need to flash it via serial. I would use something newer and erase the flash first and make sure you are using the right mode DOUT.  It is likely that the firmware you installed was not compatible with the mode of the firmware that was there.  There was a bug in esp8266 Arduino core that was addressed in core 2.4.2 that should help this in the future.

curzon01, 2018-08-06:
Your steps to reproduce (how did you tell Tasmota to load data from a https based Ota URL?) didn't worked here as I expected before I tried to reproduce your steps. Use a http Url for updates.

So my second try to reproduce also failed (because it was successful :grinning: ):

- delete a Sonoff ESP flash via serial and flashed 5.9.1 via serial to have a starting point
- now tried to use a local http url having the 5.14.0 binary minimal
- after a few seconds a successful 5.14.0 was running

So some steps are not as you described.

Frogmore42, 2018-08-06:
Flash via serial, but not DOUT mode.  Now try and update via OTA with the binary that is DOUT.  That is my guess as to what happened.

virtzilla, 2018-08-08:
`how did you tell Tasmota to load data from a https based Ota URL?`

I put the https URL in the GUI under the "OTA Url" field on the Firmware Upgrade page and clicked the "Start Upgrade" button. No errors and then dead to the world.

After several incantations, quite a bit of cursing, and a load of resets I managed to resurrect one of the dead devices and flash it using esptool. The other unit is not behaving the same way.

curzon01, 2018-08-08:
> `how did you tell Tasmota to load data from a https based Ota URL?`
> 
> I put the https URL in the GUI under the "OTA Url" field on the Firmware Upgrade page and clicked the "Start Upgrade" button. No errors and then dead to the world.

The steps are clear :grin: 

The background of my question was that AFAIK Tasmota can not load data from a TLS connection via https. The ESP8266 lib can do Ota updates via https with a hardcoded fingerprint but Tasmota has no definition for that and you did not create a fingerprint so it won't work.

What you get from an https Ota url is garbadge so it can't even you can enter a https url.

ascillato2, 2018-08-27:
Hi,

Closing this issue as there is no feedback. Please, ask to reopen if needed. Thanks.

closed 2018-08-27 by ascillato2