arendst/Tasmota #3418: Wind sensor...

jsponz, 2018-08-06:
Any idea how to connect and use a wind sensor such as: 

https://www.eltako.com/fileadmin/downloads/en/_datasheets/Datasheet_WS.pdf

Thanks!

curzon01, 2018-08-06:
This sensor provides two impulse per rotation. You can connect this sensor to a free GPIO and use a [Counter](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#sensor) to count the number of impulse per time - but without additonal info about your knowledge/environment and without a third-party system it will be difficult to evaluate the wind speed from the number of impulse counts. 

Some additonal information (from Amazon review, unfortunately only in german, use a translator if neccessary) can be found here:
[Freie Software für Raspberry Pi, incl. Formel](https://www.amazon.de/gp/customer-reviews/R3C68WVOLJ7ZTO/ref=cm_cr_othr_d_rvw_ttl?ie=UTF8&ASIN=B0018LBFG8)

Summarized:
The formular from Eltako is

`v [km/h] = 1.761 / (1 + rps) + 3.013 * rps`

where rps = rotations per second
and one full rotation are 2 impulse

Example: 
- You got `140` impulse within `60` sec. then you have `140/60/2` or `140/(2*60)` = `1.1666666667` rps
- using formular above you have a wind speed of `1.761 / (1 + (140/60/2)) + 3.013 * (140/60/2) = 4.32 km/h`

ascillato2, 2018-08-18:
Hi,

Closing this issue as your questions has been addressed. Ask to reopen this issue if you need it. Thanks

closed 2018-08-18 by ascillato2

tremichl, 2018-10-21:
Hi, I am also interested in connecting this sensor to ESP/Tasmota. Calculation of windspeed will be done in FHEM. At the moment I can only get counts, but don't know how to get impulses per time (rps). Is it possible to do that with a rule in Tasmota and how should it look like?

curzon01, 2018-10-22:
Without rules (using current dev 6.2.1.18):

GPIOx 42 (Counter1)
CounterType1 0 (Pulse Count)
TelePeriod xx (example xx=30)
you get now every 30 sec. a SENSOR msg like this:

`.../SENSOR = {"Time":"2018-10-22T07:08:58","COUNTER":{"C1":56},"ANALOG":{"A0":0}}`

next 30 sec. you get  for example
`.../SENSOR = {"Time":"2018-10-22T07:09:28","COUNTER":{"C1":59},"ANALOG":{"A0":0}}`

FHEM do extract of JSON (e.g. your Tasmota device called Sonoff):
```
# Sensor device
define Sonoff MQTT_DEVICE
attr Sonoff IODev mqttServer
...
attr Sonoff subscribeReading_sensor stat/sonoff/SENSOR
# Extract JSON as readings
define Sonoff_ej3 expandJSON Sonoff:.*:.{.*}
```
now you have the readings "Time" and "COUNTER_C1"

the rest you have to write by yourself (hopefully):
Use perl within FHEM to store your Time, Time difference from previous Time, counter difference and calc using the formular above

tremichl, 2018-10-24:
Thanks for detailed suggestion, very helpful. It would be more convenient if calculations could be done in Tasmota.  No need to send mqtt message every 10 to 30 seconds if wind speed is 0 (which is most of the time), faster response if wind is changing, no need checking counter overflow in FHEM...
I am not an programmer, but it would be great if this sensor could be integrated into Tasmota by an experient coder.

heikoh81, 2019-03-03:
Thumbs up.
I have this wind sensor in use with raspberry as posted in the link above.
However, would appreciate to have it in tasmota, connected to a Wemos D1 Mini for example.

sergios100, 2020-05-17:
Sensor is now implemented in 8.3.0 see here:
[https://github.com/arendst/Tasmota/pull/8283/](url)