arendst/Tasmota #3419: RF Bridge - Dooya RC-2700 remote - understanding "sniffed" cods

autofrank, 2018-08-06:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ X] Hardware used : RF Bridge V1
- [ X] Provide the output of command ``status 0`` :
```
status 0
09:56:29 SRC: WebConsole from 192.168.1.137
09:56:29 RSL: Group 0, Index 1, Command STATUS, Data 0
09:56:29 RSL: STATUS = {"Status":{"Module":25,"FriendlyName":["Sonoff_RF1"],"Topic":"SonoffRF1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
09:56:29 RSL: STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/SonoffRF1.ino.bin","RestartReason":"Power on","Uptime":"0T00:03:11","StartupUTC":"2018-08-06T08:53:18","Sleep":0,"BootCount":102,"SaveCount":142,"SaveAddress":"F8000"}}
09:56:29 RSL: STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-31T17:59:09","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
09:56:29 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["SonoffStarWifi",""],"TelePeriod":300,"SetOption":["00000001","55818000"]}}
09:56:29 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":536,"Free":464,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","0C000000","03B6179E","00000000"]}}
09:56:29 RSL: STATUS5 = {"StatusNET":{"Hostname":"SonoffRF1-6998","IPAddress":"192.168.1.146","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:8F:3B:56","Webserver":2,"WifiConfig":3}}
09:56:29 RSL: STATUS7 = {"StatusTIM":{"UTC":"Mon Aug 06 08:56:29 2018","Local":"Mon Aug 06 09:56:29 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:30","Sunset":"20:21"}}
09:56:29 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-08-06T09:56:29"}}
09:56:29 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-08-06T09:56:29","Uptime":"0T00:03:11","Vcc":3.238,"Wifi":{"AP":1,"SSId":"SonoffStarWifi","RSSI":68,"APMac":"06:18:D6:E1:DD:3D"}}}
```

**(Please, remember to close the issue when the problem has been addressed)**
Rf Bridge running 6.1.1 with Portisch firmware upgrade. All upgrades successful.

I have a DC2700 Dooya Remote (https://www.aliexpress.com/item/Dooya-Remote-Curtain-Accessories-Remote-Control-Transmitter-DC1600-For-DT52E/32613885486.html ) for a projector screen. It is a simple three button remote UP/STOP/DOWN

I can sniff the button presses using rfraw 166
`21:53:57 CMD: rfraw 166
21:53:57 SRC: WebConsole from 192.168.1.137
21:53:57 RSL: Group 0, Index 1, Command RFRAW, Data 166
21:53:57 RSL: RESULT = {"RfRaw":"ON"}
21:53:57 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
21:54:01 RSL: RESULT = {"RfRaw":{"Data":"AAA60602B8419AC13C55"}}
21:54:14 RSL: RESULT = {"RfRaw":{"Data":"AAA60602B8419AC155"}}
21:54:19 RSL: RESULT = {"RfRaw":{"Data":"AAA60602B8419AC11E55"}}`

This represents three button presses

- DOWN
- STOP
- UP

I can also sniff the code using rfraw 177

`21:49:54 CMD: rfraw 177
21:49:54 SRC: WebConsole from 192.168.1.137
21:49:54 RSL: Group 0, Index 1, Command RFRAW, Data 177
21:49:54 RSL: RESULT = {"RfRaw":"ON"}
21:49:54 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
21:49:57 RSL: RESULT = {"RfRaw":{"Data":"AA B1 05 12DE 05D2 02CC 0172 23A0 0123322323233232323223323232323223233232232332233223233232323232233232232323233230 55"}}
21:50:31 RSL: RESULT = {"RfRaw":{"Data":"AA B1 06 12D4 05D2 02A4 01D5 0102 226A 0123422323234242424223424242424223234242232342234223234242424242234223422342234220 55"}}
21:50:48 RSL: RESULT = {"RfRaw":{"Data":"AA B1 06 12DE 05C8 028C 0196 00FA 2242 0123322323233232323223323232323223233232232332233223233232323232233223322332233220 55"}}
21:51:15 RSL: RESULT = {"RfRaw":{"Data":"AA B1 05 12DE 05C8 02D5 0172 23A0 0123322323233232323223323232323223233232232332233223233232323232233232322323232330 55"}}
`

This represents the same three button presses

- DOWN
- STOP (pressed twice)
- UP

I can't find the Dooya protocol and having difficulty in sending the code from the bridge and control the screen

I'm also not sure if it is easier to decode/understand the rfraw166 or rfraw 177 result

Any help would be appreciated 
Thanks

autofrank, 2018-08-06:
I've been trying to decipher this using the example in the HT6P20B post  [https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/22](url)

This is what I have

Code sniffed 
21:49:57 RSL: RESULT = {"RfRaw":{"Data":"AA B1 05 12DE 05D2 02CC 0172 23A0 0123322323233232323223323232323223233232232332233223233232323232233232232323233230 55"}}

.. and the data part
AA B1 05 12DE 05D2 02CC 0172 23A0 0123322323233232323223323232323223233232232332233223233232323232233232232323233230 55

separating out the data...
AA B1 05 12DE 05D2 02CC 0172 23A0 0 12 33 22 32 32 33 23 23 23 22 33 23 23 23 23 22 32 33 23 22 32 33 22 33 22 32 33 23 23 23 23 22 33 23 22 32 32 32 33 23 0 55

I need to add the len and the repeat time
this has 51 part = len (dec 51 = 33 hex)
copy repeat of 20 (20 dec = 14 in hex)
replacing the B1 with a B0 to 'send/transmit'

and this gives me 
AA B0 33 05 14 12DE 05D2 02CC 0172 23A0 0 12 33 22 32 32 33 23 23 23 22 33 23 23 23 23 22 32 33 23 22 32 33 22 33 22 32 33 23 23 23 23 22 33 23 22 32 32 32 33 23 0 55

.. but when I enter this with an rfraw command in the console
I get the following but no screen movement

12:25:20 RSL: RESULT = {"RfRaw":"ON"}
12:25:29 CMD: rfraw AAB033051412DE05C802CF017223A0012332232323323232322332323232322323323223233223322323323232323223323223232323323055
12:25:29 SRC: WebConsole from 192.168.1.137
12:25:29 RSL: Group 0, Index 1, Command RFRAW, Data AAB033051412DE05C802CF017223A0012332232323323232322332323232322323323223233223322323323232323223323223232323323055
12:25:29 RSL: RESULT = {"RfRaw":"ON"}
12:25:31 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}

I tried adjusting the timing aspect to the most common from 10 sample commands but nothing
I tried the same with the other 2 commands and nothing either 

Any help appreciated 

PS _ I have read about adding new protocols but not sure if this is the next route

autofrank, 2018-08-06:
I resolved this with Portisch help and posted the working codes here [https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/23](url)

closed 2018-08-06 by autofrank

Portisch, 2018-08-08:
It looks like a "bug" in the uart receive from the EFM8:
https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/23#issuecomment-411288705