arendst/Tasmota #3420: Button 1 always on

bruno-wladimir, 2018-08-06:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ x] Hardware used : Sonoff T1 US
- [ ] Provide the output of command ``status 0`` :
```
When powering the Sonoff T1, button 1 is active by default, would it be a software or hardware problem? I already tested in 2 modules (2 gang and 3 gang) and the same error happens!

```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-08-06:
Have you read the wiki?
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK-and-T1-EU

bruno-wladimir, 2018-08-06:
Yes! I can flash the firmware perfectly, but the buttons are ON by default

Frogmore42, 2018-08-06:
If you mean the relay turns the power on by default, there is a setting that controls that.  It is described in the wiki, as is how to change that.

If that is not what you mean by button 1 is active by default, you will need to use some other words to describe the behavior you are seeing.  Include the web/serial log showing the bad behavior.

bruno-wladimir, 2018-08-06:
seems to be a hardware problem, even without firmware the button stays active!  I tested the same code in another module and it works perfectly

Frogmore42, 2018-08-06:
Please close the issue then.

closed 2018-08-07 by bruno-wladimir

bruno-wladimir, 2018-08-07:
@AdighaLogic  I think it's a hardware problem, even if the firmware is not active by default, your choice only applies to Firmware Tasmota right?

AdighaLogic, 2018-08-07:
Correct, I believe you're right. if the firmware is not active by default, then it must be a hardware issue.
I just flashed 2 T1 US version 2 Ch and everything went smooth. but on the first power up the first button comes on automatically, so i set option 13 to 1 and that problem went a way.