arendst/Tasmota #3421: Emulating Ifan02 with multiple relays/interlock

clyra, 2018-08-06:
Hi,

I have a 3 speed fan where the connection of the live with one of three wires set the speed. Using the interlock feature and a sonoff 4ch Pro I'm now abe to select the fan speed by sending a power on request to the correct relay (power1 = low, power2 = medium, power3 = high). It would be very nice if we could emulate the Ifan behaviour and choose the speed by sending the desired speed to a speed_command_topic like showed here (https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant). Making the request as generic as possible, it would be nice to select which relay is on by sending a value to a single topic (0=all off, 1 = relay 1, 2=relay2 and so on). ty.

digiblur, 2018-08-07:
Might be something you'd be able to do with the Tasmota rules of via MQTT with NodeRed fairly easy.

clyra, 2018-08-08:
Hi,

yes... I just figured out how to do that using rules. But I guess it will still be nice if the emulation could be done just as selecting something like "fan mode". BTW, rules support is not compiled in the bin images from here (https://github.com/arendst/Sonoff-Tasmota/releases), right?

arendst, 2018-08-11:
See readme about what version support rules.

Keep in mind that the ifan02 does more than just switching relays and connecting capacitors. For instance the low speed setting first turns on the fan to full speed and after some seconds it sets low speed. This to make the fan start at all.

digiblur, 2018-08-11:
@arendst does the esp8266 do that part or the other chip? (atmel maybe?)

clyra, 2018-08-11:
Ok. We should close the issue.

closed 2018-08-11 by clyra

limks1980, 2019-03-23:
> Hi,
> 
> yes... I just figured out how to do that using rules. But I guess it will still be nice if the emulation could be done just as selecting something like "fan mode". BTW, rules support is not compiled in the bin images from here (https://github.com/arendst/Sonoff-Tasmota/releases), right?

can post a sample of the rules files on how to control the relay

clyra, 2019-03-24:
Hi,

Sure, here is it:

Rule1 on event#speed0 do Power1 Off endon on event#speed1 do Power1 On
endon on event#speed2 do Power2 On endon on event#speed3 do Power3 On endon
Rule2 on Power1#State do status endon on Power2#State do status endon on
Power3#State do status endon

The full topic about this is here:

https://community.home-assistant.io/t/tasmota-the-desktop-fan/53523/15