arendst/Tasmota #3422: How to get/set PWM/channel values in C/C++?

renne, 2018-08-06:
Hi,

I'm working on a closed-loop light-control function which needs direct C/C++ access without MQTT parsing overhead to the PWM1-5/channel1-5 functions and variables. Where can I find them in the code?

andrethomas, 2018-08-06:
xdrv_04_light.ino, I think.

andrethomas, 2018-08-06:
Its the only place I can find that has AnalogWrite's in it...

renne, 2018-08-06:
In that case I need to know to which PINs PWM1-5 are mapped. Do you have any idea?

andrethomas, 2018-08-07:
I think the mapping will depend on how the device is configured in the web menu, or do you perhaps mean from wirish...

renne, 2018-08-07:
That's the point. I'm currently working on extending the APDS-9960 code with closed-loop light control. As there are different devices with different GPIO mappings I need a generic way to set the PWM values or at least query the GPIO mapping.

andrethomas, 2018-08-07:
Then I think the best would be to interact with xdrv_04_light.ino to add compiler directives on whether or not to use data from APDS-9960 through a shared global variable perhaps.

ascillato2, 2018-08-25:
Yes, exactly

renne, 2018-08-28:
I'm currently waiting on the Broadcom/EBV support for more information about the color-space of the APDS-9960. Until then I have to put the closed-loop light control on hold.

ascillato, 2018-08-29:
Ok, no problem, but the question of this issue has been answered? There is something else we can help?

renne, 2018-08-29:
In theory, yes. ;-)

closed 2018-08-29 by renne