arendst/Tasmota #3423: Daikin A/C remote control (Panasonic protocol)

iz3man, 2018-08-07:
i installed tasmota 6.1.1 on one of my nodemcu, as they have enough exploited i/o pins.
connected an IR receiver to D2 and  an IR transmitter to D4, and set IrSend and IrReceive accordingly.
all seems to work, and when i press buttons on my Daikin's A/C remote control, i can receive the commands, which are identified as PANASONIC protocol.

```
17:11:40 MQT: tele/sonoff/RESULT = {"IrReceived":{"Protocol":"PANASONIC","Bits":48,"Data":"E4000090"}}
17:11:46 MQT: tele/sonoff/RESULT = {"IrReceived":{"Protocol":"PANASONIC","Bits":48,"Data":"E4000010"}}
```

as this seems hex code to me, i then tried to send the exact same code to the A/C to turn it on/off, or get some reaction at all.

so in console i sent:

`IRsend {"Protocol":"PANASONIC","Bits":48,"Data":"0xE4000090"}`

and received 

`17:15:20 MQT: tele/sonoff/RESULT = {"IrReceived":{"Protocol":"PANASONIC","Bits":48,"Data":"E4000090"}}`

exactly the same as what i received when i pressed the button on the remote. but nothing happens. by being able to send/receive commands on the nodemcu directly, i guess i can rule out any wiring issues.
i even moved the IR led very close to the IR receiver of the A/C just to be sure.

any idea what's going on here, and how to trouble shoot it? i'd really love to add my A/C together with automated shades and temperature sensor to home-assistant.

thanks

andrethomas, 2018-08-07:
I'm not sure that Panasonic HVAC is supported yet - I tried using it on a Defy HVAC which I know has a Panasonic board in it and got the same result as you.

Would be interesting to see how one would make it work.

znanev, 2018-08-07:
@iz3man You can check the following excellent work on reverse engineering the Daikin A/C protocol:

https://github.com/blafois/Daikin-IR-Reverse

The code is in Java, but that shouldn't be a problem to understand how it works.

iz3man, 2018-08-07:
Excellent. Thank you!

andrethomas, 2018-08-07:
I wonder if its Daikin specific, or if the same will work with other ones identified as Panasonic... and of course someone will need to port the code :|

aymeric106, 2018-08-08:
What is the model of your Daikin Remote ?
Here you have already 2 different protocols (it seems)
http://rdlab.cdmt.vn/project-2013/daikin-ir-protocol

ascillato, 2018-08-25:
Copied from #3228 :

> In fact, it's already in the library ...
> https://github.com/arendst/Sonoff-Tasmota/tree/development/lib/IRremoteESP8266-2.2.1.02/examples/TurnOnDaikinAC
> 
> It's already in the base IR library, but not implemented in the IRhvac command.
> There's a couple of PR, for two other manufacturer
> #3227 (Fujitsu)
> and
> #3274 (midea/komico)
>
> Hope it will be implemented

ripkrizbi, 2018-08-26:
Just a short follow-up on this, there is an excellent work done on IR and HVAC at https://github.com/ToniA/arduino-heatpumpir
It seems to be well written and extendable to other HVAC systems (besides many that are already implemented), plus it is ESP8266 ready. Might be worth a while implementing it under Tasmota...

valerie16901, 2018-09-24:
> i installed tasmota 6.1.1 on one of my nodemcu, as they have enough exploited i/o pins.
> connected an IR receiver to D2 and an IR transmitter to D4, and set IrSend and IrReceive accordingly.
> all [...]

Hello I have the exact same problem but with a Panasonic TV, I am using a D1 mini with an IR shield I can see the codes from my Panasonic TV remote, however when I send the codes via console or MQTT nothing happens. Did you found a solution?

andrethomas, 2018-09-24:
@Jason2866 Posted some interesting info on Tasmota Chat but I have not been able to test it yet.

grebar, 2018-09-27:
Here below my follow-up. 
Some days ago I opened a new Issue, referencing this one and proposing just like @ripkrizbi made, the integration of https://github.com/ToniA/arduino-heatpumpir.

After a while (a few hours) @ascillato closed my issue because he tag it as duplicated so no one saw my posted [Issue](https://github.com/arendst/Sonoff-Tasmota/issues/3870).

I tried to integrate the arduino-heatpumpir in Tasmota, and, after while I bought the first problem and first success.

Briefly, I detected a problem that does not allow the integration of arduino-heatpumpir lib in Tasmota, the problem consists in an alreay used definition of the identifiers POWER_ON and POWER_OFF; I described the problem in bullet 1 of this [post](https://github.com/ToniA/arduino-heatpumpir/issues/63#issuecomment-423729181).
I am not very expert of C / C++ languages so I asked for an help on the issue desciped in the linked post.

In the mean while, in my development enviroment, I bypassed the problem described in the post renaming the duplicated identifiers used in arduino-heatpumpir.

So, to intengrate the arduino-heatpumpir in Tasmota, we need to follow the following main steps:
1) rename the POWER_ON/POWER_OFF definitions in arduino-heatpumpir 
File: all files that include POWER_ON / POWER_OFF
rename as specified below
POWER_OFF  -> HVAC_POWER_OFF
POWER_ON  -> HVAC_POWER_ON    

2) copy the arduino-heatpumpir folder in tasmota\lib folder as tasmota\lib\HeatpumpIR

3) apply the changes to xdrv_05_irremote.ino file

I attach my preliminary working version of [xdrv_05_irremote.ino](https://github.com/arendst/Sonoff-Tasmota/files/2425865/xdrv_05_irremote.ino.txt)
.

In my version of xdrv_05_irremote.ino I commented the old unused source code.

In my opinion the next steps for a full integrations are:
1) solve the issue related to duplicated definitions (POWER_ON / POWER_OFF)
2) test and complete my xdrv_05_irremote.ino changes
3) propose the integration in the tasmota to @arendst.

Gregorio

Jason2866, 2018-09-28:
Take a look here:
https://github.com/arendst/Sonoff-Tasmota/issues/2116#issuecomment-423773152

ascillato2, 2018-10-01:
Hi,

Thanks a lot for sharing your ideas. 

As Tasmota can't handle ALL possible IR protocols, due to flash size restrictions, we should be better implementing a RAW IR send/receive as requested at #2116.

Please, if you have further questions or ideas, do not hesitate on contacting us in the Tasmota Discord Chat. Thanks.

closed 2018-10-01 by ascillato2

raidolo, 2018-12-11:
There’s no need to include hearpumpir.
I’ve forked the 6.3 version with the new IRRemote standard library and coded the DAIKIN HVAC driver in the xdrv_05_irremote.ino file. Details here in this thread:

https://github.com/arendst/Sonoff-Tasmota/issues/2557#issuecomment-445367638

iz3man, 2018-12-11:
Sound great, i will try asap!!! Even though it's time to turn on the heating and not A/C ;)