arendst/Tasmota #3425: S31 module shows no power consumption

peterloron, 2018-08-07:
Hello. I have a S31 module I have flashed to 6.1.1. In general it is working fine. However, sometimes when the power is switched on (either in the Tasmota web page, or via MQTT), the power state is changed properly, but I do not see any energy monitoring data. The webpage as well as MQTT messages show the power state as ON, but using 0 Watts, 0A.

Sometimes this state persists, other times it resolves itself.

Any thoughts on debugging this?

Thanks.

Frogmore42, 2018-08-08:
Is there an actual load?
How big?
Is it on?
Logging?
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#logging
level 4 would show everything.  Then you need to capture it showing the good and the bad behavior.

peterloron, 2018-08-08:
Thank you @Frogmore42. Yes, there is an actual load (a set of LED lights). It is about 0.25A.

I have bumped up the logging, and will see if I can collect any data.

peterloron, 2018-08-09:
Ok, I have gotten some logs and it looks like there is an error in how data is coming back from the power sensor IC.

`22:09:53 CSE: Checksum failure
22:09:53 SER: Received 3C 5A 00 DA CE 4D 0A A8 02 61 FA F1 BA A9 98 55 5A 02 CA 10 00 05 C9 00
22:09:53 CSE: Checksum failure
22:09:53 SER: Received 3C 5A 00 DB 4D 4D 0A A8 02 61 FA F1 BA A9 18 55 5A 02 CA 10 00 05 C9 00
22:09:53 CSE: Checksum failure
22:09:53 SER: Received 3C 5A 00 DB 4D 4D 0A A8 02 61 FA F1 BA AA 19 55 5A 02 CA 10 00 05 C9 00
22:09:53 CSE: Checksum failure
22:09:53 SER: Received 3C 5A 00 DB 4D 4D 0A A8 02 61 FA F1 BA AB 1A 55 5A 02 CA 10 00 05 C9 00
22:09:53 CSE: Checksum failure
22:09:53 SER: Received 3C 5A 00 DB 4D 4D 0A A8 02 61 FA F1 BA AB 1A 55 5A 02 CA 10 00 05 C9 00
22:09:53 CSE: Checksum failure
22:09:54 SER: Received 3C 5A 00 DB 4D 4D 0A A8 02 61 FA F1 BA AC 1B 55 5A 02 CA 10 00 05 C9 00
22:09:54 CSE: Checksum failure
22:09:54 SER: Received 3C 5A 00 DB 4D 4D 0A A8 02 64 E2 F1 BA AD 07 55 5A 02 CA 10 00 05 C9 00
22:09:54 CSE: Checksum failure
22:09:54 SER: Received 3C 5A 00 DB 4D 4D 0A A8 02 64 E2 F1 BA AD 07 55 5A 02 CA 10 00 05 C9 00
22:09:54 CSE: Checksum failure
22:09:54 SER: Received 3C 5A 00 DB 4D 4D 0A A8 02 64 E2 F1 BA AE 08 55 5A 02 CA 10 00 05 CB 00
22:09:54 CSE: Checksum failure
22:09:54 SER: Received 3C 5A 00 DB 4D 4D 0A A8 02 64 E2 F1 BA AF 0B 55 5A 02 CA 10 00 05 CB 00
22:09:54 CSE: Checksum failure
22:09:54 SER: Received 3C 5A 00 DB 4D 4D 0A A8 02 64 E2 F1 BA AF 0B 55 5A 02 CA 10 00 05 CB 00
22:09:54 CSE: Checksum failure`

Frogmore42, 2018-08-09:
What does the good one look like?

arendst, 2018-08-11:
Just released a fix in 6.1.1c Pls give it a try and let me know if it fixes your problem.

peterloron, 2018-08-11:
@arendst Thanks! I compiled the firmware using Platform IO so I could set my wifi credentials, etc. I should be able to do a git pull and then build a new bin file I can do an OTA update, yes?

Also, recommended build settings for the S31...which platform and board?

Thanks so much!

arendst, 2018-08-11:
All info is in the git. All sonoFfs are generic esp8266 1M no spiffs DOUT. ALWAYS. See platformio.ini

peterloron, 2018-08-12:
After flashing with 6.1.1c (had to flash the minimal firmware first), the errors seem to have vanished. Thanks!

closed 2018-08-12 by peterloron

rohankapoorcom, 2018-08-18:
@arendst I'm seeing a similar problem with the S31 module. None of the power values ever change from 0.

I have the logging level set to 4 and am not seeing any sort of checksum errors.

I'm running git revision: `bb87f99ee0245d32b9805e05a05d936e808d59dc` (6.1.1.6).

`00:02:07 MQT: tele/sonoff/SENSOR = {"Time":"1970-01-01T00:02:07","ENERGY":{"Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}`

Frogmore42, 2018-08-18:
This issue was closed, so not okay to reuse it for something that is likely not even the same issue. 

Please open a new issue and make sure you provide all of the information requested in the template.  It is fine to reference this issue and even better to include all the things you did to troubleshoot the issue you are seeing.