arendst/Tasmota #3426: relay 01 does not work

bruno-wladimir, 2018-08-07:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x ] Hardware used : Sonoff T1 US
- [ ] Provide the output of command ``status 0`` :
```
I have two modules t1 and the firmware change relay 01 does not work, even with a simple code where you activate and deactivate the relay, it does not work. relay 2 works perfectly in both the tasmota and a simple blink. Anyone have any idea what it can be?

```

**(Please, remember to close the issue when the problem has been addressed)**

Frogmore42, 2018-08-08:
do not put your comments in a code block (unless it really is code 😄)

You have not included the requested information.  It is not even clear if you are using Tasmota.
With the information you did provide, it sounds like you either have a hardware failure or something is misconfigured.

bruno-wladimir, 2018-08-08:
I'm using the firmware tasmota! When I changed the original firmware, I can not get the gpio 12 (rele1) on, even using a simple code doing digitalWrite (12, HIGH / low), I do not know if I can ask this here but I did not find other forums as good as this. I saw that it has a chip (EFM8BB1) that triggers the relays, I thought it might have some relation to the problem.

Frogmore42, 2018-08-08:
This is NOT a forum.  This is a forum:
https://forum.digital-pig.com

I don't even understand what you have tried and exactly what is not working.  I am guessing that English is not your primary language, which probably makes communication harder.

This file 
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/sonoff_template.h
Indicates the Touch uses GPIO12 with a direct connection to the relay.

If it is not working.  There are many things that can cause that.  But I would check voltages to see if they are correct.  If you don't know how to do that, you are probably better off just getting another one.  If you find three with the same problem, you are probably doing something wrong in the software/configuration.

closed 2018-08-08 by bruno-wladimir