arendst/Tasmota #3427: Compile error undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'

anthonykeane, 2018-08-08:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [Sonoff Basic] Hardware used : 
- [] Provide the output of command ``status 0`` : NOT APPLICABLE
```
STATUS 0 OUTPUT HERE - BUILD ERROR NOT LOADED TO SONOFF
```

**(Please, remember to close the issue when the problem has been addressed)**

This is a BUILD error from the Arduino IDE:

Arduino: 1.8.3 (Mac OS X), TD: 1.37, Board: "Generic ESP8266 Module, 80 MHz, Flash, ck, 26 MHz, 40MHz, DOUT, 1M (no SPIFFS), 2, v1.4 Higher Bandwidth, Disabled, None, Only Sketch, 115200"

Build options changed, rebuilding all
sketch/sonoff.ino.cpp.o:(.text._Z18WifiWpsConfigBeginv+0x4): undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'
sketch/sonoff.ino.cpp.o: In function `WifiWpsConfigBegin()':
Arduino/Sonoff-Tasmota-6.1.1/sonoff/support.ino:947: undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'
Arduino/Sonoff-Tasmota-6.1.1/sonoff/support.ino:947: undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'
Arduino/Sonoff-Tasmota-6.1.1/sonoff/support.ino:947: undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'
Arduino/Sonoff-Tasmota-6.1.1/sonoff/support.ino:947: undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Generic ESP8266 Module.

Previous versions worked:

Frogmore42, 2018-08-08:
As I said, I don't use Arduino, but you probably are missing a library.  I would start over with a fresh install of Arduino and the Tasmota code base.  That will probably fix it.

grantalewis, 2018-08-08:
I'm experiencing this same issue.

anthonykeane, 2018-08-08:
SOLVED KINDA:

I moved to a Ubuntu box from a MAC ( 2 MAC's actually) and the issue went away. Identical setups on the Linux and Mac. 

Mac fails - Ubuntu works

grantalewis, 2018-08-08:
I'm using a Win10 PC. Unfortunately no easy access to a Linux box.

anthonykeane, 2018-08-08:
@grantalewis but WiFiManager lib does not seem to be used my Tasmota

I will keep trying to find an answer, let me know if you find one.

grantalewis, 2018-08-08:
Will do. Thanks.

grantalewis, 2018-08-08:
For reference, that section of code in the **support.ino** file is as follows:

```
  wps_result = status;
  if (WPS_CB_ST_SUCCESS == wps_result) {
    wifi_wps_disable();
  } else {
    snprintf_P(log_data, sizeof(log_data), PSTR(D_LOG_WIFI D_WPS_FAILED_WITH_STATUS " %d"), wps_result);
    AddLog(LOG_LEVEL_DEBUG);
    wifi_config_counter = 2;
```

...which probably concerns this section in the **user_config.h** file:

```
// -- Wifi ----------------------------------------
#define WIFI_IP_ADDRESS        "0.0.0.0"         // [IpAddress1] Set to 0.0.0.0 for using DHCP or IP address
#define WIFI_GATEWAY           "192.168.2.254"   // [IpAddress2] If not using DHCP set Gateway IP address
#define WIFI_SUBNETMASK        "255.255.255.0"   // [IpAddress3] If not using DHCP set Network mask
#define WIFI_DNS               "192.168.2.27"    // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)

#define STA_SSID1              ""                // [Ssid1] Wifi SSID
#define STA_PASS1              ""                // [Password1] Wifi password
#define STA_SSID2              ""                // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              ""                // [Password2] Optional alternate AP Wifi password
#define WIFI_CONFIG_TOOL       WIFI_WPSCONFIG    // [WifiConfig] Default tool if wifi fails to connect
                                                 //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT)
```

(Apologies if I'm posting the obvious. This is all new to me.)

grantalewis, 2018-08-08:
Fresh install of Ubuntu on an HP laptop, fresh install of all Arduino software and libraries (etc.). Error still exists.

Arduino: 1.8.5 (Linux), Board: "Generic ESP8266 Module, 80 MHz, Flash, ck, 26 MHz, 40MHz, DIO, 1M (64K SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

```
/tmp/arduino_build_82417/sketch/sonoff.ino.cpp.o:(.text._Z18WifiWpsConfigBeginv+0x4): undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'
/tmp/arduino_build_82417/sketch/sonoff.ino.cpp.o: In function `WifiWpsConfigBegin()':
/home/grant/Documents/Sonoff-Tasmota-6.1.1/sonoff/support.ino:947: undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'
/home/grant/Documents/Sonoff-Tasmota-6.1.1/sonoff/support.ino:947: undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'
/home/grant/Documents/Sonoff-Tasmota-6.1.1/sonoff/support.ino:947: undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'
/home/grant/Documents/Sonoff-Tasmota-6.1.1/sonoff/support.ino:947: undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'
collect2: error: ld returned 1 exit status
...
exit status 1
Error compiling for board Generic ESP8266 Module.

```

andyrichter, 2018-08-08:
maybe you can change revert the esp8266 driver version with boards manager to version 2.4.0

Frogmore42, 2018-08-08:
This is the reason that I like PlatformIO.  While there is a bit of a learning curve.  It and VSCode are actually really nice to use.  But, I am a trained professional 😀.

Arduino works great when your needs are not too far from Blinky.  But, when you start to need many libraries and files, the Arduino environment is extremely limiting.

It's been a long while since I last used Arduino, so maybe it has changed.

grantalewis, 2018-08-08:
Thanks for the reply. Still getting an error, but at least the error has changed. So... progress?

```
Generating function prototypes...
"/home/grant/Documents/arduino-1.8.5/portable/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/grant/Documents/arduino-1.8.5/portable/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/include" "-I/home/grant/Documents/arduino-1.8.5/portable/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include" "-I/home/grant/Documents/arduino-1.8.5/portable/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_41466/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC"  -DESP8266 "-I/home/grant/Documents/arduino-1.8.5/portable/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266" "-I/home/grant/Documents/arduino-1.8.5/portable/packages/esp8266/hardware/esp8266/2.4.0/variants/generic" "-I/home/grant/Documents/arduino-1.8.5/libraries/esp-knx-ip-0.5.1" "-I/home/grant/Documents/arduino-1.8.5/portable/packages/esp8266/hardware/esp8266/2.4.0/libraries/EEPROM" "-I/home/grant/Documents/arduino-1.8.5/portable/packages/esp8266/hardware/esp8266/2.4.0/libraries/ESP8266WiFi/src" "-I/home/grant/Documents/arduino-1.8.5/portable/packages/esp8266/hardware/esp8266/2.4.0/libraries/ESP8266WebServer/src" "-I/home/grant/Documents/arduino-1.8.5/portable/packages/esp8266/hardware/esp8266/2.4.0/libraries/Ticker" "-I/home/grant/Documents/arduino-1.8.5/portable/packages/esp8266/hardware/esp8266/2.4.0/libraries/ESP8266HTTPClient/src" "-I/home/grant/Documents/arduino-1.8.5/portable/packages/esp8266/hardware/esp8266/2.4.0/libraries/ESP8266httpUpdate/src" "-I/home/grant/Documents/arduino-1.8.5/portable/sketchbook/libraries/ArduinoJson/src" "-I/home/grant/Documents/arduino-1.8.5/portable/packages/esp8266/hardware/esp8266/2.4.0/libraries/DNSServer/src" "-I/home/grant/Documents/arduino-1.8.5/portable/packages/esp8266/hardware/esp8266/2.4.0/libraries/ESP8266mDNS" "-I/home/grant/Documents/arduino-1.8.5/portable/packages/esp8266/hardware/esp8266/2.4.0/libraries/Wire" "-I/home/grant/Documents/arduino-1.8.5/portable/sketchbook/libraries/PubSubClient/src" "-I/home/grant/Documents/arduino-1.8.5/libraries/TasmotaSerial-2.0.0/src" "-I/home/grant/Documents/arduino-1.8.5/libraries/IRremoteESP8266-2.2.1.02/src" "-I/home/grant/Documents/arduino-1.8.5/libraries/C2Programmer-1.0.0/src" "-I/home/grant/Documents/arduino-1.8.5/libraries/NeoPixelBus-2.2.9/src" "-I/home/grant/Documents/arduino-1.8.5/portable/packages/esp8266/hardware/esp8266/2.4.0/libraries/SPI" "-I/home/grant/Documents/arduino-1.8.5/libraries/Adafruit_SGP30-1.0.0.13" "/tmp/arduino_build_41466/sketch/sonoff.ino.cpp" -o "/tmp/arduino_build_41466/preproc/ctags_target_for_gcc_minus_e.cpp"
sonoff:44: error: #error Please use stable lwIP v1.4
   #error Please use stable lwIP v1.4

```

Frogmore42, 2018-08-08:
Don't use core 2.4.0.  For many people it has caused serious issues.  2.4.1 was supposed to fix them, but didn't really.  2.4.2 is supposed to fix them, but so far there is not any real confirmation.  I use 2.3.0 and have had very good luck with it.  I plan on switching to 2.4.2 as soon as it is available for PlatformIO.

Frogmore42, 2018-08-08:
For 2.3.0 you need to follow the instructions for the loader file, or you won't be able to have the correct memory layout.

grantalewis, 2018-08-08:
I’m not sure where to find that. Do you have a URL?

Frogmore42, 2018-08-09:
https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

Look at the copy files section:
ESP8266 Board version before 2.4.0

grantalewis, 2018-08-09:
Thank you VERY MUCH. In addition to the above instructions, I also found [this discussion](https://github.com/arendst/Sonoff-Tasmota/issues/3035) helpful. I was able to successfully compile 2.3.0. Now I'm ready to try my hand at the flashing process.

OperatorB, 2018-08-09:
Had the same error message with esp8266 core 2.4.2 then I downgraded it to 2.4.1. The error has gone. It compiles. Thank you guys!

consolacion, 2018-08-11:
Can confirm OperatorB.
Had the same error message with esp8266 core 2.4.2 then I downgraded it to 2.4.1. The error has gone. It compiles.  (Windows10)

digiblur, 2018-08-12:
@consolacion - don't use 2.4.0 or 2.4.1, use 2.3.0.  Frogmore42 knows what he is talking about.  

I also recommend, Platform IO, it's pretty nice, it just works how Theo intended it to be compiled.

consolacion, 2018-08-12:
@digiblur Thanks. Maybe I am wrong but I would think if it compiles it compiles. Don' t understand why a compilation in 2.3.0 would be 'better'. But I could be wrong.
Indeed I hear PlatformIO is better, but I think I'll just use the binary :-)

arendst, 2018-08-12:
@consolacion have a look here https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips#20180523---relation-tasmota-and-esp8266arduino-core-version regarding core versions and why you probably need to use 2.3.0....

Frogmore42, 2018-08-12:
Just because something compiles doesn't mean it will actually does what anyone intended or wants.  If that were the case, software development would be really easy 😀.  The fact that it compiles means the compiler understands what was asked of it.  Let's say that the developer wanted to multiply two numbers x and y.  The following will compile:

Answer = x + y;

That, of course, is the wrong answer except for a very few cases of x and y.  So, while the fact that it compiles is necessary, it is not enough.

Theo provided the link to the page that lists some of the issues with 2.4.0 and 2.4.1.  Some people have had luck with them, but I would stick with 2.3.0, at least until 2.4.2 (or later) has been proven to be working better.

consolacion, 2018-08-12:
Will do. Dank je

consolacion, 2018-08-12:
Ah yes you are right. I took a look at Theo's link. I guess I ll stick to
the binaries. I am just happy I got it compiled at all

digiblur, 2018-08-12:
It takes less than 10 seconds to switch your board manager over to 2.30 but then make sure you copy in the linker if needed. 99% there.

jwrw, 2018-08-17:
Hi - I hit this problem doing a build in IDE 1.8.5 against ESP board version 2.4.2.

The reason is in the ESP8266 library docs here [https://arduino-esp8266.readthedocs.io/en/latest/faq/readme.html](url)

Which says...
**Why can’t I use WPS ?**
WPS is disabled by default, this offers an extra 4KB in ram/heap. To enable WPS (and lose 4KB of useable ram), use this boards generator option:
./tools/boards.txt.py --allowWPS --allgen

If you want to do that, then you will need to clone the library yourself from GitHub (as the files pulled in by the IDE board manager don't have the necessary 'package' folder.

This really needs fixing in in Sonoff-Tasmota so that the WPS stuff can be safely disabled for security.

grantalewis, 2018-08-17:
I can't believe the (senseless) agony I've been through over the last two weeks. I'm happy to give more explicit instructions, but the two biggest things I've learned:

1. DON'T buy a knockoff adapter with a fake FTDI chip like [this one](https://smile.amazon.com/gp/product/B00IJXZQ7C/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1).  Spend $6 more and get one with a genuine FTDI chip like [this one](https://smile.amazon.com/gp/product/B075N82CDL/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1). It even comes with header wires. Guaranteed that $6 "saved" will buy you a world of hurt.
2. [This guy has the goods](https://www.youtube.com/watch?v=UDnNI5wkNNY). Watch his video, follow his instructions. Unless you're incredibly lucky with the minefield of compiling your own firmware, it's just not worth the time and effort to use the Arduino software. Using [ESPEasy](https://github.com/letscontrolit/ESPEasy/releases), a [pre-compiled Sonoff.bin](https://github.com/arendst/Sonoff-Tasmota/releases) file, and [Termite](https://www.compuphase.com/software_termite.htm) make the process **_so much simpler_**.

Again, happy to answer questions about the above processes.

digiblur, 2018-08-17:
I use that cheap FTDI and have flashed a ton of devices with no issues. Guess I got lucky.

I would recommend the PlatformIO to build things over ArduinoIDE any day.

grantalewis, 2018-08-17:
@digiblur Go buy a lottery ticket! And while you're out, pick me up a horseshoe and a few four-leaf clovers because I need some of your mojo.

digiblur, 2018-08-17:
Never good at the lottery. But if mine does burn out one day I am going with the Ch340g model adapter I see people recommend all the time.

arendst, 2018-08-17:
@anthonykeane about the original issue:

v6.1.1 does not compile with core version 2.4.2 as you noticed. This is due to removed default functionality in core version 2.4.2 regarding WPS support.

The issue has been solved in development version 6.1.1b

Bottomline: all current released version will fail compile with core 2.4.2 regarding WPS.

Frogmore42, 2018-08-17:
@grantalewis you are the one who got lucky.  The real FTDI is rated to NOT provide enough current to program an esp8266.  As evidenced by your experience it sometimes works.  I always use a separate low voltage power supply for powering a Sonoff.

grantalewis, 2018-08-17:
Really? In the last 24 hours I've flashed 5 Sonoff Basics and one S31, no problems at all.

arendst, 2018-08-17:
Gents, this is an issue regarding WPS mis-compile and not a FTDI facebook page. Pls stop FTDI discussions in this issue.

grantalewis, 2018-09-09:
Not sure if this is related, but yesterday I flashed a new Sonoff Basic using [the same process I mentioned above](https://github.com/arendst/Sonoff-Tasmota/issues/3427#issuecomment-413854164). Using Sonoff-Tasmota v6.2.1 I received the error "Connect failed as AP cannot be reached." I tried flashing with v6.1.1 and didn't encounter any problems.

webtop, 2018-12-10:
> This is the reason that I like PlatformIO. While there is a bit of a learning curve. It and VSCode are actually really nice to use. But, I am a trained professional 😀.
>
> Arduino works great when your [...]

I'm using PIO on Win10 and I've just started to see this error. I'm building v5.14.0 which has built before, many time. The big change on my system right now is the installation of EDP-IDF, new toolchains for ESP32 and 8266, etc.

This makes me assume this is an issue with the "build system" rather than an issue in the code.