arendst/Tasmota #3428: Wemos - Only Inputs needed in Wemos D1

Paulf007, 2018-08-08:
Is it possible to use all the IO ports just for Inputs? I know that there are 4 Switches in the module but if I wanted to add more where do I do that. Also it seems like the Switches will only work if you have a corresponding relay - Switch 1 >>> Relay 1 
Can I only enable Switches without relays.

I want to use the wemos as an alarm controller with all the inputs used for door and motion sensors. 
Best regards
Paul

Frogmore42, 2018-08-08:
There was a recent change to allow 8 inputs as switches, as I recall.  I am pretty sure other people have been successful doing that.  Have you tried it and found it didn't work?

Paulf007, 2018-08-08:
Thank you for the feedback. I upgraded to the latest fm , and did the following 
Changed the SwitchTopic to alarm 
Set the the GPIO ports to the different switches 
As it change the state of the GPOI I am expecting on Switch 1
cmnd/alarm/POWER1 = ON 
cmnd/alarm/POWER1 = OFF

But I am getting , on all Switch inputs , the following 
cmnd/alarm/POWER = ON 
cmnd/alarm/POWER = OFF
If I change the SwitchMode to 3 it changes to 
cmnd/alarm/POWER = TOGGLE

What am I missing?

jsmath, 2018-08-09:
Set "Switch1", "Switch2" etc. for each input in the Module config in the web ui.  You do not need a corresponding relay configured.  You do need to set the SwitchMode for each input you have configured.  Use SwitchMode 1 if you want "ON" to be sent when your input is high.

For example, set the following from the web Console:
SwitchTopic alarm          
SwitchMode1 1
SwitchMode2 1
SwitchMode3 1
SwitchMode4 1
SwitchMode5 1
SwitchMode6 1
SwitchMode7 1
SwitchMode8 1

When the input changes state (high, low), you should see corresponding changes ("ON", "OFF") published to these topics:
cmnd/alarm/POWER1
cmnd/alarm/POWER2
cmnd/alarm/POWER3
cmnd/alarm/POWER4
cmnd/alarm/POWER5
cmnd/alarm/POWER6
cmnd/alarm/POWER7
cmnd/alarm/POWER8

If this doesn't work, you might try to Reset your configuration from the web ui and try to reconfigure from scratch.

Paulf007, 2018-08-13:
Hello @jsmath 

Ok I reset the Wemos D1 and uploaded the latest firmware via the IDE. Here is the module settings
[image]
Next I changed the Switch Topic and Switch modes as described 
[image]
But when I pull the GPIO from high to low and the from low to high I still get the same result on the different inputs.
[image]

jsmath, 2018-08-13:
I seem to remember having an issue like this... try clearing out ButtonTopic by setting it to zero.  If that isn't it, it is probably some other setting... you may need to click the "Reset Configuration" button from the web ui and reconfigure as I stated previously... flashing new firmware doesn't reset config.

nosaj66au, 2018-08-14:
I Agree I used the same config as  jsmath with 4 switches.  The result was the same.

00:08:38 MQT: stat/Switch_106/RESULT = {"SwitchTopic":"alarm"}
00:08:41 MQT: cmnd/alarm/POWER = OFF
00:08:41 MQT: cmnd/alarm/POWER = ON
00:08:43 MQT: cmnd/alarm/POWER = OFF
00:08:43 MQT: cmnd/alarm/POWER = ON
00:08:48 MQT: cmnd/alarm/POWER = OFF
00:08:49 MQT: cmnd/alarm/POWER = ON
00:08:51 MQT: cmnd/alarm/POWER = OFF
00:08:52 MQT: cmnd/alarm/POWER = ON

Only when I add relays did I get 
MQT: cmnd/alarm/POWER1 = OFF
MQT: cmnd/alarm/POWER1 = ON
MQT: cmnd/alarm/POWER2 = OFF
MQT: cmnd/alarm/POWER2 = ON
MQT: cmnd/alarm/POWER3 = OFF
MQT: cmnd/alarm/POWER3 = ON
MQT: cmnd/alarm/POWER4 = OFF
MQT: cmnd/alarm/POWER4 = ON

Jason

jsmath, 2018-08-14:
Ok sorry, in my configuration i have 2 relays and 8 switches.  Maybe you do need at least one relay to make this work?  Seems like a bug to me.

nosaj66au, 2018-08-14:
Yes you are right. Once I added one relay I got the expected result.

00:00:14 MQT: tele/Switch_106/SENSOR = {"Time":"1970-01-01T00:00:14","Switch1":"ON","Switch2":"ON","Switch3":"ON","Switch4":"ON"}
00:00:19 MQT: cmnd/alarm/POWER4 = OFF
00:00:19 MQT: cmnd/alarm/POWER4 = ON
00:00:22 MQT: cmnd/alarm/POWER3 = OFF
00:00:22 MQT: cmnd/alarm/POWER3 = ON
00:00:24 MQT: cmnd/alarm/POWER2 = OFF
00:00:24 MQT: cmnd/alarm/POWER2 = ON
..
..

Paulf007, 2018-08-15:
I can also confirm that once you add a rely it works as expected. This could then be a feature request ?

ascillato2, 2018-08-25:
Hi,

No need for a feature request. It is already implemented what you are requesting.

> Is it possible to use all the IO ports just for Inputs?

Yes, but only up to 8.

> Can I only enable Switches without relays.

Yes, but you need to set switchtopic and switchmode. See wiki at: https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic

Also, you can use rules in order to define the behaviour of the switches you want (send a MQTT message, etc). See rules example in the wiki at: https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#12-controlling-timers-enable-from-a-switch

_(If you think your issue is solved, please close it. Thanks)_

jsmath, 2018-08-25:
@ascillato2 Even after setting the switchtopic and switchmode properly, it is necessary to add at least 1 relay.  Try it without a relay... it doesn't work.  This is annoying since you lose a pin by needing to assign a relay that isn't being used.

ascillato, 2018-08-25:
Hi,

Sorry, I could not reproduce the issue you are explaining.

Using last Tasmota, I configure a nodemcu with just switches without relays.
After that I put in the console as the wiki example states:

```
switchmode1 1
rule 1
rule on Switch1#state=1 do Timers 0 endon on Switch1#state=0 do Timers 1 endon
```

And everytime I press the switch, I receive the command TIMERS 1 and TIMERS 0 when I release the switch.

```
16:39:00 RUL: SWITCH1#STATE=0 performs "Timers 1"
16:39:00 MQT: stat/living/RESULT = {"Timers":"ON"}
16:39:00 MQT: stat/living/RESULT = {"Timers1":{"Timer1":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer2":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer3":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer4":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0}}}
16:39:00 MQT: stat/living/RESULT = {"Timers2":{"Timer5":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer6":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer7":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer8":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0}}}
16:39:00 MQT: stat/living/RESULT = {"Timers3":{"Timer9":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer10":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer11":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer12":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0}}}
16:39:00 MQT: stat/living/RESULT = {"Timers4":{"Timer13":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer14":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer15":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer16":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0}}}
16:39:04 RUL: SWITCH1#STATE=1 performs "Timers 0"
16:39:04 MQT: stat/living/RESULT = {"Timers":"OFF"}
16:39:04 MQT: stat/living/RESULT = {"Timers1":{"Timer1":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer2":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer3":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer4":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0}}}
16:39:04 MQT: stat/living/RESULT = {"Timers2":{"Timer5":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer6":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer7":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer8":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0}}}
16:39:04 MQT: stat/living/RESULT = {"Timers3":{"Timer9":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer10":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer11":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer12":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0}}}
16:39:04 MQT: stat/living/RESULT = {"Timers4":{"Timer13":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer14":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer15":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer16":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0}}}
```

So, please, review your configuration.

Frogmore42, 2018-08-25:
@jsmath is not using rules.  It appears that when using switchtopic you do have to have at least one relay defined and if MQTT goes out, that relay will be controlled by all of the switches.  It is probably not really desirable behavior.  But, it looks like it probably works okay if one does use rules as @ascillato suggested.

jsmath, 2018-08-25:
Exaclry, does not work unless you use rules. Rules should not be required to work around such a normal use case for switches in my opinion. Is this really so hard to fix? Fixed on my config with a few lines of code. And also rules don't currently support sending retained mqtt messages... So not even a workaround for me.

Frogmore42, 2018-08-25:
@jsmath  submit a PR with your solution, that way the people who decide can look at it and see if it fits in with the direction they want.

ascillato, 2018-08-25:
@jsmath 

PR #3593 with command PUBLISH2 for sending retained MQTT messages.

jsmath, 2018-08-25:
@ascillato Good to hear about the PUBLISH2 command!

ascillato, 2018-08-25:
@Paulf007

So, the original topic for this issue was **to use only inputs for sending MQTT messages**.

A solution that works with actual version:

* Inputs configured as switch
* Type in the console:
```
switchmode1 1
switchmode2 1
switchmode3 1
.
.
switchmode8 1
rule 1
rule on Switch1#state=1 do publish cmnd/topic/switch1 1 endon on Switch2#state=1 do publish cmnd/topic/switch2 1 endon ...... on Switch8#state=1 do publish cmnd/topic/switch8 1 endon
```

If you want the mqtt message to be retained by your mqtt broker change `publish` to `publish2`

@jsmath
If you consider it necessary, please open a feature request to modify how Tasmota works related to switchtopic, switchmode and rules, and explain it as detailed as can be, so Theo can consider it. Thanks.

Is there something else related to this issue that needs attention?

ascillato2, 2018-08-27:
Hi,

Closing this issue as there is no feedback. Please, ask to reopen if needed. Thanks.

closed 2018-08-27 by ascillato2