arendst/Tasmota #3429: Pulsetime to turn power off for a while

i3laze, 2018-08-08:
I’d like to implement AlwaysOn scenario with PulseTime to toggle power off for 0.5 sec.

How-to?

Jason2866, 2018-08-08:
Take a look a command pulstime and command backlog in wiki

curzon01, 2018-08-08:
As you did not give more details and did not fill out the template that's all what we have

i3laze, 2018-08-08:
**curzon01,** I'll give backlog a try.. still it's complicated.

Testing "GPIO14: virtual Relay2" now with PulseTime (with Interlock enabled it should cause Relay1 to temporary trigger Off).

i3laze, 2018-08-08:
And here's the answer that I was looking for: (by trial and error)

```
PowerOnState 5
PulseTime x
```

Device is ALWAYS ON, and toggling it Off reverts back to On in PulseTime msec.

closed 2018-08-08 by i3laze

curzon01, 2018-08-09:
Sorry, but for the people who want help on issues it's neccessary to have the issue template info previously.

Did you test it also with current 6.1.1 too? You are using an outdated version 5.14.0 and the Wiki should describe the current version.