arendst/Tasmota #3430: Individual commands for Hue, Saturation and Brightness

titus-leistner, 2018-08-08:
I'm currently working on a HomeKit-MQTT-Bridge. HomeKit sends **different messages for H, S and B**. I can cache the current values for each device and concat them together if a new value occurs, but that is kind of a dirty hack and leads to more problems.

Therefore it would be a nice feature to have **individual commands for Hue, Saturation and Brightness**.

gitolicious, 2018-08-14:
That would also be helpful for my [MagicHome LED controller](https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-LED-strip-controller) which has R, G, B and W values. Tried solving this in [NodeRed](https://nodered.org/) but ended up with a giant mess of conversions in order to link it to [openHab](https://www.openhab.org) items.

kevinshane, 2018-08-25:
yes, please! It would be awesome when adding this type of bulb into homekit!

ascillato, 2018-08-28:
Hi,

With the PR #3615 now has been added individual `HSBCOLOR` commands for better integration to HomeKit and OpenHab.

Usage:

`HSBCOLOR1 <x>` to set Hue
`HSBCOLOR2 <x>` to set Saturation
`HSBCOLOR3 <x>` to set Brightness

Still is supported previous command HSBCOLOR <hue>,<sat>,<bri>

Hope this solves your issue.

Please, test it.

closed 2018-08-29 by ascillato2

kevinshane, 2018-08-29:
Hi, not sure if it's a bug, when I type comand in webconsole
`HsbColor2 50` or `HsbColor3 20` it always reflect to `HsbColor1`, the `2` and `3` donsn't work at all?
```
10:36:52 CMD: HsbColor1 30 //Here I change Hue to 30, it's working
10:36:52 MQT: stat/ailight/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FFA34700","HSBColor":"30,72,100","Channel":[100,64,28,0]}
10:36:52 MQT: stat/ailight/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FFA34700","HSBColor":"30,72,100","Channel":[100,64,28,0]}
10:37:03 CMD: HsbColor3 20 //Here I want to change Bri to 20, but result in Hue changed
10:37:03 MQT: stat/ailight/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF844700","HSBColor":"20,72,100","Channel":[100,52,28,0]}
10:37:03 MQT: stat/ailight/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF844700","HSBColor":"20,72,100","Channel":[100,52,28,0]}
10:37:20 CMD: HsbColor2 50 //Here I want to change Sat to 50, but result in Hue again
10:37:20 MQT: stat/ailight/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FFE04700","HSBColor":"50,72,100","Channel":[100,88,28,0]}
10:37:20 MQT: stat/ailight/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FFE04700","HSBColor":"50,72,100","Channel":[100,88,28,0]}
```

arendst, 2018-08-29:
Investigating...

ascillato, 2018-08-29:
Ups!

Thanks :+1:

ascillato, 2018-08-29:
@kevinshane Can you test is again with latest version? Thanks :+1:

kevinshane, 2018-08-30:
Confirmed the latest version is working like a charm ^^
Thanks @ascillato for the PR and @arendst for the fix