arendst/Tasmota #3431: Sonoff Basic losing I2C connection (BMP280) after a while

bearli97, 2018-08-08:
Hi!

I am using a BM**P**280 connected by I2C with this wiring:
```
BME280-3.3V -> Sonoff-3.3V
BME280-GND -> Sonoff-GND
BME280-SCL -> Sonoff-TX
BME280-SDA -> Sonoff-RX
```

Module configuration:
```
GPIO1 Serial Out -> 05 I2C SCL
GPIO3 Serial In -> 06 I2C SDA
```
After powering the Sonoff it is showing me good sensor readings, but with an error:
```
17:28:36 SHT: Sensor did not ACK command
17:28:37 MQT: tele/sonoff-kulicht/SENSOR = {"Time":"2018-08-08T17:28:37","BMP280":{"Temperature":29.3,"Pressure":1007.8},"TempUnit":"C"}
```
 After a while, however, it delivers temperatures like -138°C:
```
17:38:41 MQT: tele/sonoff-kulicht/SENSOR = {"Time":"2018-08-08T17:38:41","BMP280":{"Temperature":-134.6,"Pressure":1072.0},"TempUnit":"C"}
```
Then, `i2cscan` in the console makes it say:
```
17:59:29 MQT: stat/sonoff-kulicht/RESULT = {"I2CScan":"No devices found"}
```
Tried Sonoff-Tasmota 5.14.0 and 6.1.1 as well. A different module (BM**E**280) results in the same.

Any ideas?

Marvin

[image]

Einstein67, 2018-08-08:
Is "Serial Log" disabled ind Log-Setings?

Frogmore42, 2018-08-08:
I don't have any basic devices, but I have a few TH16 and other esp8266 based ones.  I have several BME280, HTU21 and SHT15.  I also tried a BMP280.  They all work fine.  Do you have a pull-up (or does the sensor have one).  I2C requires a pull-up.  Some sensors include one, but some don't.  I also use GPIO 4 and 14 to avoid the conflict with TX and RX.

andrethomas, 2018-08-09:
The BME280 I2C address will be 0x76 if the SDO pin is grounded or 0x77 if the SDO pin is pulled high. Most likely if not connected it will float so the I2C address may change during usage - Connect SD0 to ground and do i2cscan command, also try to connect to 3.3V and do i2cscan command. If left floating it will change and cause the sensor to appear not responding for Tasmota firmware.

andrethomas, 2018-08-10:
@bearli97 Did you test successfully?

bearli97, 2018-08-17:
@andrethomas I soldered SDO to VCC and the adress was 0x77 like you said. It then did work for almost 24 hours, then back to "No devices found". Now I am trying 0x76 with the same module, if the problem persists I'll try another module.
Thank you so far for your help.

andrethomas, 2018-08-17:
Possibly a faulty chip I guess - I'm not clear on what the CSB pin is for.

Usually, the only reasons these chips fail is because they revert to SPI mode, but having tied the SDO to GND or VCC would have forced it to stay in I2C mode.

andrethomas, 2018-08-17:
Ok, from datasheet

[image]

Connect CSB pin to VCC/3.3V also !

andrethomas, 2018-08-17:
So something new learnt

CSB->GND = SPI mode, CSB->VCC = I2C mode
SDO = I2C address selection only

bearli97, 2018-08-18:
Sometimes in life, all it needs is somebody who can read. It's running smooth since. None of the help articles I found so far did describe the necessity of this setting. Thank you @andrethomas

Frogmore42, 2018-08-18:
The one I have has the proper pull-UPs and pull-DOWNs.
https://www.amazon.com/HiLetgo-Precision-BMP280-3-3-Atmospheric-Pressure/dp/B01ICN5QPU
Take a look at the schematic, last picture.

The picture you provided is the same.  I suspect that you have an intermittent connection/poor solder joint from poor quality control of the reflow or solder paste operations during manufacturing.

Connecting pins (other than power) directly to VCC is generally considered a bad idea by Electrical Engineers, use a pull-up instead.

andrethomas, 2018-08-18:
@Frogmore42 I'd agree with you if it was an IO pin that needed to be externally biased for dual operation and you wanted to bias it high or low by default, but in this particular case it needs to stay to 3V3 or VDDio

[image]

Both are high impedance already so no need to use resistors... in fact, the resistor on CSB may actually cause malfunction if value too high.

andrethomas, 2018-08-18:
@bearli97 If you believe your issue is resolved, please remember to close this issue. Thanks

closed 2018-08-19 by bearli97

curlyel, 2019-07-30:
> Both are high impedance already so no need to use resistors... in fact, the resistor on CSB may actually cause malfunction if value too high.

Many thanks for pointing this out!
Connecting CSB to VDD seems to solve the issue to me too.

BUT: **In addition to it** and to the two 100nF capacitors C1/C2 on board, I've added a bigger capacitance (10µF) between VDD and GND. **This made the pressure readings a lot more stable**.

Before that, pressure reading was _jumping_ around by more than +/- 1 hPa. After adding this capacitor, I got exactly the same readings for many, many seconds and if it changes at all then just by +/- 0.1 hPa.

G-Christ, 2020-07-20:
Hi,
I have this version of BME280 https://www.amazon.de/gp/product/B077PNKCQ6/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&th=1
connected to SonOff basic (V2) : GPIO1 -> I2C SCL (5) and GPIO3 -> I2C SDA (6)
I2CScan returns: {"I2CScan":"Device(s) found at 0x76"} 

But I get no Temperature/humidity/pressure readings at all. And the Tasmota main page don’t even show the placeholders for these values.
If connect a DHT11 sensor instead, GPIO3 -> DHT11(1) I get the readings right away.

What am I missing or doing wrong? 

@andrethomas  any idea?

Hope somebody can help

ascillato, 2020-07-20:
You need to update to Tasmota-sensors.bin where that sensor is supported. 

In Tasmota.bin and Tasmota-lite.bin, that sensor's driver is not included in the firmware.

G-Christ, 2020-07-21:
Hi Adrian,
This is exactly what I was missing.
Thanks a lot for your quick and helpfull response 😊

Med venlig hilsen
Ghassan Christensen