arendst/Tasmota #3438: How can I upload binary to more than 10 devices at a time

upendrawww9, 2018-08-09:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ X] Hardware used :  ESP8266
- [ X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE

**NEW REQUEST**

**How can I upload binary to more than 10  ro 15 devices at a time**

Thanks
Upendra

 
```

**(Please, remember to close the issue when the problem has been addressed)**

upendrawww9, 2018-08-09:
Can i push binary to device?

meingraham, 2018-08-09:
You can perform an automated "bulk" update with TasmoAdmin.

DrZzs Tutorial - www.youtube.com/watch?v=vJUhRyi3-BQ

upendrawww9, 2018-08-09:
Great info

Thanks meingraham

arendst, 2018-08-11:
Or simply use the provided firmware server in the tools folder and let the devices pull the correct firmware using the command ``upload 1`` or ``upload <version number>``.

They retry a few times if the server is overloaded.

upendrawww9, 2018-08-11:
Thanks arendst

 I have checked with upload 1 option it works fine with device. I will update the behavior for bulk.