arendst/Tasmota #3439: Alexa Skill to control Sonoff-Tasmota

luisfpinto, 2018-08-09:
Hello all,
Today the Yeti Alexa Skill has been officially released to everyone. 

This means that if you have Sonoff-Tasmota devices and Yeti installed in your mobile or tablet, you'll be able to control your devices using your voice

[How to connect Yeti to Amazon Alexa
](https://getyeti.co/posts/connecting-yeti-and-amazon-alexa)

Right now only Yeti charms (ambiances) are compatible with Alexa but soon we will be implementing independent device control

I want to say thank you to all of you who particiaped in the Beta, it's been nice to get direct feeback directly from you

Let me know if you have any question, issue and of course feedback!

Frogmore42, 2018-08-09:
So is this cloud to cloud?
What is the expected latency?
I have Alexa talking to SmartThings which is talking to some non Phillips Hue blubs.  That takes several seconds.

I also have Alexa talking Hue protocol locally to HA Bridge, talking MQTT to some Sonoff devices.  Here the latency is under 1 second, which is at the edge of tolerable.

3 seconds is amusingly funny, luckily my wife enjoys the joke too 😀

luisfpinto, 2018-08-10:
Expected latency shouldn't be a lot. I mean communication is Alexa -> Yeti -> Sonoff (Local HTTP request)