arendst/Tasmota #3441: Compiling Tasmota fails.

msiemers, 2018-08-09:
Hi, i have a Problem by compiling a new Tasmota with Platform.io in Visual Studio Code.

I get this Failure:

Compiling .pioenvs\sonoff\src\sonoff.ino.cpp.o
Compiling .pioenvs\sonoff\lib604\IRremoteESP8266-2.2.1.02\ir_Sherwood.cpp.o
Compiling .pioenvs\sonoff\lib604\IRremoteESP8266-2.2.1.02\ir_Sony.cpp.o
Compiling .pioenvs\sonoff\lib604\IRremoteESP8266-2.2.1.02\ir_Trotec.cpp.o
Compiling .pioenvs\sonoff\lib604\IRremoteESP8266-2.2.1.02\ir_Whynter.cpp.o
D:/VSC Working Folder/Sonoff-Tasmota-development/sonoff/settings.ino: In function 'void SettingsDefaultSet2()':
**D:/VSC Working Folder/Sonoff-Tasmota-development/sonoff/settings.ino:472:35: error: 'MQTT_CMND_HOLD' was not declared in this scope**
strlcpy(Settings.state_text[3], MQTT_CMND_HOLD, sizeof(Settings.state_text[3]));
^
Compiling .pioenvs\sonoff\lib38b\ESP8266WiFi\ESP8266WiFi.cpp.o
Compiling .pioenvs\sonoff\lib38b\ESP8266WiFi\ESP8266WiFiAP.cpp.o
*** [.pioenvs\sonoff\src\sonoff.ino.cpp.o] Error 1

Can anybody help me for this?

msiemers, 2018-08-09:
The Problem only exist by Compiling with Platform.io under Visual Studio. Under Atom no Problem.
I have no Idea!

wasi54, 2018-08-09:
Most likely the library storage differs from Atom and platformio. Platformio uses a local (per user) storage in the users home and/or a global storage. Atom usually stores the library in its own directory on the user home.
For Platform just run on command line
pio update
pio upgrade

Then try again to compile

Jason2866, 2018-08-09:
I use VSC without any problems. Why do you use a extra path for VSC?
D:\VSC Working Folder\
I use for Atomio and VSC just one Path to Tasmota. 
D:\Sonoff-Tasmota-development\
I always use the trash symbol before compiling with the other platform.
If i get strange errors that has always helped (in both platforms)!