arendst/Tasmota #3444: Sò

antonioman86, 2018-08-09:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

Frogmore42, 2018-08-09:
If you have to ask the question, you probably don't have the necessary skills and knowledge required to fix it, if it really is broken.

This isn't a forum where you should try and get help on how to use Sonoff devices.  There are some places where there are people who discuss these kinds of things.  Your problem doesn't sound Tasmota related, so any forum about esp8266 based hardware might be fine.

It is possible that your problem occurred while trying to flash Tasmota and that caused it to appear non-functional.  If so, do a little search and see all of the solutions that worked for the others that came before.

It is also possible that you have destroyed your Sonoff Touch.  I have found the Sonoff devices to be reasonably reliable, but I have about a 20% failure rate.  That is not good, but I believe some of them might have gotten damaged during the time my FTDI was outputting 5V on VCC.

So, it certainly is not too hard to destroy one.  Since they are quite cheap, I just use one of my spare ones.  If you don't have any spares, I suggest that you get some.

antonioman86, 2018-08-10:
Thanks a lot..

antonioman86, 2018-08-10:
sorry, I need to reopen...  i reflashed Tasmota and it works....
But now, the problem is this:

- platform: mqtt                               
  name: "Luce bagno"
  state_topic: "stat/luce-bagno/POWER"
  command_topic: "cmnd/luce-bagno/POWER"
  qos: 1
  payload_on: "ON"
  payload_off: "OFF"
  retain: true
  
- platform: mqtt                            
  name: "Lampadario Filippo"
  state_topic: "state/lampadario-filippo/POWER"
  command_topic: "cmnd/lampadario-filippo/POWER"
  qos: 1
  payload_on: "ON"
  payload_off: "OFF"
  retain: true

The first works in home-assistant..
change state from off to on and on to off with light and switch group..
The second not..
What's the problem?
in sonoff configuration topics are correct
also mqtt..

Frogmore42, 2018-08-10:
Since you have provided none of the information required by the template for issues, no one can really know what is the issue.

However, from what you did provide, you are using retain for your messages.  That will cause problems.  I suggest you read and reread the wiki on this:
https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant

You likely have an incorrect configuration.

antonioman86, 2018-08-10:
Thanks again...

closed 2018-08-10 by antonioman86