arendst/Tasmota #3446: Upload stalls at 86%.

phathak, 2018-08-10:
Hi,

I'm attempting to upload the firmware to a Sonoff basic R2 V1.0 using Atom and PlatformIO under High Sierra. I specified the upload port to be `upload_port = /dev/cu.usbserial-A50285BI
` in platformio.ini. The building goes well, then the upload stalls at 86%. As you can see in the screenshot below, the upload has been going on for 11 minutes. 

[image]

Any help is appreciated.
Thank you

phathak, 2018-08-10:
> Try lowering the baud rate. Maybe it is getting some error that it is not detecting properly.

Ok I’ll try that tonight. Thank you

Frogmore42, 2018-08-16:
Since apparently that worked, please close this issue.

phathak, 2018-08-16:
I was able to flash 4 out 5 sonoffs I have by retrying 5-6 times and eventually it would flash. There’s still one that doesn’t want to go through. Changed the upload speed and it didn’t catch any error and would stall the same way.

Thank you

closed 2018-08-16 by phathak