arendst/Tasmota #3448: Upload Pictures BlitzwoIF SHP2

helmar74, 2018-08-10:
Hi,

I'm creating a Wiki Page for the BlitzwoIF SHP2.
Could someone Upload the pictures, so that I can link it?
Thanks a lot!
[image]
[image]
[image]
[image]

curzon01, 2018-08-10:
You already has done this by creating and drop your images within this issue :grin:
- https://user-images.githubusercontent.com/25499089/43945384-3534d26a-9c82-11e8-8a46-95bb2a82ee65.png
- https://user-images.githubusercontent.com/25499089/43945385-354fab6c-9c82-11e8-9d13-d405102a96bd.png
- https://user-images.githubusercontent.com/25499089/43945387-358d35c2-9c82-11e8-8031-27754090e50c.jpg

helmar74, 2018-08-10:
Haha! Thanks a lot.
I didn't know ;-)

closed 2018-08-10 by helmar74