arendst/Tasmota #3449: Possible add thermistor as temperature sensor

Fires04, 2018-08-10:
**Have you look for this feature in other issues and in the wiki?**
Yes.

**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Add thermistor to Sonoff basic. To use sonoff basic as floor heater controler. It is possible to add thermistor to read floor temperature ? 

**Describe alternatives you've considered**
Use Wedos D1 mini and custom firmware. 

**Additional context**

Jason2866, 2018-08-10:
Why not connect a BMP280 or a DS18B20 (waterproof) temperatur sensor?
Take a look in the wiki. Under rules you find a example for a Thermostat

Jason2866, 2018-08-10:
You can monitor floor temperature too with the suggestions i made.

Fires04, 2018-08-10:
Yes but I want to use current thermistor what is in concrete. No add any
new.. I can't find any connection diagram in wiki .. I probably need use
Analog pin in ESP and create voltage divider to read voltage 0-1V from
thermistor.

Frogmore42, 2018-08-10:
I would bet that you will get very inconsistent results if you use the internal ADC.  I use an external ADS1115 and an external 3V3 regulator.  I use one channel to measure the 3V3 and then a different channel to measure the output from a resistor voltage divider sensor.  This works well.  Prior to doing that, I tried running it off of the internal 3V3 and even the 5V in, but I noticed that the current draw of the esp8266 was enough to impact the measurement significantly.

You will need to do some testing to determine the value for the other resistor.  It depends on the accuracy and repeatability you desire/need.  If you only care if it is really, really, really hot or not, the internal ADC could do that, but I would expect that the repeatability might be off by 10 degrees or even more.

Frogmore42, 2018-08-11:
This site has theory on how to connect the thermistor 
https://www.petervis.com/electronics%20guides/calculators/thermistor/thermistor.html

Vin of those diagrams goes to the output of the 3V3 regulator and ADC input 0 (as a reference to make the math easier/more accurate)

Vout of those diagrams goes to the ADC input 1

the bottom line is ground

the values you need for the resistor will vary depending on the temperatures you need and the thermistor you have in the floor already.  If you don't care much about accuracy, you could use the 3V3 from the Sonoff and the built in ADC, just make sure you size the other resistor to limit the voltage to the ADC to a safe (for the input) value.

Frogmore42, 2018-08-13:
If this answers your question, please close the issue.

closed 2018-08-13 by Fires04