arendst/Tasmota #3451: OTA File not found ?

laurentand, 2018-08-10:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ]X Searched the problem in the forum (https://forum.digital-pig.com/)
- [ X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ X] Hardware used : Sonoff S20 (but any sonoff is failing) 
- [X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE

Hi, 
here's the debug window of a MQTT

10/08/2018 à 16:55:57node: 7236a223.f0961c
stat/sonoffp1/RESULT : msg.payload : string[71]
"{"Upgrade":"Version 5.14.0 from http://192.168.0.9/tasmota/sonoff.bin"}"
10/08/2018 à 16:56:09node: 7236a223.f0961c
stat/sonoffp1/UPGRADE : msg.payload : string[27]
"Failed File not Found (404)"

But the file exists ! and i can download it in my browser. 
The WS is running on a synology on the LAN. 
The bin file is working correctly and was successfully uploaded using serial connexion on 5+ devices POW, S20, T1 , Basic) 

![sonoff-bin](https://user-images.githubusercontent.com/15147663/43965526-c93de310-9cbf-11e8-995f-2d6536526921.jpg)

![sonoff2](https://user-images.githubusercontent.com/15147663/43965772-6ddd6a80-9cc0-11e8-8e29-58245800b3af.jpg)

**(Please, remember to close the issue when the problem has been addressed)**

laurentand, 2018-08-10:
I tried something else:
- Modified the user_config.h to have the correct url. (previously was default value) 
- Upgraded to a minimal build using the file upload option : worked. 
- tried OTA again : Worked ? 

What was failing ? the url used is the one builded and not the one we see in the text box ?

closed 2018-08-11 by laurentand

crispy78, 2020-01-30:
Hi there,

I've had some issues while upgrading from 8.1.0.3 to 8.1.0.5.
It gives a 404 error, while the file is there to download.
I've already done 20 or so devices manually, but OTA would be nice :-)

11:04:52 MQT: 59/stat/RESULT = {"OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin"}
11:04:55 CMD: upgrade 1
11:04:55 MQT: 59/stat/RESULT = {"Upgrade":"Version 8.1.0.3 from http://thehackbox.org/tasmota/tasmota.bin"}
11:04:58 HTP: Web server stopped
11:04:59 HTP: Web server active on lavalamp-woonkamer with IP address 192.168.178.59
11:05:00 HTP: Web server stopped
11:05:01 HTP: Web server active on lavalamp-woonkamer with IP address 192.168.178.59
11:05:02 HTP: Web server stopped
11:05:03 HTP: Web server active on lavalamp-woonkamer with IP address 192.168.178.59
11:05:04 HTP: Web server stopped
11:05:05 HTP: Web server active on lavalamp-woonkamer with IP address 192.168.178.59
11:05:06 HTP: Web server stopped
11:05:07 HTP: Web server active on lavalamp-woonkamer with IP address 192.168.178.59
11:05:09 MQT: 59/stat/UPGRADE = {"Upgrade":"Failed File Not Found (404)"}

stuermera, 2020-03-21:
Same problem here with 16 devices and Tasmota 8.1.0.3. File not found.

une18, 2020-04-21:
I can not update my devices, how to solve? Manually uploading gives error "Upload Failed Upload buffer miscompare"

ascillato, 2020-04-21:
Use the file MINIMAL first to free space in flash.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)