arendst/Tasmota #3452: Zigbee with Ta

L00000813, 2018-08-10:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

L00000813, 2018-08-10:
I currently have Tasmota installed on a number of Sonoff devices communicating with a home IoT system hosted on a Raspberry Pi. I am looking to integrate some Zigbee sensors into my network as they are wireless, and have low battery consumption. Mainly temperature sensors around the home.

**I basically want to get the Zigbee temperature readings on the PI** and then I will control my Sonoff's accordingly.

I understand that WIFi and Zigbee are two different protocols. Has anyone done something similar - any advice on how best to go about achieving this.

Jason2866, 2018-08-10:
You can use a Home Automation which "speaks" mqtt and zigbee.
For example you can use your existing RasPi to install the HA system OpenHab which support both.
You can find help in the OpenHab Community 
https://community.openhab.org

mihalski, 2018-08-13:
Home Assistant ( https://www.home-assistant.io/ ) is a great solution. Many people are using it in this manner and much MUCH more.

fabiosoft, 2019-07-19:
I think you need external hardware (like an usb dongle) to read zigbee radio frequency...than translate to mqtt should be easy with tool like zigbee2mqtt