arendst/Tasmota #3453: Translated values for Payload_available and Payload_unavailable

dzieciakowy, 2018-08-10:
_A clear and concise description of what the bug is._

I am using Sonoff-Tasmota 6.1.1 compiled with MY_LANGUAGE set to pl-PL
This causes Sonoff to send some mqtt messages translated to polish (translated values marked in bold font):
00:00:07 MQT: homeass/switch/sonoff_1/config = {"name":"sonoff_plug_01","command_topic":"cmnd/sonoff/POWER","state_topic":"stat/sonoff/RESULT","value_template":"{{value_json.POWER}}","payload_off":"OFF","payload_on":"ON","availability_topic":"tele/sonoff/LWT","**payload_available":Aktywny**","**payload_not_available":"Nieaktywny"**} (Zachowane)

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues and in the wiki
- [Sonoff S20] Hardware used : 
- [x] Provide the output of command ``status 0`` :
```
00:01:31 CMD: status 0
00:01:31 MQT: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["sonoff_plug_01"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:01:31 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:01:30","StartupUTC":"","Sleep":0,"BootCount":22,"SaveCount":129,"SaveAddress":"FB000"}}
00:01:31 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-08-03T20:45:46","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:01:31 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NVTL",""],"TelePeriod":300,"SetOption":["000A8009","55818000"]}}
00:01:31 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":538,"Free":464,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000415","0FDAE794","0C000000","23B6179E","00000000"]}}
00:01:31 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6466","IPAddress":"10.18.11.37","Gateway":"10.18.11.1","Subnetmask":"255.255.255.0","DNSServer":"10.18.11.1","Mac":"EC:FA:BC:BC:D9:42","Webserver":2,"WifiConfig":2}}
00:01:31 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"10.15.21.35","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_86D942","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:01:31 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:01:31 1970","Local":"Thu Jan 01 00:01:31 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:01:31 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:01:31"}}
00:01:31 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:01:31","Uptime":"0T00:01:30","Vcc":3.152,"POWER":"ON","Wifi":{"AP":1,"SSId":"NNNN","RSSI":86,"APMac":"00:72:43:1F:47:A4"}}}
```

**To Reproduce**
_Steps to reproduce the behavior:_
Set define MY_LANGUAGE to pl-PL in user_config.h and build tasmota

**Expected behavior**
_A clear and concise description of what you expected to happen._
Locale setting should not affect message values. I expect Payload_Available to return "Active" and not "Aktywny"

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2018-08-11:
From HomeAssistant MQTT docs:

> The mqtt binary sensor platform optionally supports an availability_topic to receive online and offline messages (birth and LWT messages) from the MQTT device. During normal operation, if the MQTT cover device goes offline (i.e., publishes payload_not_available to availability_topic), Home Assistant will display the binary sensor as unavailable. If these messages are published with the retain flag set, the binary sensor will receive an instant update after subscription and Home Assistant will display the correct availability state of the binary sensor when Home Assistant starts up. If the retain flag is not set, Home Assistant will display the binary sensor as unavailable when Home Assistant starts up. If no availability_topic is defined, Home Assistant will consider the MQTT device to be available.

So all Tasmota does here is tell HAss what the LWT messages are for offline and online.

dzieciakowy, 2018-08-12:
So i should be able to reconfigure HAss to read this configuration from this message? Right now i have it set up in the configuration.yaml so whenever i change tasmota locale, i have to modify HAss config.
Thanks, i'll look it up. 

Nevertheless it seems like an inconsistency in tasmota firmware - payload availability it is the only configuration setting that is affected by locale - ex. "Payload" uses "ON" or "OFF" in both languages.

curzon01, 2018-08-12:
payload uses StateText1...StateText4, see [MQTT Commands](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#mqtt-specific) StateText&lt;x&gt;

ascillato2, 2018-08-18:
Hi,

Closing this issue as your questions has been addressed. Ask to reopen this issue if you need it. Thanks

closed 2018-08-18 by ascillato2