arendst/Tasmota #3455: Sonoff 4Ch restarts when switching load

r3ndszergazda, 2018-08-10:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : **Sonoff 4CH Module**
- [x] Provide the output of command ``status 0`` :
```
22:33:57 CMD: status 0
22:33:57 SRC: WebConsole from 0.0.0.0
22:33:57 RSL: Received Topic /status, Data Size 1, Data 0
22:33:57 RSL: Group 0, Index 1, Command STATUS, Data 0
22:33:57 RSL: stat/sonoff/STATUS = {"Status":{"Module":7,"FriendlyName":["Sonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"sonoff","ButtonTopic":"0","Power":8,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
22:33:57 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:01:14","StartupUTC":"2018-08-10T21:32:43","Sleep":0,"BootCount":44,"SaveCount":145,"SaveAddress":"FB000"}}
22:33:57 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-31T17:59:09","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:33:57 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"XXX","LogPort":514,"SSId":["BBB","CCC"],"TelePeriod":10,"SetOption":["00008009","55818000"]}}
22:33:57 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":536,"Free":464,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","0C000000","03B6179E","00000000"]}}
22:33:57 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1300","IPAddress":"YYY","Gateway":"YYZ","Subnetmask":"255.255.255.0","DNSServer":"YYZ","Mac":"60:01:94:89:05:14","Webserver":2,"WifiConfig":2}}
22:33:57 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"AAA","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_890514","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:33:57 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Fri Aug 10 21:33:57 2018","Local":"Fri Aug 10 22:33:57 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:36","Sunset":"20:15"}}
22:33:57 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-08-10T22:33:57"}}
22:33:57 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-08-10T22:33:57","Uptime":"0T00:01:14","Vcc":3.156,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","Wifi":{"AP":1,"SSId":"BBB","RSSI":86,"APMac":"A0:AB:1B:42:1B:A1"}}}
```
Dear All!

I had this unit for a while. After I've bought it I've flashed Tasmota on it, and then I've put it away. Today I thought I would play again with it & loaded the latest FW ver via serial*.

The issue is, that no matter which channel I use (1,2,3,4), if there is load on the channel (let's say a lamp, a mobile charger, so anything ranging from a couple watts to a hundred) the box reboots after switching. It happens with the physical switch buttons, through the web ui, and via mqtt as well. If I pull the loads out of it: everything is fine. Switches well, doesn't reboot.

I guess it's some sort of an HW issue, but still hope there is a solution to it:)

Thanks in advance for any help!
Regards

*Problem was present with older Tasmota version as well

andrethomas, 2018-08-10:
Most likely the 3v3 regulator is defective - another possibility is the 5V or 3V3 stabilising capacitors - I do not have one of these so do not know what the PCB looks like inside... just some thoughts.

andrethomas, 2018-08-10:
[image]

The marked part is the 5V -> 3V3 regulator I believe - it should have 5V on two of the pins and 3V3 on two of the others - if not then faulty... could be it has partially failed and causes the 3V3 rail to drop below operational voltage of esp module when 5V rail is loaded with having to drive a relay.

ascillato2, 2018-08-18:
Hi,

Have you managed to solve your issue?

r3ndszergazda, 2018-08-18:
Yes: I've bought another one:) Tried swapping the regulator but that didn't make a difference. It's probably the ESP that's bad.

closed 2018-08-18 by r3ndszergazda

jojkos, 2018-10-03:
So buying another one helped? I have exactly the same issue and don't know what to do with it.

r3ndszergazda, 2018-10-03:
> So buying another one helped? I have exactly the same issue and don't know what to do with it.

Yes, that was the solution. I think I've fried the old one partially when I've connected it to 5V logic.