arendst/Tasmota #3456: Sonoff POW R2 shows 0 values while heavy load is running (Heater 2kw)

laurentand, 2018-08-11:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
09:44:53 MQT: stat/sonoffpow1/STATUS = {"Status":{"Module":43,"FriendlyName":["Sonoffpow1"],"Topic":"sonoffpow1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
09:44:53 MQT: stat/sonoffpow1/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://xxxxxxxxx/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T13:47:35","StartupUTC":"2018-08-10T17:57:18","Sleep":0,"BootCount":18,"SaveCount":96,"SaveAddress":"F7000"}}
09:44:53 MQT: stat/sonoffpow1/STATUS2 = {"StatusFWR":{"Version":"6.1.1c","BuildDateTime":"2018-08-10T19:13:18","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
09:44:54 MQT: stat/sonoffpow1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"xxxxxxxxx","LogPort":514,"SSId":["xxxxxxxx",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000001"]}}
09:44:54 MQT: stat/sonoffpow1/STATUS4 = {"StatusMEM":{"ProgramSize":492,"Free":508,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["0000040C","0FCAE794","00000000","B7B6179E","00000000"]}}
09:44:54 MQT: stat/sonoffpow1/STATUS5 = {"StatusNET":{"Hostname":"sonoffpow1-7579","IPAddress":"xxxxxx","Gateway":"xxxxxxx","Subnetmask":"255.255.255.0","DNSServer":"xxxxxxx","Mac":"xxxxxxxxxxxx","Webserver":2,"WifiConfig":2}}
09:44:54 MQT: stat/sonoffpow1/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxxx","MqttPort":1883,"MqttClientMask":"xxxxxx","MqttClient":"xxxxxxx","MqttUser":"xxxxxxx","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:44:54 MQT: stat/sonoffpow1/STATUS7 = {"StatusTIM":{"UTC":"Sat Aug 11 07:44:54 2018","Local":"Sat Aug 11 09:44:54 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"06:37","Sunset":"21:13"}}
09:44:54 MQT: stat/sonoffpow1/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
09:44:54 MQT: stat/sonoffpow1/STATUS10 = {"StatusSNS":{"Time":"2018-08-11T09:44:54","ENERGY":{"Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
09:44:54 MQT: stat/sonoffpow1/STATUS11 = {"StatusSTS":{"Time":"2018-08-11T09:44:54","Uptime":"0T13:47:36","Vcc":3.458,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxxxx","RSSI":60,"APMac":"xxxxxxxxxxxxx"}}}

``` 
Sonoff POW R2 shows  0V and 0A while a 2Kw+ heater is working. 
I checked the input / output  live neutral  wiring and verified (AC to ground) that the live is the real live. 

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2018-08-11:
Is it a real R2? Try the other pow module option pls.

laurentand, 2018-08-11:
As i only changes user settings to have default wifi / mqtt , would this make any change ?

laurentand, 2018-08-11:
@curzon01 as i juste change wifi / and Mqtt settings to have them work "by default" i'm not sure that would make a difference ?

@arendst  Yes, it's a sonoff R2

As these are picture from an unused, i'll do some testing with this one to see if it's behaving the same way with
- my compiled bin
- the default one. 

[image]
[image]

arendst, 2018-08-11:
Alas, it's a Sonoff Pow v2.0 NOT a R2.

Quick check: change module to Sonoff Pow et voila, it works. Pls close this issue if correct.

laurentand, 2018-08-11:
Ok, did it and it's working. thanks. 
As the wiring was like the R2 and it was written V2 on the PCB i assumed that was a R2.

closed 2018-08-11 by laurentand