arendst/Tasmota #3458: Sonoff Pow after Flash only Red LED works

Heartbeatss1, 2018-08-11:
Hello Guys,
i hope you can help me.
I have two Pows here and i tried both to flash them with the newest Tasmota Software.
But after Flashing them, they didnt work correctly. Only the red led lights up and no Wifi connection.

I flashed it on Win10 and with the Atom software.
Does anyone else hase this Problems?

Jason2866, 2018-08-11:
Did you dowmload the precompiled version from here?
Please give us some more infos, without more infos we cant help

Heartbeatss1, 2018-08-11:
Hey Jason,
i downloaded the completed zip from here and and compile it with atom
newest version. 
I tried it with sonoff-minimal/Full/normale etc.
and with every core version

Heartbeatss1, 2018-08-11:
this is my platformio file
`; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter, extra scripting
;   Upload options: custom port, speed and extra flags
;   Library options: dependencies, extra library storages
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/en/stable/projectconf.html

[platformio]
src_dir = sonoff

; *** Uncomment one of the lines below to build/upload only one environment
;env_default = sonoff
env_default = sonoff-minimal
;env_default = sonoff-classic
;env_default = sonoff-knx
;env_default = sonoff-allsensors
;env_default = sonoff-BG
;env_default = sonoff-BR
;env_default = sonoff-CN
;env_default = sonoff-CZ
;env_default = sonoff-DE
;env_default = sonoff-ES
;env_default = sonoff-FR
;env_default = sonoff-GR
;env_default = sonoff-HU
;env_default = sonoff-IT
;env_default = sonoff-NL
;env_default = sonoff-PL
;env_default = sonoff-PT
;env_default = sonoff-RU
;env_default = sonoff-UK
;env_default = sonoff-TW

[common] ; ************************************************************
; *** Esp8266 core for Arduino version 2.3.0
platform = espressif8266@1.5.0
; *** Esp8266 core for Arduino version 2.4.0
;platform = espressif8266@1.6.0
; *** Esp8266 core for Arduino version 2.4.1
;platform = espressif8266@1.7.3
; *** Esp8266 core for Arduino current version
platform = espressif8266

framework = arduino
board = esp01_1m
board_build.flash_mode = dout

; *** Fix espressif8266@1.7.0 induced undesired all warnings
build_unflags = -Wall

build_flags =
  -Wl,-Tesp8266.flash.1m0.ld
  -DVTABLES_IN_FLASH
  -DPIO_FRAMEWORK_ARDUINO_LWIP_HIGHER_BANDWIDTH
;  -DUSE_CONFIG_OVERRIDE

; *** Fix Esp/Arduino core 2.4.x induced Tasmota unused floating point includes
;extra_scripts = pio/strip-floats.py

; *** Serial Monitor options
monitor_speed = 115200

; *** Select ONE from below THREE upload options.

; *** OPTION1 - Upload Serial reset method for Wemos and NodeMCU
upload_speed = 115200
upload_resetmethod = nodemcu
[... 242 lines cut ...]

[env:sonoff-UK]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board}
board_build.flash_mode = ${common.board_build.flash_mode}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags} -DMY_LANGUAGE=uk-UK
monitor_speed = ${common.monitor_speed}
upload_speed = ${common.upload_speed}
upload_port = ${common.upload_port}
extra_scripts = ${common.extra_scripts}

[env:sonoff-TW]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board}
board_build.flash_mode = ${common.board_build.flash_mode}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags} -DMY_LANGUAGE=zh-TW
monitor_speed = ${common.monitor_speed}
upload_speed = ${common.upload_speed}
upload_port = ${common.upload_port}
extra_scripts = ${common.extra_scripts}
`

Heartbeatss1, 2018-08-12:
okay, now i know what you mean.
I tried it today with the newest version 6.1.1 and the 5.14., but at booth the same behavior

curzon01, 2018-08-12:
When you flash it, you are using an USB adapter (which has a COM port under your Win10) to serial flash your Sonoff. Ok.

After you have flashed your Sonoff, reconnect your enviroment same like you want to flash (this restarts your Sonoff) but do not flash at, open a terminal program (Termite, putty whatever) to visibile the serial output of this COM port using 115200 baud.

Then you will hopefully see more what's happening with your firmware on your Sonoff.

closed 2018-08-22 by ascillato2