arendst/Tasmota #3460: Wrong object name in Mqtt payload.

centralna, 2018-08-12:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

centralna, 2018-08-12:
I have issue with object name in Mqtt payload. DS18B20-1 is not OK for node red extract Function. How I change to DS18B201.

var value = msg.payload.BS18B20-1.Temperature;
msg.payload = value;
return msg;

Unexpected "1"

Kind Regards

curzon01, 2018-08-12:
try 
`var value = msg.payload['BS18B20-1']['Temperature'];`

fluppie, 2018-08-12:
Maybe you have the same problem as this guy? https://github.com/arendst/Sonoff-Tasmota/issues/3268
Looks like a wrongly formatted JSON string.

Frogmore42, 2018-08-12:
JavaScript doesn't like dashes in the name, since it can't tell the difference between a dash and a minus sign.  Since they use the same ASCII character, it has to tell by context.  Since, apparently, JS doesn't require whitespace, a name with a dash in it looks like you want to subtract the two named values.  This, of course, is not what you want.  So, you have to either, not put dashes in variable names, or use the alternative syntax for object specification that makes it easy for JS to understand what you really meant.

centralna, 2018-08-13:
Thanks for reply. This mqtt bug message I getting from tasmota 6.1.0. I like to know, where in tasmota file, is posible to changed this wrong minus sign.
Regards

arendst, 2018-08-13:
You should not need to change Tasmota. 

Search the issues for a much better solution as many users already stumbled across your problem providing a simple local solution

centralna, 2018-08-13:
Thanks for help. Working.

var temp=msg.payload['DS18B20-1'].Temperature;

var msg = {

payload:temp

};
return msg;

closed 2018-08-13 by centralna