arendst/Tasmota #3461: MQTT problem

rt400, 2018-08-12:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [X ] Hardware used : 
- [ X] Provide the output of command ``status 0`` :
```
12:35:47 MQT: stat/Kitchen/STATUS = {"Status":{"Module":30,"FriendlyName":["Kitchen","Hall","Salon"],"Topic":"Kitchen","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":8,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
12:35:47 MQT: stat/Kitchen/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.0.254:9999/data/firmwares/sonoff.bin","RestartReason":"Power on","Uptime":"0T03:30:23","StartupUTC":"2018-08-12T06:05:24","Sleep":0,"BootCount":226,"SaveCount":1147,"SaveAddress":"F4000"}}
12:35:47 MQT: stat/Kitchen/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-23T11:18:24","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:35:47 MQT: stat/Kitchen/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"","LogPort":514,"SSId":["Mejahez",""],"TelePeriod":300,"SetOption":["0000A029","55818000"]}}
12:35:47 MQT: stat/Kitchen/STATUS4 = {"StatusMEM":{"ProgramSize":537,"Free":464,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","0C000000","23B6179E","00000000"]}}
12:35:47 MQT: stat/Kitchen/STATUS5 = {"StatusNET":{"Hostname":"Kitchen","IPAddress":"192.168.0.70","Gateway":"192.168.0.138","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.254","Mac":"60:01:94:97:22:B0","Webserver":2,"WifiConfig":5}}
12:35:47 MQT: stat/Kitchen/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.254","MqttPort":1883,"MqttClientMask":"KitchenLight","MqttClient":"KitchenLight","MqttUser":"homeassistant","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:35:47 MQT: stat/Kitchen/STATUS7 = {"StatusTIM":{"UTC":"Sun Aug 12 09:35:47 2018","Local":"Sun Aug 12 12:35:47 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":3,"Sunrise":"07:38","Sunset":"22:11"}}
12:35:47 MQT: stat/Kitchen/STATUS10 = {"StatusSNS":{"Time":"2018-08-12T12:35:47"}}
12:35:47 MQT: stat/Kitchen/STATUS11 = {"StatusSTS":{"Time":"2018-08-12T12:35:47","Uptime":"0T03:30:23","Vcc":3.116,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"Mejahez","RSSI":68,"APMac":"80:2A:A8:54:8A:49"}}}
12:35:59 MQT: stat/Kitchen/RESULT = {"POWER1":"ON"}
```

**(Please, remember to close the issue when the problem has been addressed)**
hi .
i have a HA system with mosquitto broker on it. running on Ubuntu server 18 .
all my device with TASMOTA 6.1.1v firmware.
now , where i power off all the devices and turn them on , some of them become ON Although they were turned off before the shutdown, and reversed.
i try for ling time to figure the problem. but i cant found nothing..
any help will be useful for me.
best regards..

Jason2866, 2018-08-12:
Your HA / broker has stored retained messages. You have to delete this.
This is not a Tasmota issue!
Dont use function retain messages in your HA system, it makes more problems than it solves

rt400, 2018-08-12:
thanks..
i did it...and its looks that is work OK now..
but why we use retain flag?

Frogmore42, 2018-08-13:
Here is a good tutorial on retained messages.
https://www.hivemq.com/blog/mqtt-essentials-part-8-retained-messages

Here is the topic for home Assistant
https://www.home-assistant.io/components/switch.mqtt/

They appear to think retain is a good idea.  It might be if devices didn't remember their state and all control messages were sent with retain, so it really was the last state.  Since Tasmota devices remember their last state and it appears the retained message is not always correct, using retained messages to control Tasmota devices probably causes more issues than it helps.

Frogmore42, 2018-08-16:
Please close this issue, since it has been resolved.

closed 2018-08-18 by ascillato2