arendst/Tasmota #3462: OTA Update from 6.1.1 Minimal to 6.1.1 doesnt complete

SurfingSpider, 2018-08-12:
Can anybody help please? I am trying to update OTA. I flashed minimal but now it wont flash the full version. I have no idea why? in the console it does say "RestartReason":"Blocked Loop"

Here is a section of the console readout.
```

10:39:16 MQT: stat/sonoff7/STATUS = {"Status":{"Module":8,"FriendlyName":["Sonoff7 - Fireplace"],"Topic":"sonoff7","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
10:39:16 MQT: stat/sonoff7/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"https://github.com/arendst/Sonoff-Tasmota/releases/download/v6.1.1/sonoff-minimal.bin","RestartReason":"Blocked Loop","Uptime":"0T00:09:15","StartupUTC":"2018-08-12T09:30:01","Sleep":0,"BootCount":45,"SaveCount":911,"SaveAddress":"FB000"}}
10:39:16 MQT: stat/sonoff7/STATUS2 = {"StatusFWR":{"Version":"6.1.1-minimal","BuildDateTime":"2018-07-31T09:40:44","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
10:39:16 MQT: stat/sonoff7/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["Pretty Fly For A WiFi",""],"TelePeriod":300,"SetOption":["00000029","55818000"]}}
10:39:16 MQT: stat/sonoff7/STATUS4 = {"StatusMEM":{"ProgramSize":406,"Free":596,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","00082580","0C000002","21021702","00000000"]}}
10:39:16 MQT: stat/sonoff7/STATUS5 = {"StatusNET":{"Hostname":"sonoff7-4489","IPAddress":"192.168.0.61","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"EC:FA:BC:8A:B1:89","Webserver":2,"WifiConfig":2}}
10:39:16 MQT: stat/sonoff7/STATUS6 = {"StatusMQT":{"MqttHost":"hassio","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8AB189","MqttUser":"homeassistant","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:39:16 MQT: stat/sonoff7/STATUS7 = {"StatusTIM":{"UTC":"Sun Aug 12 09:39:16 2018","Local":"Sun Aug 12 10:39:16 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
10:39:16 MQT: stat/sonoff7/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
10:39:16 MQT: stat/sonoff7/STATUS10 = {"StatusSNS":{"Time":"2018-08-12T10:39:16"}}
10:39:16 MQT: stat/sonoff7/STATUS11 = {"StatusSTS":{"Time":"2018-08-12T10:39:16","Uptime":"0T00:09:15","Vcc":3.208,"POWER":"ON","Wifi":{"AP":1,"SSId":"Pretty Fly For A WiFi","RSSI":50,"APMac":"C0:3E:0F:39:B6:BD"}}}
```

Jason2866, 2018-08-12:
OTA upgrade and file upgrade via webconsole are 2 different things.
If you have your firmware file locally use file upgrade method.

[image]

SurfingSpider, 2018-08-12:
Thank you for responding Jason 2866.

I have tried both methods. When I upload the sonoff.bin file I get....

```
00:00:00 Project sonoff Sonoff7 - Fireplace (Topic sonoff7, Fallback DVES_8AB189, GroupTopic sonoffs) Version 6.1.1-minimal-2_3_0
00:00:00 WARNING This version does not support persistent settings
00:00:00 WIF: Connecting to AP1 Pretty Fly For A WiFi in mode 11N as sonoff7-4489...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff7-4489 with IP address 192.168.0.61
00:00:04 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/sonoff7/LWT = Online (retained)
00:00:05 MQT: cmnd/sonoff7/POWER = 
00:00:05 MQT: stat/sonoff7/RESULT = {"POWER":"ON"}
00:00:05 MQT: stat/sonoff7/POWER = ON (retained)
00:00:05 MQT: tele/sonoff7/INFO1 = {"Module":"Sonoff S2X","Version":"6.1.1-minimal","FallbackTopic":"DVES_8AB189","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/sonoff7/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff7-4489","IPAddress":"192.168.0.61"}
00:00:05 MQT: tele/sonoff7/INFO3 = {"RestartReason":"Blocked Loop"}
00:00:06 MQT: stat/sonoff7/RESULT = {"POWER":"ON"}
00:00:06 MQT: stat/sonoff7/POWER = ON (retained)
21:20:11 MQT: tele/sonoff7/STATE = {"Time":"2018-08-12T21:20:11","Uptime":"0T00:00:13","Vcc":3.208,"POWER":"ON","Wifi":{"AP":1,"SSId":"Pretty Fly For A WiFi","RSSI":56,"APMac":"C0:3E:0F:39:B6:BD"}}
21:25:25 MQT: tele/sonoff7/STATE = {"Time":"2018-08-12T21:25:25","Uptime":"0T00:05:27","Vcc":3.208,"POWER":"ON","Wifi":{"AP":1,"SSId":"Pretty Fly For A WiFi","RSSI":54,"APMac":"C0:3E:0F:39:B6:BD"}}
21:30:30 APP: Serial logging disabled
21:30:46 MQT: tele/sonoff7/STATE = {"Time":"2018-08-12T21:30:46","Uptime":"0T00:10:48","Vcc":3.208,"POWER":"ON","Wifi":{"AP":1,"SSId":"Pretty Fly For A WiFi","RSSI":56,"APMac":"C0:3E:0F:39:B6:BD"}}
21:36:23 MQT: tele/sonoff7/STATE = {"Time":"2018-08-12T21:36:23","Uptime":"0T00:16:25","Vcc":3.207,"POWER":"ON","Wifi":{"AP":1,"SSId":"Pretty Fly For A WiFi","RSSI":56,"APMac":"C0:3E:0F:39:B6:BD"}}
21:41:31 MQT: tele/sonoff7/STATE = {"Time":"2018-08-12T21:41:31","Uptime":"0T00:21:33","Vcc":3.208,"POWER":"ON","Wifi":{"AP":1,"SSId":"Pretty Fly For A WiFi","RSSI":56,"APMac":"C0:3E:0F:39:B6:BD"}}
``` 

When I do a OTA update before it restarts I get ....

```
00:00:00 Project sonoff Sonoff7 - Fireplace (Topic sonoff7, Fallback DVES_8AB189, GroupTopic sonoffs) Version 6.1.1-minimal-2_3_0
00:00:00 WARNING This version does not support persistent settings
00:00:00 WIF: Connecting to AP1 Pretty Fly For A WiFi in mode 11N as sonoff7-4489...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on sonoff7-4489 with IP address 192.168.0.61
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/sonoff7/LWT = Online (retained)
00:00:05 MQT: cmnd/sonoff7/POWER = 
00:00:05 MQT: stat/sonoff7/RESULT = {"POWER":"ON"}
00:00:05 MQT: stat/sonoff7/POWER = ON (retained)
00:00:05 MQT: tele/sonoff7/INFO1 = {"Module":"Sonoff S2X","Version":"6.1.1-minimal","FallbackTopic":"DVES_8AB189","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/sonoff7/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff7-4489","IPAddress":"192.168.0.61"}
00:00:05 MQT: tele/sonoff7/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: stat/sonoff7/RESULT = {"POWER":"ON"}
00:00:06 MQT: stat/sonoff7/POWER = ON (retained)
00:00:06 MQT: stat/sonoff7/RESULT = {"POWER":"ON"}
00:00:06 MQT: stat/sonoff7/POWER = ON (retained)
21:46:31 MQT: tele/sonoff7/STATE = {"Time":"2018-08-12T21:46:31","Uptime":"0T00:00:13","Vcc":3.208,"POWER":"ON","Wifi":{"AP":1,"SSId":"Pretty Fly For A WiFi","RSSI":58,"APMac":"C0:3E:0F:39:B6:BD"}}
21:49:04 MQT: stat/sonoff7/RESULT = {"OtaUrl":"https://github.com/arendst/Sonoff-Tasmota/releases/download/v6.1.1/sonoff.bin"}
21:49:04 MQT: stat/sonoff7/RESULT = {"Upgrade":"Version 6.1.1-minimal from https://github.com/arendst/Sonoff-Tasmota/releases/download/v6.1.1/sonoff.bin"}
21:49:07 HTP: Web server stopped
21:49:08 HTP: Web server active on sonoff7-4489 with IP address 192.168.0.61
21:49:11 HTP: Web server stopped
21:49:11 HTP: Web server active on sonoff7-4489 with IP address 192.168.0.61
21:49:13 HTP: Web server stopped
21:49:13 HTP: Web server active on sonoff7-4489 with IP address 192.168.0.61
21:49:15 HTP: Web server stopped
21:49:15 HTP: Web server active on sonoff7-4489 with IP address 192.168.0.61
21:49:17 HTP: Web server stopped
21:49:18 HTP: Web server active on sonoff7-4489 with IP address 192.168.0.61
21:49:22 MQT: stat/sonoff7/UPGRADE = Failed HTTP error: connection refused
```

Jason2866, 2018-08-12:
Last idea from my side. Download precompiled sonoff minimal from here and do upload this file.
After that try update to full version. If this doesnt work i see only the way via serial flash.

SurfingSpider, 2018-08-12:
thanks for your advise. I tried it but no joy.

Frogmore42, 2018-08-13:
Well then you will need to flash via serial.  That will always work, unless what you are trying to flash to is defective, even then, the flashing will work, but the device won't do what you want.

ascillato2, 2018-08-18:
Hi,

Have you managed to solve your issue?

SurfingSpider, 2018-08-18:
Afraid not. I will flash it via serial. Just not got round to it yet.

pmknowles, 2018-08-30:
Just to add to this I've had a couple of devices flashed OTA to 6.11 minimal. It reports upload successful but the device doesn't come back online.
Connecting to serial the device is running in AP mode. Sending the wifi details is unsuccessful because the details are not persisted (according to the message). Using WPS is also unsuccessful - it connects but loses the config on reboot. The only way to recover is to reflash via FTDI.

ascillato2, 2018-09-03:
@SurfingSpider

As this issue is on HOLD since 13 aug, I'm closing it temporarily until you have time to continue with it. Please, as soon as you have more information/questions, ask to reopen this issue and we gladly continue with it. Thanks.

closed 2018-09-03 by ascillato2

SurfingSpider, 2018-09-03:
Hi I re flashed by USB FTDI last week and it is back working again.