arendst/Tasmota #3464: Sonoff SC Temperature always the same.

Landixus, 2018-08-12:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
11:49:51 MQT: tele/sonoffSC1/SENSOR = {"Time":"2018-08-12T11:49:51","SonoffSC":{"Temperature":28.0,"Humidity":14.0,"Light":10,"Noise":20,"AirQuality":100},"TempUnit":"C"}
```

It is always 28.0, i have the Orginal DHT in the SC Box.
All other values are changing.
DHT Module defect? Or need i to define the DHT Module in Source?

arendst, 2018-08-12:
Enable more logging with command ``weblog 4`` and look for a message like
```
14:15:31 SER: Received AT+UPDATE="humidity":87,"temperature":23,"light":9,"noise":2,"dusty":1
```
Verify if the temperature does not change. If so the sensor is either broke or the SC is in a constant temperature location ;-)

Landixus, 2018-08-13:
Ah, temperature do change.
The problem is that the DHT inside is not accurate.
I think i will change in the future to BME 280 or DHT22
Thanks for the great software. :)

closed 2018-08-13 by Landixus