arendst/Tasmota #3466: Allow Time#Minute Modulo n Trigger

fxkai, 2018-08-12:
We want to turn on a pump every 5 minutes for 20 seconds using a sonoff Pow. Currently we solve this by this nasty piece of code:

```
# Create self repeating rule
rule1 on Rules#Timer=1 do backlog power on; ruletimer 300 endon
# Auto Turn off after 20 seconds
pulsetime 200
# Enable timer/trigger
ruletimer 1
```

Using a minute trigger which accepts a modulo would be easy to implement and solve the problem in a clean fashion such as :

```
rule 1 on Time#Minute%5 do ... endon
```

curzon01, 2018-08-13:
That's no nasty, but you can do it without pulsetime

`rule1 on Rules#Timer=1 do backlog power on;delay 200;power off;ruletimer 300 endon`

fxkai, 2018-08-13:
Thanks, unfortunately the piece of code you suggest did not work for us since the delay was not respected (the pump turned off after 1 second)

curzon01, 2018-08-13:
No it works as expected (using current v6.1.1 from [Release](https://github.com/arendst/Sonoff-Tasmota/releases))

`dev/stat/tasmota/esp01/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-31T17:59:09","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}`

Disable previous set PulseTime, this interfer with your rule 
`PulseTime1 0`

```
# Disable PulseTime
PulseTime1 0
# create rule
rule1 on Rules#Timer=1 do backlog power on;delay 200;power off;ruletimer 300 endon
# enable rule
rule1 1
# trigger rule
ruletimer1 1

```
Result
```
11:55:21 MQT: dev/stat/tasmota/esp01/RESULT = {"POWER1":"ON"}
11:55:21 MQT: dev/stat/tasmota/esp01/POWER1 = ON
11:55:21 MQT: dev/stat/tasmota/esp01/RESULT = {"Delay":200}
11:55:41 MQT: dev/stat/tasmota/esp01/RESULT = {"POWER1":"OFF"}
11:55:41 MQT: dev/stat/tasmota/esp01/POWER1 = OFF
11:55:41 MQT: dev/stat/tasmota/esp01/RESULT = {"RuleTimer1":300}
12:00:41 MQT: dev/stat/tasmota/esp01/RESULT = {"POWER1":"ON"}
12:00:41 MQT: dev/stat/tasmota/esp01/POWER1 = ON
12:00:41 MQT: dev/stat/tasmota/esp01/RESULT = {"Delay":200}
12:01:01 MQT: dev/stat/tasmota/esp01/RESULT = {"POWER1":"OFF"}
12:01:01 MQT: dev/stat/tasmota/esp01/POWER1 = OFF
12:01:01 MQT: dev/stat/tasmota/esp01/RESULT = {"RuleTimer1":300}

```

arendst, 2018-08-13:
I have to agree with @curzon01 

But as I liked the modulo option too I just released 6.1.1.6 with this option. Keep in mind to use the "|" sign for modulo as the "%" is already used for variables and discriminating them makes the code unwanted larger.

So in your case this worked for me:
```
rule1 on Time#Minute|5 do backlog power on;delay 200;power off endon
rule1 1
rule1 4
```
results in this:
```
13:10:00 RUL: TIME#MINUTE|5 performs "backlog power on;delay 200;power off"
13:10:00 SRC: Rule
13:10:00 RSL: Group 0, Index 1, Command BACKLOG, Data power on;delay 200;power off
13:10:00 SRC: Backlog
13:10:00 RSL: Group 0, Index 1, Command POWER, Data on
13:10:00 MQT: stat/ring2/POWER = {"POWER":"ON"}
13:10:00 MQT: stat/ring2/POWER = ON
13:10:00 SRC: Backlog
13:10:00 RSL: Group 0, Index 1, Command DELAY, Data 200
13:10:00 MQT: stat/ring2/DELAY = {"Delay":200}
13:10:20 SRC: Backlog
13:10:20 RSL: Group 0, Index 1, Command POWER, Data off
13:10:20 MQT: stat/ring2/POWER = {"POWER":"OFF"}
13:10:20 MQT: stat/ring2/POWER = OFF
```

closed 2018-08-15 by arendst

fxkai, 2018-08-18:
Hi Guys,

i just checked again and for me the delay version as stated above does not work. Any ideas?

```
11:33:57 CMD: rule1 on Rules#Timer=1 do backlog power on; delay 200; power off; ruletimer 200 endon 
11:33:57 RSL: RESULT = {"Rule1":"ON","Once":"OFF","Free":431,"Rules":"on Rules#Timer=1 do backlog power on; delay 200; power off; ruletimer 200 endon "}
11:34:16 CMD: ruletimer 1
11:34:16 RSL: RESULT = {"RuleTimer1":1}
11:34:18 RUL: RULES#TIMER=1 performs "backlog power on; delay 200; power off; ruletimer 200"
11:34:18 RSL: RESULT = {"POWER1":"ON"}
11:34:18 RSL: POWER1 = ON
11:34:18 RSL: RESULT = {"POWER1":"OFF"}
11:34:18 RSL: POWER1 = OFF
11:34:18 RSL: RESULT = {"Delay":200}
```

fxkai, 2018-08-18:
can we reopen it or open a new issue?

arendst, 2018-08-18:
Pls enable more logging with weblog 3 to find out where the POWER1 OFF comes from.

RuneKyndal, 2018-09-05:
im having a similar issue.

im using a Sonoff BASIC

the built in relay controls some power

im using an external relay,  via GPIO14 (30 Relay 2i)  as its active low
i need to toggle it ON for a period  and then back off to open my garage door

im posting to topic cmnd/sonoff/backlog             POWER1 ON; delay 50; POWER1 OFF

this  DOES turn on the built in relay for 5 secs

> 
backlog | POWER1 ON; delay 50; POWER1 OFF
stat/sonoff/RESULT | {"POWER1":"ON"}
stat/sonoff/POWER1 | ON
stat/sonoff/RESULT | {"Delay":50}
stat/sonoff/RESULT | {"POWER1":"OFF"}
stat/sonoff/POWER1 | OFF
>

same command for the other relay
im posting to topic cmnd/sonoff/backlog        POWER2 ON; delay 50; POWER2 OFF

gives only about 100ms delay,   with an extra delayed "OFF"  
or indeed an extra OFF right after ON ?

cmnd/sonoff/backlog | POWER2 ON; delay 50; POWER2 OFF
stat/sonoff/RESULT | {"POWER2":"ON"}
stat/sonoff/POWER2 | ON
stat/sonoff/RESULT | {"POWER2":"OFF"}
stat/sonoff/POWER2 | OFF
stat/sonoff/RESULT | {"Delay":50}
stat/sonoff/RESULT | {"POWER2":"OFF"}
stat/sonoff/POWER2 | OFF

ideas?