arendst/Tasmota #3467: Sonnoff Tasmota does not access wifi.

klrock, 2018-08-12:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ x] Hardware used : Sonoff Basic TH_V1.1
- [x ] Provide the output of command ``status 0`` : cannot do
```
STATUS 0 OUTPUT HERE
```
After several fails, I finally loaded Tasmota (Sonohh.bin - 537Kb) on my Sonoff Basic (TR version 1.1) due to the great support here.  It was apparently sucessful as shown below.

C:\Windows\system32>esptool.py --port COM6 write_flash -fs 1MB -fm dout 0x0 c:\o
ther\sonoff.bin\sonoff.bin
esptool.py v2.3.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 549632 bytes to 374449...
Wrote 549632 bytes (374449 compressed) at 0x00000000 in 33.3 seconds (effective
132.0 kbit/s)...
Hash of data verified.
Leaving...
Hard resetting via RTS pin...

Now I am still trying to get the module onto my wifi unsuccessfully.  I have tried each of the following methods several times.  Is there a way to confirm that Tasmota was successfully loaded?

3 short presses: Start Wifi smartconfig allowing for SSID and Password configuration using an Android mobile phone with the ESP8266 SmartConfig app. The LED will blink during the config period. A single button press during this period will abort and restart sonoff.
esptouch fail
4 short presses: Start Wifi manager providing an Access Point with IP address 192.168.4.1 and a web server allowing the configuration of Wifi. The LED will blink during the config period. A single button press during this period will abort and restart sonoff.
192.168.4.1 took too long to respond.
5 short presses: Start Wifi Protected Setup (WPS) allowing for SSID and Password configuration using the router's WPS button or webpage. The LED will blink during the config period. A single button press during this period will abort and restart sonoff.
Asus - Fail due to WPS time out!
Also checked for Sonoff-xxxx access point on my iphone but none found.

**(Please, remember to close the issue when the problem has been addressed)**

Frogmore42, 2018-08-13:
Since you have Windows, you can use teraterm
https://ttssh2.osdn.jp/index.html.en

Connect the sonoff like you did when you were flashing it, but do not connect gpio 0 to ground, because you want it to boot normally.  Use TeraTerm to connect to the serial port and get the log and the results of status 0.  You can also use the commands to configure it for your WiFi.

klrock, 2018-08-13:
I also tried Putty, serial, COM6 and serial category.

I got same experience as others:
Doesn’t allow me to type anything.
The black window opens, green cursor flashes and that’s it.
Doesn’t accept any keyboard input. (Yes, the keyboard works)

Frogmore42, 2018-08-13:
Are you sure COM6 is the correct port?
TeraTerm will work with any COM port in my experience, even a double digit one (I have never managed to get all the way to three digits for COM ports, but I expect it would work for that too), but that COM port has to actually be preset and visible to the system.  If you open up Device Manager:
<Windows Key>X 
M

Then open up Ports (COM & LPT), what does it show?

What are you using for your adapter.  Many FTDI ones do not provide enough power to flash/work correctly without an external 3V3 supply.

While there are not a lot of things you have to do get flashing to work, there are several that if you don't do it exactly right will result in a bad experience.

klrock, 2018-08-13:
Yes COM6 had been taken from Device Manager previously and it is still the same CP210x.

Reading the Teraterm documentation, I found that the number of COMs is controlled by the maxcomport=x line in tereaterm.ini. I have edited the max to 10, but when I try to save it I get "access denied". I don't understand this as I am administrator.

Frogmore42, 2018-08-13:
Remove the read only flag on the file.  You can do this from Explorer by right click --> Properties

klrock, 2018-08-14:
So I finally got Tera Term to read COM6. However the display does not show anything when I type. Every few minutes some gibberish shows up such as
€ý
!9zIøÿÿàóSXüÐ
):ÿvþèAáÀ»¹³€¨ÿ€`Á

Where from here?

klrock, 2018-08-14:
I am not sure what caused the tasmota-Sonoff to finally connect, but seeing the gibberish in Tera Term, I decided to see if the AP was available. It was and I signed on to complete the wifi logon. It is now visible on the router and I will start configuration tomorrow.
Any suggestions on MQTT?

Frogmore42, 2018-08-14:
You need to set the baud rate in teraterm.  I believe it is 115200.

For MQTT, I would highly recommend it.
If you wanted to know which one.  I use mosquitto, as do many other people.

klrock, 2018-08-14:
Thanks for all the help.

Great support.

closed 2018-08-14 by klrock