arendst/Tasmota #3469: Integrating custom DIY RF/IR bridge

bilogic, 2018-08-13:
**Is your feature request related to a problem? Please describe.**
My DIY RF/IR bridge does not have MQTT or SmartConfig

**Describe the solution you'd like**
Integrate my bridge with Tasmota properly so that it can utilize future versions of Tasmota.

I have been looking at Tasmota's source, there are ```xdrv*```, ```xplg*``` and ```xsns*```, but I can't make sense of them.

What files should I create to integrate my custom code cleanly?
I have a Wemos D1 mini as a development board.

**Why not use the Sonoff RF bridge?**
My bridge can receive and transmit IR/315/433. From what I know, the Sonoff bridge supports only 433.

Thank you.

Jason2866, 2018-08-13:
Without a exact description of your diy bridge nobody can implement or can give you help.

bilogic, 2018-08-13:
I'm planning to write the code myself, I just don't know how to name them and which folder to create them in (integration).

1. IR is just a infrared LED bulb.
2. Both 315/433 RF modules are these https://www.ebay.com/itm/2PCS-433Mhz-WL-RF-transmitter-and-receiver-link-kit-for-Arduino-ARM-MCU-Good/311569769620?epid=1773673828&hash=item488b016494:g:9EIAAOSwvv9Ze-1-

So, I ask only 1 question (with *cleanly* being the keyword):

> What files should I create to integrate my custom code cleanly?

Maybe you can tell me what other info you need and I can answer them.

bilogic, 2018-08-13:
Ok thanks, let me have a look, but I still would like to understand the naming convention of ```xdrv*```, ```xplg*``` and ```xsns*``` so that I can customize for other stuff, e.g. LoRa bridge. 

What transmitter do you use then?

Jason2866, 2018-08-13:
Receiver
https://www.google.de/imgres?imgurl=https%3A%2F%2Fi.ebayimg.com%2Fimages%2Fg%2Fez8AAOSwImRYQnX3%2Fs-l300.jpg&imgrefurl=https%3A%2F%2Fwww.ebay.co.uk%2Fitm%2FNew-RXB6-433Mhz-Superheterodyne-Wireless-Receiver-Module-for-Arduino-ARM-AVR-%2F291961239406&docid=d1GYujmhCn3dNM&tbnid=W2xuG6MYa6P8zM%3A&vet=1&w=300&h=300&bih=768&biw=1024&ved=2ahUKEwj_6az8vercAhVIK1AKHZpPDDwQxiAoCnoECAEQMA&iact=c&ictx=1

Jason2866, 2018-08-13:
Transmitter from Qiachip is ok... 
The transmitter and receiver in the Rf-Bridge are real good...
Thats the reason i use the bridge for Rf433 Mhz devices. It is hard to get this quality by a Diy project.

Frogmore42, 2018-08-13:
@bilogic  read this: https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-API

Frogmore42, 2018-08-13:
and this https://github.com/arendst/Sonoff-Tasmota/wiki/PID-Control-with-Sonoff-Devices

Jason2866, 2018-08-14:
Maybe this is interesting
https://github.com/1technophile/OpenMQTTGateway

bilogic, 2018-08-15:
@Jason2866 thank you! It was what I have in mind.

closed 2018-08-15 by bilogic